tttapa/Control-Surface #74: Transpose NoteButton via a NC pushbutton?

samlochner, 2019-12-11:
I'm trying to figure out how to transpose a NoteButton up one octave. Is there a simple way to do this without having to redeclare the NoteButton?

Also, I'm hoping to make the transpose occur when I push a normally closed push button. Is this something that can be done with the Control Surface library?

Thank you kindly,
Sam

tttapa, 2019-12-11:
I added an example, but I didn't test it yet.

<https://tttapa.github.io/Control-Surface-doc/Doxygen/db/d5c/Transpose-Octave-NC-Button_8ino-example.html>

tttapa, 2019-12-11:
Note that I fixed a refactoring bug as well, so you'll have to update to the latest version of the library.

samlochner, 2019-12-12:
Thanks tttapa,

If I'm not using bankable, just NoteButtons, what would it look like?

I tried:
NoteButton button8 = {transposer, 8, {72, CHANNEL_1},127};

But I see there's no slot for the transposer parameter.

Thanks,
Sam

tttapa, 2019-12-12:
You have to use the bankable version (transposing is a special version of banking, both shift the address up or down).

samlochner, 2019-12-12:
Hi tttapa, the usage of bankable is a bit over my head at the moment.

What would the bankable equivalent declaration of this be:
NoteButton button8 = {8, {72, CHANNEL_1},127};

Thanks,
Sam

samlochner, 2019-12-12:
got it!

Trying it out, I see that when I press my transpose pushbutton, it goes up an octave, and then when I push it again, it goes down an octave back to the original. Is there a way to make it so it just keeps going up octaves? So each time I push it it goes up one more octave? I'll then have another button for going down octaves.

This is great, thanks!

tttapa, 2019-12-12:
Something like this?

```cpp
#include <Control_Surface.h>
 
USBMIDI_Interface midi;
 
// Transposer that transposes one octave (12 semitones) at a time,
// from 0 to +5 octaves.
Transposer<0, +5> transposer(12);
 
// Push button on pin 2 to transpose up, pin 3 to transpose down.
IncrementDecrementSelector<6> selector = {transposer, {2, 3}};
 
using namespace MIDI_Notes;
Bankable::NoteButton notebutton = {
  transposer,      // bank/transposer
  8,               // pin
  {72, CHANNEL_1}, // address/note
  127,             // velocity
};
 
void setup() {
  selector.invert(); // normally closed button
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
}
```

samlochner, 2019-12-12:
You nailed it, very nice.

Last request, is it possible to have a homing button, so a separate pushbutton brings it back to original octave?

Really appreciate all you help on this tttapa

tttapa, 2019-12-12:
You can press the two buttons at once to reset it.

samlochner, 2019-12-12:
Wow, that's convenient, such a well thought out library.
Closing this thread.
Thanks again for all your help!

closed 2019-12-12 by samlochner

STOP63, 2023-12-04:
`#include <Control_Surface.h>
 
USBMIDI_Interface midi;
 
// Transposer that transposes one octave (12 semitones) at a time,
// from 0 to +5 octaves.
Transposer<0, +5> transposer(12);
 
// Push button on pin 2 to transpose up, pin 3 to transpose down.
IncrementDecrementSelector<6> selector = {transposer, {2, 3}};
 
using namespace MIDI_Notes;
Bankable::NoteButton notebutton = {
  transposer,      // bank/transposer
  8,               // pin
  {72, CHANNEL_1}, // address/note
  127,             // velocity
};
 
void setup() {
  selector.invert(); // normally closed button
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
}`

I have experimented with various octave transposers, but unfortunately, none of them are proving effective. The script consistently plays note 72, and I've observed that the buttons on pins 2 and 3 do not yield any noticeable results. I would greatly appreciate any assistance or guidance in resolving this issue. Thank you.

EugeneSolomko, 2024-02-28:
> `#include <Control_Surface.h>
> 
> USBMIDI_Interface midi;
> 
> // Transposer that transposes one octave (12 semitones) at a time, // from 0 to +5 octaves. Transposer<0, +5> transposer(12);
> 
> // Push button on pin 2 to transpose up, pin 3 to transpose down. IncrementDecrementSelector<6> selector = {transposer, {2, 3}};
> 
> using namespace MIDI_Notes; Bankable::NoteButton notebutton = { transposer, // bank/transposer 8, // pin {72, CHANNEL_1}, // address/note 127, // velocity };
> 
> void setup() { selector.invert(); // normally closed button Control_Surface.begin(); }
> 
> void loop() { Control_Surface.loop(); }`
> 
> I have experimented with various octave transposers scripts , but unfortunately, none of them is working. The script (above) consistently plays note 72, and I've observed that the buttons on pins 2 and 3 do not yield any noticeable results. I would greatly appreciate any assistance or guidance in resolving this issue. Thank you.

#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Instantiate a Transposer that can transpose from 5 octaves down to 5 octaves up
Transposer<-5, +5> transposer(12);

// Instantiate a Selector to change the transposition
IncrementDecrementSelector<transposer.getNumberOfBanks()> selector {
  transposer,
  {10, 11}, // pins for buttons 
  Wrap::Clamp,
};

// Instantiate an array of NoteButton objects
Bankable::NoteButton buttons[] {
  {transposer, 2, MIDI_Notes::C[4]},
  {transposer, 3, MIDI_Notes::D[4]},
  {transposer, 4, MIDI_Notes::E[4]},
  {transposer, 5, MIDI_Notes::F[4]},
  {transposer, 6, MIDI_Notes::G[4]},
  {transposer, 7, MIDI_Notes::A[4]},
  {transposer, 8, MIDI_Notes::B[4]}, 
  {transposer, 9, MIDI_Notes::C[5]},
};
void setup() {
  //selector.invert(); // normally closed button
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

THIS ONE WORKED FOR ME