tttapa/Control-Surface #75: Teensy 4

rogerarends, 2019-12-13:
Hi
Can you add support to your library for the Teesy 4 for usbmidi. Also MCU::LCD for one channel

tttapa, 2019-12-14:
The Teensy Core doesn't support USB MIDI for the Teensy 4 yet.

The LCD doesn't really have a notion of tracks. It's just a buffer of 120 characters. If you only need a certain range of characters, you can specify the length and the offset of the range you want: <https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d2e/classMCU_1_1LCD.html#ae03259fb1de07a86f794e670c51f114a>

rogerarends, 2019-12-15:
https://forum.pjrc.com/threads/58654-Teensyduino-1-49-Beta-2 

Add Teensy 4.0 USB MIDI, Keyboard, Mouse, Joystick ...

So when you do have the time could you? putting the project on hold till end Jan
Thanks

closed 2019-12-15 by rogerarends

tttapa, 2019-12-15:
I won't have time until my exams are over in early February.

rogerarends, 2019-12-15:
No problem.

tttapa, 2019-12-18:
It was easier than expected, did a quick test and both sending and receiving notes seemed to work fine.

I haven't tested multiple USB MIDI cables or USB Audio, but I don't really expect any problems.

commit c318c75f (2019-12-18): Cleanup USB MIDI, add Teensy 4 support
Fix #75