tttapa/Control-Surface #77: Using Control-Surface for Piezo Drum MIDI trigger and Teensy 4.0

rcappuccio, 2019-12-20:
Hi, I would like to use your wonderful library for developing a Piezo Drumkit.
I noticed that piezos are not mentioned among the inputs covered by your library.
Would it be difficult to implement them?
Anyway, would you be so kind to provide an example code for that purpose?
I tested the examples and this one works with my mux and some piezos:

#include <Control_Surface.h>

CD74HC4067 mux = { 14, {0, 1, 2, 3} };

void setup() {
  // put your setup code here, to run once:
  USBMIDI_Interface midi;

  Serial.begin(115200);
  mux.begin(); // Initialize multiplexer
}

void loop() {
  for (pin_t pin : mux.pins()) {
    int reading = analogRead(pin);
    if (reading > 20) {
        Serial.print(reading);
    } else {
        Serial.print(0);
    }
    Serial.print('\t');
  }
  Serial.println();
}

tttapa, 2019-12-20:
I don't have time to write and test an example right now.

Some remarks: 
- In your code, he USBMIDI_Interface only exists in the setup function. You have to make it global if you want to use it in the loop.
- You have to call `midi.begin()` in the setup.
- You'll have to find a way to do edge detection to know when to send a MIDI Note On event, you can't continuously send it in the same way you're doing now.
- You'll also have to find a way to know when to send the Note Off event. Just sampling the input once to check if the amplitude is lower than a threshold won't work, as piezos generate an exponentially decaying sinusoidal signal. If you randomly measure close to a zero crossing of the sine wave, you would incorrectly think that the signal has died off, even though this isn't the case. You'll have to use some kind of maximum peak detector to know when the peaks of the sine wave are below a threshold.  
Another approach would be to start a timer when sending a Note On even, and just send the Note Off even x milliseconds later.
- Piezos can generate very high voltages, both positive and negative. If you connect them to an Arduino's inputs directly, you will probably damage it. Piezos generate sinusoidal AC signals. The Arduino cannot handle the negative voltages.

Here's an example that demonstrates how to send MIDI notes using the library: <https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d8d/Send-MIDI-Notes_8ino-example.html>

rcappuccio, 2019-12-20:
Hi,

thanks for your answer.
I decided to give your library a try because all my previous attempts with the Teensyduino and other libraries failed.
Some time ago I used the Arduino Uno and I had no problems.
The problem with the Teensy 4.0 is that it is not 5 volts tolerant and, more important, it is too fast for the multiplexers, generating a lot of crosstalk, but when I added a delay in the other mux libraries, they stopped working.
Your library, instead, works correctly just out of the box, even at 600MHz.
I know that my code, as it is, is not usable for the Piezo Drum trigger. Paul Stoffregen answered a question in the Teensz forum, publishing a nice example code for the capture of the electric peak and I will follow that trace.
What I was asking myself is: would it be easy to create a class in your code for a piezo input?
What class could I use as a basis for it?
Would that class represent a nice addition to your library?

Thanks

Roberto

Da: tttapa
Inviato: venerdì 20 dicembre 2019 13:24
A: tttapa/Control-Surface
Cc: rcappuccio; Author
Oggetto: Re: [tttapa/Control-Surface] Using Control-Surface for Piezo DrumMIDI trigger and Teensy 4.0 (#77)

I don't have time to write and test an example right now.
Some remarks:
• In your code, he USBMIDI_Interface only exists in the setup function. You have to make it global if you want to use it in the loop.
• You have to call midi.begin() in the setup.
• You'll have to find a way to do edge detection to know when to send a MIDI Note On event, you can't continuously send it in the same way you're doing now.
• You'll also have to find a way to know when to send the Note Off event. Just sampling the input once to check if the amplitude is lower than a threshold won't work, as piezos generate an exponentially decaying sinusoidal signal. If you randomly measure close to a zero crossing of the sine wave, you would incorrectly think that the signal has died off, even though this isn't the case. You'll have to use some kind of maximum peak detector to know when the peaks of the sine wave are below a threshold.
Another approach would be to start a timer when sending a Note On even, and just send the Note Off even x milliseconds later.
• Piezos can generate very high voltages, both positive and negative. If you connect them to an Arduino's inputs directly, you will probably damage it. Piezos generate sinusoidal AC signals. The Arduino cannot handle the negative voltages.
Here's an example that demonstrates how to send MIDI notes using the library: https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d8d/Send-MIDI-Notes_8ino-example.html

tttapa, 2019-12-20:
> Your library, instead, works correctly just out of the box, even at 600MHz.

I'm glad to hear that :)

> Would that class represent a nice addition to your library?

Sure!

Creating a class for it should be relatively easy. You can use the [`NoteButton`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d80/NoteButton_8hpp_source.html) class as a reference, and especially its parent class [`MIDIButton`](https://tttapa.github.io/Control-Surface-doc/Doxygen/dd/d47/Abstract_2MIDIButton_8hpp_source.html).

The basic structure is as follows:
- All MIDI elements inherit from a common base class, [`MIDIOutputElement`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/d41/MIDIOutputElement_8hpp_source.html#l00009) (actually, it's an alias for `AH::Updatable`).  
You have to override the `begin` method, which is called once upon initialization in the setup, and the `update` method, which is called continuously in the loop, and should read the input and send the MIDI messages. I think [`MIDIButton`](https://tttapa.github.io/Control-Surface-doc/Doxygen/dd/d47/Abstract_2MIDIButton_8hpp_source.html) is a good and simple example that you can base your class on.
- The logic for reading the input is in the `MIDI_Outputs/Abstract` folder, `MIDIButton` in this example. The classes in the `Abstract` folder are generic in the MIDI Sender type, which means that `MIDIButton` can serve as a base for both `NoteButton` and `CCButton`.
- The `MIDI_Outputs` folder contains these `NoteButton` and `CCButton` classes, and all they do is basically passing the right arguments to the `MIDIButton` constructor, and provide more specific documentation.
- The MIDI Senders, like [`DigitalNoteSender`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/daa/DigitalNoteSender_8hpp_source.html) are in the `MIDI_Senders` folder. You'll probably have to create a note sender that supports velocity as well.

If you face any problems, just let me know! I'm studying for my exams right now, so I won't be able to write much code, but I may be able to point you in the right direction.

Thanks,

Pieter

rcappuccio, 2020-01-14:
Hi, 
I successfully compiled your first example in the Arduino IDE.
I would like to work with PlatformIO, though, but when I try to compile the first example, for a Teensy 4.0 board, I get the following errors:

Processing teensy40 (platform: teensy; board: teensy40; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy 4.6.0 > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 1MB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1), framework-arduinoteensy 1.148.0 (1.48)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 93 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Control Surface> 1.1.0
|   |-- <SPI> 1.0
|   |-- <Encoder>
|   |-- <Audio> 1.3
|   |   |-- <SerialFlash>
|   |   |   |-- <SPI> 1.0
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|   |   |-- <SD> 1.2.2
|   |   |   |-- <SPI> 1.0
|   |-- <FastLED> 3.3.1
|   |   |-- <SPI> 1.0
|   |   |-- <SoftwareSerial> 1.0
|   |   |-- <DmxSimple> 3.1
|   |-- <SoftwareSerial> 1.0
|   |-- <MIDIUSB> 1.0.4
|-- <MIDI Library> 4.3.1
|   |-- <MIDIUSB> 1.0.4
Building in release mode
Compiling .pio\build\teensy40\src\main.cpp.o
Archiving .pio\build\teensy40\libb0e\libSerialFlash.a
Compiling .pio\build\teensy40\libc06\Wire\WireIMXRT.cpp.o
Compiling .pio\build\teensy40\libc06\Wire\WireKinetis.cpp.o
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\teensy40\libc06\Wire\utility\twi.c.o
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:9: warning: ISO C99 requires whitespace after the macro name
Compiling .pio\build\teensy40\lib69b\SD\File.cpp.o
Compiling .pio\build\teensy40\lib69b\SD\SD.cpp.o
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
Archiving .pio\build\teensy40\libc06\libWire.a
Compiling .pio\build\teensy40\lib69b\SD\cache_t3.cpp.o
In file included from C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/Control_Surface.h:117:0,
Compiling .pio\build\teensy40\lib69b\SD\card_t3.cpp.o
                 from src\main.cpp:13:
Compiling .pio\build\teensy40\lib69b\SD\dir_t3.cpp.o
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp:284:5: error: 'usb_packet_t' does not name a type
     usb_packet_t *rx_packet = nullptr;
     ^
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp: In member function 'void CS::USBMIDI_Interface::writeUSBPacket(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp:87:41: error: 'usb_midi_write_packed' was not declared in this scope
                               (d2 << 24));      // data 2
                                         ^
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp: In member function 'virtual CS::MIDI_read_t CS::USBMIDI_Interface::read()':
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp:185:17: error: 'rx_packet' was not declared in this scope
             if (rx_packet == nullptr) { // If there's no previous packet
                 ^
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp:189:21: error: 'MIDI_RX_ENDPOINT' was not declared in this scope    
                     MIDI_RX_ENDPOINT); // Read a new packet from the USB buffer
                     ^
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp:189:37: error: 'usb_rx' was not declared in this scope
[... 11 lines cut ...]
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp:215:36: error: 'MIDI_RX_ENDPOINT' was not declared in this scope    
                 rx_packet = usb_rx(MIDI_RX_ENDPOINT); // Read the next packet
                                    ^
C:\Users\roberto\.platformio\lib\Control Surface_ID6793\src/MIDI_Interfaces/USBMIDI_Interface.hpp:215:52: error: 'usb_rx' was not declared in this scope
                 rx_packet = usb_rx(MIDI_RX_ENDPOINT); // Read the next packet
                                                    ^
src\main.cpp: At global scope:
src\main.cpp:16:19: error: 'CS::USBMIDI_Interface midi' redeclared as different kind of symbol
 USBMIDI_Interface midi;
                   ^
In file included from C:\Users\roberto\.platformio\packages\framework-arduinoteensy\libraries\MIDI\src/midi_Defs.h:30:0,
                 from C:\Users\roberto\.platformio\packages\framework-arduinoteensy\libraries\MIDI\src/MIDI.h:30,
                 from src\main.cpp:12:
C:\Users\roberto\.platformio\packages\framework-arduinoteensy\libraries\MIDI\src/midi_Namespace.h:31:66: note: previous declaration 'namespace midi { }'
 #define BEGIN_MIDI_NAMESPACE            namespace MIDI_NAMESPACE {
[... cut ...]
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
*** [.pio\build\teensy40\src\main.cpp.o] Error 1
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
===================================================================== [FAILED] Took 7.17 seconds =====================================================================
The terminal process terminated with exit code: 1

tttapa, 2020-01-16:
PlatformIO doesn't have Teensy 4.0 USB MIDI support yet. This was added very recently in Teensyduino 1.49.

You could either wait for PIO to update the Teensy Core, use the Arduino IDE for now, or manually link the latest Teensyduino version into your PIO installation: https://forum.pjrc.com/threads/58781-PlatformIO-how-to-compile-for-Teensy-4-0-with-beta-core

For Teensy, you can use the following in your `platformio.ini` file:
```ini
[env:teensy]
platform = teensy
board = teensy36
framework = arduino
build_flags = -D USB_MIDI_SERIAL
lib_ignore = MIDIUSB
```

rcappuccio, 2020-01-16:
Hi,
thanks for the answer.
I copied the cores from the new Teensyduino 1.49 into Platformio.
The missing part was:

build_flags = -D USB_MIDI_SERIAL

Now it works.

Regards