tttapa/Control-Surface #79: Cannot compile your library on windows/platformio as the paths are too long for windows python

justin-romano, 2019-12-30:
I *think* (not certain because the console only lets me see 100 characters) the only issue is you need to prepend \\\\\\\\?\\\\ to your path if it is going to exceed 260 characters. 

e.g. in my case
```
\\\\?\\d:\\docs\\PlatformIO\\Projects\\espmidihost\\.pio\\libdeps\\esp32dev\\_tmp_installing-cbxnqbhl-package\\.\\examples\\1. MIDI 
```
This is an legacy limitation for windows and very annoying.
In any case it wont install the library.
platformio.ini
```ini
lib_deps =
    [env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_port = com3
monitor_speed = 112500
lib_deps =
    6793
```
error message
```console
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing id=6793
Using cache: C:\Users\Administrator\.platformio\.cache\f7\b3b844e5433ea226ddde1a015a8cf5f7
Unpacking...
FileNotFoundError: [WinError 206] The filename or extension is too long: 'd:\\docs\\PlatformIO\\Projects\\espmidihost\\.pio\\libdeps\\esp32dev\\_tmp_installing-cbxnqbhl-package\\.\\examples\\1. MIDI 
Output\\2. Buttons & Switches\\4. Push Buttons to Increment or Decrement Controls and to Scroll through Lists\\CCIncrementDecrementButtons':
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
```

tttapa, 2019-12-30:
Thanks for letting me know. The problem with filename length is a problem with Windows, and I think it has to be fixed/circumvented in the PlatformIO builder.

I'll use a shorter path for now.

closed 2019-12-31 by tttapa

justin-romano, 2019-12-31:
> Closed #79 <https://github.com/tttapa/Control-Surface/issues/79> via
> 101919d

Yeah i was thinking you might not be able to fix it from your side but i
contacted you because you most likely care about it. Cheers

tttapa, 2019-12-31:
Absolutely, I'm not a Windows user myself, so these are the kinds of problems that I wouldn't be able to find otherwise.  
Thanks again.

commit 101919d6 (2019-12-31): Shorter filenames for examples (Windows limitation)
Fix #79