tttapa/Control-Surface #80: Potentiometer "map" "calibrate"

SaturnAmps, 2020-01-04:
Uploading the medium-old CCpotentiometer-map code exactly as it was written, (no changes) makes the potentiometer not work at all it does not even flash the red led in the pro micro board when moving the pot as if it is not reading the pot movement.

tttapa, 2020-01-04:
The old examples are no longer supported. I will probably update some of them when I find the time.

For me, it doesn't compile (and it shouldn't compile either):
```
invalid conversion from 'uint8_t (*)(uint8_t) {aka unsigned char (*)(unsigned char)}' to 'CS::MappingFunction {aka unsigned int (*)(unsigned int)}' [-fpermissive]
```

The map behavior was changed some time ago (<https://github.com/tttapa/Control-Surface#recent-breaking-changes>), and now all values are between 0 and 16383 (i.e. everything is 128 times larger). The signature for the map function is now `analog_t function(analog_t raw_value)`.

I cannot test anything at the moment, but you could try this:

```cpp
/**
 * @brief   This is an example of the `CCPotentiometer` class of the 
 *          Control_Surface library, and specifically, how to use the `map`
 *          function to account for non-linearities of the potentiometer.
 * 
 * Connect a potentiometer to analog pin A0. This will be the MIDI channel 
 * volume of channel 1.  
 * Map it in your DAW or DJ software.
 * 
 * Written by Pieter P, 16-06-2019  
 * https://github.com/tttapa/Control-Surface
 */

// Include the Control Surface library
#include <Control_Surface.h>

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Create a new instance of the class `CCPotentiometer`, called `potentiometer`,
// on pin A0, that sends MIDI messages with controller 7 (channel volume)
// on channel 1.
CCPotentiometer potentiometer = {
  A0, {MIDI_CC::Channel_Volume, CHANNEL_1}
};

// The filtered value read when potentiometer is at the 0% position
constexpr analog_t minimumValue = 255;
// The filtered value read when potentiometer is at the 100% position
constexpr analog_t maximumValue = 16383 - 255;

// A mapping function to eliminate the dead zones of the potentiometer:
// Some potentiometers don't output a perfect zero signal when you move them to
// the zero position, they will still output a value of 1 or 2, and the same
// goes for the maximum position.
analog_t mappingFunction(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

void setup() {
    // Add the mapping function to the potentiometer
    potentiometer.map(mappingFunction);
    // Initialize everything
    Control_Surface.begin();
    Serial.begin(115200);
}

void loop() {
    // Update the Control Surface (check whether the potentiometer's
    // input has changed since last time, if so, send the new value over MIDI).
    Control_Surface.loop();
    // Use this to find minimumValue and maximumValue: it prints the raw value
    // of the potentiometer, without the mapping function
    Serial.println(potentiometer.getRawValue());
}
```

SaturnAmps, 2020-01-04:
It worked as intended when I changed: 
The filtered value read when potentiometer is at the 100% position from:
constexpr analog_t maximumValue = 16383 - 255; 
To:
constexpr analog_t maximumValue = 16383;
Tested it on reaper and it works.
Thank you for your quick response and solution.

closed 2020-01-04 by SaturnAmps

diegogauffin, 2020-03-28:
hi! this library it's awsome ! 
I have question about this issue, i have a list of bankable potentiometers (on a MUX), and want to MAP one of them, how could i write the code for that ? I ll put here the code i am using now ...

#include <Control_Surface.h> 

 #include <Selectors/IncrementDecrementSelector.hpp>
 
USBMIDI_Interface midi;  
 
CD74HC4067 mux = {
  A3,       
  {2, 3, 4, 5} 
};

Bank<4> bank(2);

IncrementDecrementSelector<4> selector = {
    bank,       // Bank to manage
    {6, 16},     // push button pins (increment, decrement)
    Wrap::Wrap, // Wrap around
};
IncrementDecrementSelectorLEDs<4> bankSelector = {
  bank,
  {16, 6},         // button pins
  {0,1,19,20}, // LED pins
};

Bankable::CCPotentiometer myPots[] = {
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(0),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_1,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
    {                             // myPot1
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(1),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_2,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
    {                             // myPot2
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(2),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_3,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
     {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(3),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_4,      // (3.1)
            CHANNEL_1,                    // (3.2)
[... 130 lines cut ...]
    return map(raw, minimumValue, maximumValue, 0, 16383);  
  
};

Bankable::NoteButtonLatching switches[] = {
  {{bank, BankType::CHANGE_ADDRESS}, 7, MCU::MUTE_1},
  {{bank, BankType::CHANGE_ADDRESS}, 8, MCU::MUTE_2},
  {{bank, BankType::CHANGE_ADDRESS}, 9, MCU::MUTE_3},
  {{bank, BankType::CHANGE_ADDRESS}, 10, MCU::MUTE_4},
  {{bank, BankType::CHANGE_ADDRESS}, 18, MCU::SELECT_2},
  {{bank, BankType::CHANGE_ADDRESS}, 15, MCU::SELECT_1},
};

void setup (){
  potentiometer.map(mappingFunction);
Control_Surface.begin(); 

}
 
void loop() {
  Control_Surface.loop(); 
}
   

THANK YOU!!!