tttapa/Control-Surface #81: Multiple IncrementDecrement Buttons

rogerarends, 2020-01-08:
How would I add 3 or more, eg;
one for JOG_Wheel and Master volume

#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Instantiate a CCIncrementDecrementButtons object
CCIncrementDecrementButtons buttons = {
  {5, 6},              // Button pins: 5 increments, 6 decrements
  MCU::V_POT_1,        // Increment/Decrement CC address
  1,                   // Multiplier
  MCU::V_POT_SELECT_1, // Reset note address
};

void setup() {
  // Use the Mackie Control protocol for sending relative MIDI CC messages.
  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);

  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the control surface
}

tttapa, 2020-01-08:
Just create more instances:

```cpp
// Instantiate a CCIncrementDecrementButtons object
CCIncrementDecrementButtons vpotbtn1 = {
  {5, 6},              // Button pins: 5 increments, 6 decrements
  MCU::V_POT_1,        // Increment/Decrement CC address
  1,                   // Multiplier
  MCU::V_POT_SELECT_1, // Reset note address
};

CCIncrementDecrementButtons vpotbtn2 = {
  {7, 8},              // Button pins: 7 increments, 8 decrements
  MCU::V_POT_2,        // Increment/Decrement CC address
  1,                   // Multiplier
  MCU::V_POT_SELECT_2, // Reset note address
};
```
The reset note is optional, see [**documentation**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d60/classCCIncrementDecrementButtons.html#ab1c0779d2b64afc66abbc0fa6c5f0dd3):

```cpp
CCIncrementDecrementButtons vpotbtn2 = {
  {9, 10},              // Button pins: 9 increments, 10 decrements
  MCU::JOG_WHEEL,    // Increment/Decrement CC address
  1,                   // Multiplier
};
```
Master volume uses Pitch Bend, not relative CC, so it won't work with `CCIncrementDecrementButtons`.  
It's pretty straightforward to implement the same functionality, but I would suggest taking a step back to learn the language using some simpler examples first.

Once you know some more C++, you can use this as inspiration: <https://tttapa.github.io/Control-Surface-doc/Doxygen/db/d20/Abstract_2MIDIIncrementDecrementButtons_8hpp_source.html>

In the `send` method, add the delta to a `position` variable, constrain the position to the interval [0, 16383], and send the position over MIDI, not the delta.  
You can get rid of the reset method and the ResetSender.

rogerarends, 2020-01-08:
I'm impatient. I like building stuff, and troubleshooting hardware (IT)  and programming was always my downfall.., but I'm learning slowly. 

Thanks for your help thus far. Project almost complete.

closed 2020-01-18 by rogerarends