tttapa/Control-Surface #82: Boot Splash Screen

rogerarends, 2020-01-08:
#76  To get the display splash screen working, add a displaySplash method to the MySSD1306DisplayInterface class that draws the bitmap and calls display.

Don't have clue to what I'm doing... after several attempts I finally changed display to displaySplash (like this)

//-------------------------------  Display interface

class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
 public:
  MySSD1306_DisplayInterface(Adafruit_SSD1306 &displaySplash)
    : SSD1306_DisplayInterface(displaySplash) {}

then added the first display with a 5 second delay in setup.

// Setup

void setup() {
  // Correct relative mode for MCU rotary encoders
  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface
    display_A.display();
  delay(5000);
}

It compiled and I uploaded the sketch. on reboot all 4 screens are blank for 5 seconds then the "normal" stuff shows. (vpot, track name, ect.) 

Do I copy the splash. h file to the control surface library and where do I put it if yes. Is there anything I should add or change to DisplayInterfaceSSD1306.hpp or cpp?

tttapa, 2020-01-08:
The first change you made only changed the name of a parameter, it has no effect on the code whatsoever.

You could try something like this:

```cpp
void setup() {
  // Correct relative mode for MCU rotary encoders
  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface
  
  // Display a splash screen for five seconds
  display.clear();
  display.drawXBitmap(0, 0, 
                      XBM::rec_rdy_14B.bits, 
                      XBM::rec_rdy_14B.width,
                      XBM::rec_rdy_14B.height, 
                      WHITE);
  display.display();
  delay(5000);
}
```
Replace the variable name  `display` to `display_A` if necessary. Change `XBM::rec_rdy_14B` to whatever image you want to display at startup.

I have no time to test it right now.

rogerarends, 2020-01-08:
removed  display.clear();
then changed display to the screens and it works. 
thanks

closed 2020-01-08 by rogerarends