tttapa/Control-Surface #83: no such file or directory

benwadub, 2020-01-09:
hi, i m really new to Arduino, I d like to make a simple midi controller via midi din
when I include the library I have a problem with the midi interface, ide say :
SerialMIDI_Interface.hpp: No such file or directory

what am I doing wrong?
I just dowloaded Arduino IDE 
thanks if someone can help me!

tttapa, 2020-01-09:
I fixed the link to the [**installation instructions**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html), and the relative include paths in the documentation should be correct now, thanks for letting me know.

Most of the time, you don't have to include any specific header files, just including `<Control_Surface.h>` will be enough. When using rotary encoders, you have to include `<Encoder.h>` before `<Control_Surface.h>`, to let the Control Surface library know to enable encoder support.

I'm adding a SerialMIDI_Interface example right now, I'll push it to GitHub in a couple of minutes.

Having 16 rotary encoders is going to be difficult on an Uno, since it has only 20 IO pins. You need 32 for 16 encoders. The problems with encoders is that you have to read them at a high rate, preferably with interrupts. There is no support for rotary encoders on multiplexers or other ExtIO chips yet.  
I've ordered some hardware to experiment with it, so it might come in the future, but definitely not before early February.

The [**RotaryEncoder**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/d1f/RotaryEncoder_8ino-example.html) example should get you started for a limited number of encoders.

benwadub, 2020-01-10:
Thank you very much!

benwadub, 2020-01-10:
I already ordered my encoders do you think it is possible that they will be usable?
I thought that I could write that sketch as a learner but I think that I will have to learn a lot of things before my controller will be functional!

tttapa, 2020-01-11:
I've done a quick test with an MCP23017 I2C port expander with interrupt capabilities, but I couldn't get it to work. You have to re-enable interrupts inside of an ISR in order for the I2C driver to work. It should be possible, but I couldn't find any information online. If there's nothing else going on in your loop, you could probably just poll them, though, that's much easier than using interrupts.  
I don't have time to look into it in depth right now.

You can try getting it to work with 9 encoders first, only using the Arduino's pins (D2-D13 and A0-A5).

benwadub, 2020-01-11:
Yes i ll try like this! I saw notes and volts made a sketch for a controller with 16 encoders on a teensy, I should buy a teensy instead and keep the Arduino for another project In the future!
Thanks again for your help!

tttapa, 2020-01-11:
Yes, the encoder example will work fine on a Teensy as well. It has interrupt capability on all pins, and has more pins than the Arduino Uno.

You can test with a few encoders on your Arduino now, and then port it to the Teensy when it arrives.

benwadub, 2020-01-13:
I just received my order, I made a mistake, I ordered encoders clickable, will the encoder li tart will work? Is there a way to use also the click to send another cc?

tttapa, 2020-01-13:
> I just received my order, I made a mistake, I ordered encoders clickable, will the encoder li tart will work? Is there a way to use also the click to send another cc?

I see no reason why they wouldn't work. The "click" is just a momentary push button, you can use it with the [`CCButton`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/daa/classCCButton.html#details) and [`CCButtonLatched`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da6/classCCButtonLatched.html#details) classes.

benwadub, 2020-01-13:
i have now a problem, the ice tell me: Encoder.h: No such file or directory

my sketch is: 

#include <MIDI.h>
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
 
// Select the serial port to use.
auto &serial = Serial;
// Instantiate a Serial MIDI interface at the default MIDI baud rate.
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};
 
// You can also use the following serial MIDI interfaces:
// 
// The Serial port that is connected to your computer over USB:
//
//    USBSerialMIDI_Interface midi = 115200;
//
// A hardware serial port:
// 
//    HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
NoteButton button = {
  5,                       // Push button on pin 5
  {note(C, 4), CHANNEL_1}, // Note C4 on MIDI channel 1
};
 
// Instantiate the LED that will light up when middle C is playing
NoteValueLED led = {
  LED_BUILTIN,             // Pin of built-in LED
  {note(C, 4), CHANNEL_1}, // Note C4 on MIDI channel 1
};

 / Instantiate a CCRotaryEncoder object
CCRotaryEncoder enc = {
  {2, 3},       // pins
  MCU::V_POT_1, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

did I made something wrong?

tttapa, 2020-01-13:
Did you follow the [installation instructions](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html) for the dependencies?

Also, don't include MIDI.h, it's a different library, and it might interfere.

benwadub, 2020-01-13:
i had not installed the PJRC library! I had just upgrade the ice with teensyduino!
now compilation is done!
i ll try to put it on the uno and test my first sketch and encoder!!
thanks you very very very much!!

benwadub, 2020-01-13:
can you please just show me how to write the line of the rotary encoder to send by exemple cc82 on channel 5
instead of the mou v_pot_1 please?

tttapa, 2020-01-13:
Just replace `MCU::V_POT_1` with `{82, CHANNEL_5}`.

benwadub, 2020-01-14:
Nothing is working, I think I have a wiring problem, I ll check if I can find a tutorial that show how I must wire my out midi socket, button and rotary

tttapa, 2020-01-14:
It's recommended to try your code with a debug MIDI interface first: https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/d81/Debug-MIDI-Interface_8ino-example.html

benwadub, 2020-01-14:
I ll try it thanks again!
I see sometimes encoders are wired with resistors and sometimes no, do you think I need some?

tttapa, 2020-01-14:
No, the Encoder library enables the internal pull-up resistors. Just connect the common pin to ground, and the two signal pins to the input pins of the Arduino.

benwadub, 2020-01-14:
I still have nothing working, is there something in this sketch that I had to modify?
do I need to unplug the uno from my Mac and power it separately for this work?
I really don t know where to look now! here the sketch

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
 #include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
// Select the serial port to use.
auto &serial = Serial;
// Instantiate a Serial MIDI interface at the default MIDI baud rate.
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
};

// You can also use the following serial MIDI interfaces:
// 
// The Serial port that is connected to your computer over USB:
//
//    USBSerialMIDI_Interface midi = 115200;
//
// A hardware serial port:
// 
//    HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
NoteButton button = {
  5,                       // Push button on pin 5
  {note(C, 2), CHANNEL_5}, // Note C4 on MIDI channel 1
};

 // Instantiate a CCRotaryEncoder object
CCRotaryEncoder enc = {
  {2, 3},       // pins
  {82, CHANNEL_5}, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2020-01-15:
(Sorry, had an exam today, so I couldn't respond earlier.)

I've just tried your code on an Arduino UNO with a rotary encoder on pins 2&3, and everything seems to work fine.

I suspect that either your encoder is connected incorrectly, or that your MIDI interface is connected incorrectly. 

What MIDI device are you connecting the Arduino to? How did you connect it?

benwadub, 2020-01-15:
no problem! you are very very nice to answer each of my question you can take 24h to respond :-) !
I try to control an electron digitakt with this code, I checked if all parameters were well seated to receive notes and cc with my midi keyboard and all is fine! I think that I make a mistake with the hardware because it s my very first project! 
maybe you can show me the wiring on frizzing or a photo?
on the monitor serial in ide I can see some activity when I turn the encoder...
sorry to be the noob of the year :-)

tttapa, 2020-01-16:
Did you try the debug MIDI interface?

Try this code, upload it, and open the Serial monitor with a baud rate of 115200 baud.

```cpp
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library

USBDebugMIDI_Interface midi = 115200;

using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
NoteButton button = {
  5,                       // Push button on pin 5
  {note(C, 2), CHANNEL_5}, // Note C4 on MIDI channel 1
};

 // Instantiate a CCRotaryEncoder object
CCRotaryEncoder enc = {
  {2, 3},       // pins
  {82, CHANNEL_5}, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

The correct MIDI connections can be found on midi.org

[image]

benwadub, 2020-01-16:
I didn’t try it, because I don t use the midi usb interface but the serial, i really don’t know how to write the debug for the serial midi interface, i m really new to programming I am a cook and not a programmer :-)

tttapa, 2020-01-16:
The serial connection is connected to the serial USB interface on an Arduino UNO. I posted the code you need in my previous reply.

benwadub, 2020-01-16:
Thank you so much!!

benwadub, 2020-01-16:
everything seems to be good
17:51:03.594 -> Control Change  	Channel: 5	Data 1: 0x52	Data 2: 0x01	Cable: 0
17:51:05.306 -> Control Change  	Channel: 5	Data 1: 0x52	Data 2: 0x01	Cable: 0
17:51:07.679 -> Control Change  	Channel: 5	Data 1: 0x52	Data 2: 0x7f	Cable: 0
17:51:08.784 -> Control Change  	Channel: 5	Data 1: 0x52	Data 2: 0x7f	Cable: 0
17:51:12.092 -> Note On         	Channel: 5	Data 1: 0x24	Data 2: 0x7f	Cable: 0
17:51:12.645 -> Note Off        	Channel: 5	Data 1: 0x24	Data 2: 0x7f	Cable: 0

benwadub, 2020-01-16:
all is working now!!! I have now to find how to use my encoder to use it as a send for my delay and reverb, for the moment it pass from zero to 127 an from  127 to 0, I ll try also to add the other encoders!
again a big thank you!
if you come in France come to taste my kitchen for free! I m far better at cooking than programming!!

tttapa, 2020-01-16:
Glad to hear :)

Rotary encoders send relative CC events (i.e. a value of `1` means `+1` and a value of `127` means `-1`). If that's not what your MIDI device is expecting, you could try the [**CCAbsoluteEncoder**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d2f/classCCAbsoluteEncoder.html#a35bfb124c30d2d7dba9894c53f91d13d) class instead.

benwadub, 2020-01-16:
i ll try to write it thanks!

benwadub, 2020-01-16:
ok I did it and it work like a charm!
to add other encoder I just have to copy paste and use all the others pin from 4 to 13?

tttapa, 2020-01-16:
Yes.

benwadub, 2020-01-16:
cool! I just read that analog pin can be used as digital ones do you confirm it? and if it s ok how must I write the code for it please?

benwadub, 2020-01-16:
impossible to find how to write the others encoder I tried this way but expected primary expression before enc

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
 #include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
// Select the serial port to use.
auto &serial = Serial;
// Instantiate a Serial MIDI interface at the default MIDI baud rate.
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
};

// You can also use the following serial MIDI interfaces:
// 
// The Serial port that is connected to your computer over USB:
//
//    USBSerialMIDI_Interface midi = 115200;
//
// A hardware serial port:
// 
//    HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
NoteButton button = {
  5,                       // Push button on pin 5
  {note(C, 2), CHANNEL_1}, // Note C4 on MIDI channel 1
};

 // Instantiate a CCRotaryEncoder object
 CCAbsoluteEncoder enc = {
  {2, 3},       // pins
  {82, CHANNEL_1}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent}
 CCAbsoluteEncoder enc = {
  {4, 5},       // pins
  {82, CHANNEL_2}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent}
};
void setup() {
  RelativeCCSender::setMode ;SIGN_MAGNITUDE ,
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2020-01-16:
The "analog" pins are digital pins in the first place, they just happen to be connected to the ADC as well, as an extra feature.

To use it in your code, just enter the pin number as before. Instead of using `2`, use `A0`, for example.

benwadub, 2020-01-16:
thanks! and to write more than one encoder how must I write it please? I tried to adapt the array I found in the wiki with the cc potentiometer but I don t understand how this work :-(

tttapa, 2020-01-16:
Each variable has to have a different name. You also have a mismatch of braces in your code above.

You have two options: create multiple variables, or create a single array.

```cpp
// Instantiate a CCAbsoluteEncoder object
CCAbsoluteEncoder enc1 = {
  {2, 3},       // pins
  {82, CHANNEL_1}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc2 = {
  {4, 5},       // pins
  {82, CHANNEL_2}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
```
```cpp
// Instantiate an array of CCAbsoluteEncoder objects
CCAbsoluteEncoder encoders[] = {
  {
    {4, 5},       // pins
    {82, CHANNEL_1}, // MIDI address (CC number + optional channel)
    16, // optional multiplier if the control isn't fast enough
    4, //number of detent
  },
  {
    {4, 5},       // pins
    {82, CHANNEL_2}, // MIDI address (CC number + optional channel)
    16, // optional multiplier if the control isn't fast enough
    4, //number of detent
  },
};
```
Note how I just added an extra pair of braces around the initializers for the array, and separated the elements with commas.  
You can clearly see the general pattern:
```
Type var = initializer for one element;

Type array[] = {
  initializer for one element,
  initializer for one element, 
  ...
};
```
Where `Type` = `CCAbsoluteEncoder`, and the initializer for a single element is `{pins, address, multiplier, detents}´.

benwadub, 2020-01-17:
hi everything is working!!! thanks again!

I can t make anything to thank you so I give the code here, maybe you can use it as an example for someone like me who want to make a controller over midi din with rotary encoders

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
 #include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
// Select the serial port to use.
auto &serial = Serial;
// Instantiate a Serial MIDI interface at the default MIDI baud rate.
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
};

// You can also use the following serial MIDI interfaces:
// 
// The Serial port that is connected to your computer over USB:
//
//    USBSerialMIDI_Interface midi = 115200;
//
// A hardware serial port:
// 
//    HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
//NoteButton button = {
  //5,                       // Push button on pin 5
  //{note(C, 2), CHANNEL_5}, // Note C4 on MIDI channel 1
//};

// Instantiate a CCAbsoluteEncoder object
CCAbsoluteEncoder enc1 = {
  {2, 3},       // pins
  {82, CHANNEL_1}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc2 = {
  {4, 5},       // pins
  {82, CHANNEL_2}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc3 = {
  {6, 7},       // pins
  {82, CHANNEL_3}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc4 = {
  {8, 9},       // pins
  {82, CHANNEL_4}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc5 = {
  {10, 11},       // pins
  {82, CHANNEL_5}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc6 = {
  {12, 13},       // pins
  {82, CHANNEL_6}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc7 = {
  {A0, A1},       // pins
  {82, CHANNEL_7}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc8 = {
  {A2, A3},       // pins
  {82, CHANNEL_8}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

benwadub, 2020-01-17:
ok so next step is adding multiplexer but again a problem when compiling :-(
here is the full code
at the end o obtain mux does not name a type!
I picked peace of code in different of your example, I worked on it for 3 hours but can t find what is the problem!

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
 #include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
 #include <Arduino_Helpers.h> // Include the Arduino Helpers library
#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>
 
// Select the serial port to use.
auto &serial = Serial;
// Instantiate a Serial MIDI interface at the default MIDI baud rate.
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
};

// You can also use the following serial MIDI interfaces:
// 
// The Serial port that is connected to your computer over USB:
//
//    USBSerialMIDI_Interface midi = 115200;
//
// A hardware serial port:
// 
//    HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
//NoteButton button = {
  //5,                       // Push button on pin 5
  //{note(C, 2), CHANNEL_5}, // Note C4 on MIDI channel 1
//};

// Instantiate a CCAbsoluteEncoder object
CCAbsoluteEncoder enc1 = {
  {2, 3},       // pins
  {82, CHANNEL_1}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc2 = {
  {4, 5},       // pins
  {82, CHANNEL_2}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc3 = {
  {6, 7},       // pins
  {82, CHANNEL_3}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc4 = {
  {8, 9},       // pins
  {82, CHANNEL_4}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc5 = {
  {10, 11},       // pins
  {82, CHANNEL_5}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc6 = {
  {12, 13},       // pins
  {82, CHANNEL_6}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc7 = {
  {A0, A1},       // pins
  {82, CHANNEL_7}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc8 = {
  {A2, A3},       // pins
  {82, CHANNEL_8}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};

// Instantiate a multiplexer
CD74HC4051 mux = {
  A5,       // Analog input pin
  {2, 3, 4} // Address pins S0, S1, S2
};

//Instantiate multiplexer for 8 switch
CCButton Button[] = {
  {mux.pin(0), {82, CHANNEL_1}},
  {mux.pin(1), {82, CHANNEL_2}},
  {mux.pin(2), {82, CHANNEL_3}},
  {mux.pin(3), {82, CHANNEL_4}},
  {mux.pin(4), {82, CHANNEL_5}},
  {mux.pin(5), {82, CHANNEL_6}},
  {mux.pin(6), {82, CHANNEL_7}},
  {mux.pin(7), {82, CHANNEL_8}},
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
 mux.begin();                  // Initialize multiplexer
 mux.pinMode(0, INPUT_PULLUP); // Set the pin mode (setting it for one pin of
                                // the multiplexers sets it for all of them)
void loop() {
  for (pin_t pin = 0; pin < mux.getLength(); ++pin) {
    Serial.print(mux.digitalRead(pin));
    Serial.print('\t');
  }
  Serial.println();
}
  Control_Surface.loop(); // Update the Control Surface
}

benwadub, 2020-01-17:
i fixed a problem with the } at the end but now in debug I got this

Control Change  	Channel: 1	Data 1: 0x5e	Data 2: 0x7f	Cable: 0
0	1	1	1	0	1	1	1	
0	1	1	1	0	1	1	1	
0	1	1	1	0	1	1	1	
and the page is scrolling again and again... even if I remove all my switch from the uno, I d like these switch to send mute with a cc94

tttapa, 2020-01-18:
Well, that's to be expected, since you are printing the states of the pins of the mux in the loop:
```cpp
  for (pin_t pin = 0; pin < mux.getLength(); ++pin) {
    Serial.print(mux.digitalRead(pin));
    Serial.print('\t');
  }
  Serial.println();
```
I suggest checking out some of the built-in Arduino examples to understand what that code is doing.

Watch out, you're using the same pins for your mux as for the encoder! If you have the encoder connected, this will damage your Arduino.

benwadub, 2020-01-18:
ok so I think I wrote an other error
my button goes on TWO channels and don t go to midi din, I have this on debug
Control Change  	Channel: 4	Data 1: 0x5e	Data 2: 0x7f	Cable: 0
Control Change  	Channel: 2	Data 1: 0x5e	Data 2: 0x7f	Cable: 0
Control Change  	Channel: 2	Data 1: 0x5e	Data 2: 0x00	Cable: 0
Control Change  	Channel: 4	Data 1: 0x5e	Data 2: 0x00	Cable: 0

here is the code
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
 USBDebugMIDI_Interface midi = 115200;
 //#include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
 #include <Arduino_Helpers.h> // Include the Arduino Helpers library
#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>
 
// Select the serial port to use.
auto &serial = Serial;
// Instantiate a Serial MIDI interface at the default MIDI baud rate.
//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
//};

// You can also use the following serial MIDI interfaces:
// 
// The Serial port that is connected to your computer over USB:
//
//    USBSerialMIDI_Interface midi = 115200;
//
// A hardware serial port:
// 
//    HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
//NoteButton button = {
  //5,                       // Push button on pin 5
  //{note(C, 2), CHANNEL_5}, // Note C4 on MIDI channel 1
//};

// Instantiate a CCAbsoluteEncoder object
CCAbsoluteEncoder enc1 = {
  {5, 6},       // pins
  {82, CHANNEL_1}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc2 = {
  {7, 8},       // pins
  {82, CHANNEL_2}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc3 = {
  {9, 10},       // pins
  {82, CHANNEL_3}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc4 = {
  {11, 12},       // pins
  {82, CHANNEL_4}, // MIDI address (CC number + optional channel)
  8, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc5 = {
  {A0, A1},       // pins
  {82, CHANNEL_5}, // MIDI address (CC number + optional channel)
  8, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc6 = {
  {12, 13},       // pins
  {82, CHANNEL_6}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc7 = {
  {A0, A1},       // pins
  {82, CHANNEL_7}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc8 = {
  {A2, A3},       // pins
  {82, CHANNEL_8}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};

// Instantiate a multiplexer
CD74HC4051 mux = {
  A5,       // Analog input pin
  {2, 3, 4} // Address pins S0, S1, S2
};

//Instantiate multiplexer for 8 switch
CCButton button[8] = {
  {mux.pin(0), {83, CHANNEL_1}},
  {mux.pin(1), {94, CHANNEL_2}},
  {mux.pin(2), {94, CHANNEL_3}},
  {mux.pin(3), {94, CHANNEL_4}},
  {mux.pin(4), {94, CHANNEL_5}},
  {mux.pin(5), {94, CHANNEL_6}},
  {mux.pin(6), {94, CHANNEL_7}},
  {mux.pin(7), {94, CHANNEL_8}},
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface

 
 mux.begin();                  // Initialize multiplexer
 mux.pinMode(0, INPUT_PULLUP); // Set the pin mode (setting it for one pin of
}                           // the multiplexers sets it for all of them)
void loop() {
  
    
  
 

  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2020-01-18:
> my button goes on TWO channels

This sounds like one of the address lines of the mux isn't connected correctly.

benwadub, 2020-01-18:
oh damn i ordered a CD74HC4050E and I was looking on notes and volts tutorial for wiring and he use a CD74HC4051 hope I ll find how to wire mine!

tttapa, 2020-01-18:
They are the same chip. The correct wiring diagram with explanation of the pins is the datasheet from the manufacturer.

benwadub, 2020-01-18:
I just checked the connection and I think all is good according to notes and volt tutorial
vcc to 5v
E, vee and god to ground 
s0 to 2
S1 to3
2 to4 
and Z to A5

microswitch to Y0
i plugged only this on the board ans I still have message on channel 1,3,4 and 5 and it doesn't show a cc number but :

21:06:17.473 -> Control Change  	Channel: 5	Data 1: 0x5e	Data 2: 0x7f	Cable: 0
21:06:17.510 -> Control Change  	Channel: 5	Data 1: 0x5e	Data 2: 0x00	Cable: 0
21:06:20.662 -> Control Change  	Channel: 5	Data 1: 0x5e	Data 2: 0x7f	Cable: 0
21:06:20.700 -> Control Change  	Channel: 5	Data 1: 0x5e	Data 2: 0x00	Cable: 0
21:09:55.729 -> Control Change  	Channel: 5	Data 1: 0x5e	Data 2: 0x7f	Cable: 0
21:09:55.762 -> Control Change  	Channel: 5	Data 1: 0x5e	Data 2: 0x00	Cable: 0

code is
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
USBDebugMIDI_Interface midi = 115200;
 //#include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
 #include <Arduino_Helpers.h> // Include the Arduino Helpers library
#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>
 
// Select the serial port to use.
auto &serial = Serial;
// Instantiate a Serial MIDI interface at the default MIDI baud rate.
//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
//};

// You can also use the following serial MIDI interfaces:
// 
// The Serial port that is connected to your computer over USB:
//
//    USBSerialMIDI_Interface midi = 115200;
//
// A hardware serial port:
// 
//    HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
//NoteButton button = {
  //5,                       // Push button on pin 5
  //{note(C, 2), CHANNEL_5}, // Note C4 on MIDI channel 1
//};

// Instantiate a CCAbsoluteEncoder object
CCAbsoluteEncoder enc1 = {
  {5, 6},       // pins
  {82, CHANNEL_1}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc2 = {
  {7, 8},       // pins
  {82, CHANNEL_2}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc3 = {
  {9, 10},       // pins
  {82, CHANNEL_3}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc4 = {
  {11, 12},       // pins
  {82, CHANNEL_4}, // MIDI address (CC number + optional channel)
  8, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc5 = {
  {A0, A1},       // pins
  {82, CHANNEL_5}, // MIDI address (CC number + optional channel)
  8, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc6 = {
  {12, 13},       // pins
  {82, CHANNEL_6}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc7 = {
  {A0, A1},       // pins
  {82, CHANNEL_7}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc8 = {
  {A2, A3},       // pins
  {82, CHANNEL_8}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};

// Instantiate a multiplexer
CD74HC4051 mux = {
  A5,       // Analog input pin
  {2, 3, 4} // Address pins S0, S1, S2
};

//Instantiate multiplexer for 8 switch
CCButton button[] = {
  {mux.pin(0), {82, CHANNEL_1}},
  {mux.pin(1), {94, CHANNEL_2}},
  {mux.pin(2), {94, CHANNEL_3}},
  {mux.pin(3), {94, CHANNEL_4}},
  {mux.pin(4), {94, CHANNEL_5}},
  {mux.pin(5), {94, CHANNEL_6}},
  {mux.pin(6), {94, CHANNEL_7}},
  {mux.pin(7), {94, CHANNEL_8}},
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface

 
 mux.begin();                  // Initialize multiplexer
 mux.pinMode(0, INPUT_PULLUP); // Set the pin mode (setting it for one pin of
}                           // the multiplexers sets it for all of them)
void loop() {
  for (pin_t pin = 0; pin < mux.getLength(); ++pin) {
    
    
  }
  

  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2020-01-18:
> i plugged only this on the board ans I still have message on channel 1,3,4 and 5 and it doesn't show a cc number but :
>
> 21:06:17.473 -> Control Change Channel: 5 Data 1: 0x5e Data 2: 0x7f Cable: 0 [...]

That looks alright to me. What do you mean it doesn't show a CC number? The CC number is Data 1: 0x5e, which is 82 in hexadecimal, i.e. the CC number you specified in your code for the button on pin 0 of the mux.

benwadub, 2020-01-19:
Ok so you think the code is alright like this?

tttapa, 2020-01-19:
Yes, apart from the unnecessary bits of code left behind from the examples you copied it from. You only need the Encoder and Control_Surface includes. You don't need to call mux.begin or mux.pinMode, Control_Surface.begin() will do this for you. The for loop in your loop function is empty, so you might as well just delete it.

benwadub, 2020-01-19:
Thanks I ll clean it, and for my mux sending on different channel do you seen another problem on code if my wiring is good?

tttapa, 2020-01-19:
Ah, I missed the channel number. I'm studying for an exam right now, I'll have a better look at it tomorrow.

benwadub, 2020-01-19:
thanks again! I must admit I don t understand a lot of thing on what I m doing! I m not sure I ll be able to write the final code a day!

tttapa, 2020-01-20:
I cannot reproduce your problem. The channel seems to be correct. Could you post your code and wiring diagram?

benwadub, 2020-01-20:
hello! so here is the code
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
USBDebugMIDI_Interface midi = 115200;
 //#include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
 #include <Arduino_Helpers.h> // Include the Arduino Helpers library
#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>
 
// Select the serial port to use.
auto &serial = Serial;
// Instantiate a Serial MIDI interface at the default MIDI baud rate.
//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
//};

// You can also use the following serial MIDI interfaces:
// 
// The Serial port that is connected to your computer over USB:
//
//    USBSerialMIDI_Interface midi = 115200;
//
// A hardware serial port:
// 
//    HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
using namespace MIDI_Notes;
 
// Instantiate a NoteButton object
//NoteButton button = {
  //5,                       // Push button on pin 5
  //{note(C, 2), CHANNEL_5}, // Note C4 on MIDI channel 1
//};

// Instantiate a CCAbsoluteEncoder object
CCAbsoluteEncoder enc1 = {
  {5, 6},       // pins
  {82, CHANNEL_1}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc2 = {
  {7, 8},       // pins
  {82, CHANNEL_2}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc3 = {
  {9, 10},       // pins
  {82, CHANNEL_3}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc4 = {
  {11, 12},       // pins
  {82, CHANNEL_4}, // MIDI address (CC number + optional channel)
  8, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc5 = {
  {A0, A1},       // pins
  {82, CHANNEL_5}, // MIDI address (CC number + optional channel)
  8, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc6 = {
  {12, 13},       // pins
  {82, CHANNEL_6}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc7 = {
  {A0, A1},       // pins
  {82, CHANNEL_7}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};
// Instantiate another CCAbsoluteEncoder object
CCAbsoluteEncoder enc8 = {
  {A2, A3},       // pins
  {82, CHANNEL_8}, // MIDI address (CC number + optional channel)
  16, // optional multiplier if the control isn't fast enough
  4, //number of detent
};

// Instantiate a multiplexer
CD74HC4051 mux = {
  A5,       // Analog input pin
  {2, 3, 4} // Address pins S0, S1, S2
};

//Instantiate multiplexer for 8 switch
CCButton button[] = {
  {mux.pin(0), {82, CHANNEL_1}},
  {mux.pin(1), {94, CHANNEL_2}},
  {mux.pin(2), {94, CHANNEL_3}},
  {mux.pin(3), {94, CHANNEL_4}},
  {mux.pin(4), {94, CHANNEL_5}},
  {mux.pin(5), {94, CHANNEL_6}},
  {mux.pin(6), {94, CHANNEL_7}},
  {mux.pin(7), {94, CHANNEL_8}},
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface

 
 mux.begin();                  // Initialize multiplexer
 mux.pinMode(0, INPUT_PULLUP); // Set the pin mode (setting it for one pin of
}                           // the multiplexers sets it for all of them)
void loop() {
  
  

  Control_Surface.loop(); // Update the Control Surface
}

and a superb drawing of my wiring :-)
[image]

and still message on channel 1 and 3 with the switch
21:31:05.020 -> Control Change  	Channel: 3	Data 1: 0x5e	Data 2: 0x00	Cable: 0
21:31:06.379 -> Control Change  	Channel: 1	Data 1: 0x52	Data 2: 0x7f	Cable: 0
21:31:06.379 -> Control Change  	Channel: 3	Data 1: 0x5e	Data 2: 0x7f	Cable: 0
21:31:06.519 -> Control Change  	Channel: 1	Data 1: 0x52	Data 2: 0x00	Cable: 0
21:31:06.519 -> Control Change  	Channel: 3	Data 1: 0x5e	Data 2: 0x00	Cable: 0

and curious not the same message when push the button and when it come back

LukaszChrzanowski, 2020-01-20:
You problably know it - only for reminder :) - but for testing purposes (eg. only one encoder connected to A0 on CD74HC4051) You should **ground other "input" pins A1-A7** of multiplexer CD74HC4051  (to any Arduino GND).

It's CMOS - You will get a ton of garbage signals from "air". Try even blowing on multiplexer to get interesting results :)

this should get Your's case solved

benwadub, 2020-01-20:
thanks Lucas, but I just found the problem, one of my cable was a faulty one apparently, I changed one by one each wire and I found that the one on S1 was a problem! with another cable all goes to one channel!!!
but why does my switch send 2 different message? 
22:01:37.936 -> Control Change  	Channel: 1	Data 1: 0x52	Data 2: 0x7f	Cable: 0
22:01:38.082 -> Control Change  	Channel: 1	Data 1: 0x52	Data 2: 0x00	Cable: 0
22:01:38.392 -> Control Change  	Channel: 1	Data 1: 0x52	Data 2: 0x7f	Cable: 0
22:01:38.535 -> Control Change  	Channel: 1	Data 1: 0x52	Data 2: 0x00	Cable: 0
 and is it possible to see the cc number in the monitor instead of the strange numbers :-)?

LukaszChrzanowski, 2020-01-20:
that was my problem too, about week ago.

1) if You ave any usuned pins from CD74HC4051 -> ground it!
2) check wiring - if You use breadboard - it will work like that :/ usually it's faulty connections - try soldier You project in prototype board. This really helps. :)

benwadub, 2020-01-20:
ok so all is working fine now except 2 encoders sending on 2 channels but I think it s again faulty cables I ll see it later! I just add another mix for adding 8 switch that seems to work too
could you just look the way I wrote the second mix and tell me if it s ok? it s working but I don t know if there is a better way to write it!

tttapa, 2020-01-21:
> but why does my switch send 2 different message?

It sends one message when you press it, and another when you release it.

> and is it possible to see the cc number in the monitor instead of the strange numbers :-)?

The strange number is the cc number. It's hexadecimal (base 16), which is much more intuitive when dealing with binary protocols. Most MIDI implementation documents use hexadecimal to specify the CC numbers etc.  
0x52 = 5×16+2 = 80 + 2 = 82.

If you really wanted to, you could change it to decimal by editing the [source code](https://github.com/tttapa/Control-Surface/blob/fa2e4a49da50ffb0e7c659deff0420e2cfd288ae/src/MIDI_Interfaces/DebugMIDI_Interface.cpp#L83).

> I just add another mix for adding 8 switch that seems to work too
could you just look the way I wrote the second mix and tell me if it s ok? it s working but I don t know if there is a better way to write it!

```cpp
CD74HC4051 muxes[] = {
  { // mux 1
    A4, // Analog input pin
    {2, 3, 4} // Address pins S0, S1, S2
  },
  { // mux 2
    A5, // Analog input pin
    {2, 3, 4} // Address pins S0, S1, S2
  },
};
```

benwadub, 2020-01-21:
Thanks! I was nearly good! I begin to understand lot of things with you! Really thank you for support!

benwadub, 2020-01-21:
I m going to order what I need to solder one a board, I have a question that come to mind, when you solder you put the encoder in the hole of the board and solder the cable on the other side of the board directly on the pin of the encoder? Same thing for the switch?

tttapa, 2020-01-21:
The encoder goes on the top side of the PCB (the side without copper), and then you solder it in place at the bottom. Then you insert the wire in the top side as well, through the hole next to the pin of the encoder you want to connect it to. Then you just solder it to the bottom as well. If you're using strip board, you might have to cut the trace. If you're using perfboard with single-hole copper, you can bend the wire towards the pin of the encoder before soldering, so you can easily connect it to the pin using some solder.

There are other ways, of course, but I find this one to be the easiest and cleanest.

[30 more replies]