tttapa/Control-Surface #84: Need help with bankable buttons

samlochner, 2020-01-10:
Hi, if I have a series of note buttons like in the code below, is there a better way to declare them (fewer lines of code)? The mapping from arduino pin to midi note doesn't necessarily have a particular pattern, so it has to be specified for each. I'll also have to invert them all. Suggestions?

Thanks,
Sam

```
Bankable::NoteButton button34 = {transposer,34,{69, CHANNEL_1},127};
Bankable::NoteButton button33 = {transposer,33,{58, CHANNEL_1},127};
Bankable::NoteButton button32 = {transposer,32,{70, CHANNEL_1},127};
Bankable::NoteButton button31 = {transposer,31,{59, CHANNEL_1},127};
Bankable::NoteButton button30 = {transposer,30,{71, CHANNEL_1},127};
Bankable::NoteButton button29 = {transposer,29,{60, CHANNEL_1},127};
Bankable::NoteButton button28 = {transposer,28,{72, CHANNEL_1},127};
```

tttapa, 2020-01-10:
You can create an array of `NoteButton`s:

```cpp
Bankable::NoteButton buttons[] = {
  {transposer, 34, {69, CHANNEL_1}, 127},
  {transposer, 33, {58, CHANNEL_1}, 127},
  {transposer, 32, {70, CHANNEL_1}, 127},
  {transposer, 31, {59, CHANNEL_1}, 127},
  {transposer, 30, {71, CHANNEL_1}, 127},
  {transposer, 29, {60, CHANNEL_1}, 127},
  {transposer, 28, {72, CHANNEL_1}, 127},
};
```

To invert them all, just loop over the array, and invert every one of them:
```cpp
void setup() {
  for (auto &button : buttons)
    button.invert();
  Control_Surface.begin();
}
```

Alternatively, you can use the [NoteButtons](https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/dbd/classNoteButtons.html#details) class. This only works if the addresses of all buttons are a fixed interval apart.

```cpp
Bankable::NoteButtons<7> notebuttons = {
  transposer, 
  {28, 29, 30, 31, 32, 33, 34}, // pins
  {60, CHANNEL_1},              // base address
  {1, 0},                       // address increment
};
```

Pin 28 will be address `{60, CHANNEL_1}`, pin 29 will be address `{60, 1} + {1, 0} = {61, CHANNEL_1}`, pin 30 will be `{61, 1} + {1, 0} = {62, CHANNEL_1}`, etc.

In that case, you can just use `notebuttons.invert()` in the setup to invert all of the buttons.

samlochner, 2020-01-10:
Understood, thanks!

closed 2020-01-10 by samlochner