tttapa/Control-Surface #85: Set manufacturer, model and midi device name

LukaszChrzanowski, 2020-01-10:
Hello,

I'm creating controller for myself currently for DAW Propellerhead Reason, and for this I need to create special mapping for their protocol "Remote" for communication with my newly created Control Surface (I've created my control surface to control mixer effects, and for each time in every project I've to map knobs from my controller to wirtual knobs in mixer - with that mapping it will be assigned automatically).

To have this mapping working - I've to set Manufacturer, Model and MIDI Device name. 

I've tried to find this in Your's library, but I cannot find it - can You give me a hint? :)

tttapa, 2020-01-11:
The low-level USB descriptors are handled by either the MIDIUSB library or by the Teensy Core, depending on which board you're using.

The Teensy names are defined here: https://github.com/tttapa/Control-Surface/blob/bc9b22273e2c6ff4fc6c4640e8029f1c4105ebb2/src/MIDI_Interfaces/Teensy-USB-MIDI-name.c  
You can find more information on the PJRC forums.

MIDIUSB is harder to change, you'll have to edit the library. I don't have any experience with this, so I can't really help you.

2kohm, 2020-01-12:
in your main sketch folder, create a file named **usb_desc.c**

```
#include <usb_names.h>

#define MANUFACTURER_NAME {'M', 'Y'}
#define MANUFACTURER_NAME_LEN 2
#define PRODUCT_NAME {'D', 'E', 'V', 'I', 'C', 'E'}
#define PRODUCT_NAME_LEN 6
struct usb_string_descriptor_struct usb_string_manufacturer_name = {
    2 + MANUFACTURER_NAME_LEN * 2,
    3,
    MANUFACTURER_NAME};

struct usb_string_descriptor_struct usb_string_product_name = {
    2 + PRODUCT_NAME_LEN * 2,
    3,
    PRODUCT_NAME};
```

LukaszChrzanowski, 2020-01-12:
I use Arduino Micro, so Teensy is out of scope for me.

I've found something like this: http://liveelectronics.musinou.net/MIDIdeviceName.php but not tried yet

2kohm - I suppose this will be then automatically added to my project? I will try tomorrow and give feedback how it goes. :)

tttapa, 2020-01-12:
2kohm's solution only works for Teensies. The liveelectronics link should work for Arduino Micro.

mostr, 2020-04-06:
@LukaszChrzanowski did you make it work with the instructions from liveelectronics? I changed the name but can't see proper board name. I'm on Arduino Leonardo

pukulsesuatu, 2022-07-29:
@LukaszChrzanowski  I Following this instruction and work for Me, I am using Arduino micro [https://www.youtube.com/watch?v=6cskac0Dzjo](url)

starwalker00, 2023-08-23:
> @LukaszChrzanowski I Following this instruction and work for Me, I am using Arduino micro [https://www.youtube.com/watch?v=6cskac0Dzjo](url)

Thank you, this worked great for me.

---

Basic instructions are :
---
1. locate __boards.txt__ file, on windows it is somewhere in :

    - C:\Program Files (x86)\Arduino\hardware\arduino\avr\

    - C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

2. change desired name, for instance for Leonardo in __boards.txt__ :

    - from `leonardo.build.usb_product="Arduino Leonardo"`
    - to `leonardo.build.usb_product="My Custom Device"`
3. Restart Arduino IDE
4. Compile and upload code of your application.