tttapa/Control-Surface #86: Joystick Class to control Surround Panning

kevinagnes, 2020-01-12:
Hello tttapa!
Firstly, thanks for the amazing library and the dedication you spend with this gem.

I was able to follow on your suggestion from Issue #76 of Borrowed Rotary Encoder
and managed to create a Bankable version of the shared-encoder branch.

Now, I would like to implement a Joystick class that can handle Logic Pro's Surround Panner. I was able to build and test the code bellow without your library, however, I have no idea where to start in order to build the class using your library.

......................(0,1)....................
.................................................		  	
....(-30,1).......................(+30,1)
.................................................
.................................................						    
..(-90,1).........(0,0).........(+90,1)
..................................................
..................................................
....(-120,1)..................(+120,1)...				
..................................................
..................(+-180,1)..................

value_converted_mapped
UP(512,1023)(0,1)(0,1)
LEFT(0,512)(-90,1)(32,1)
RIGHT(1023,512)(+90,1)(96,1)
DOWN(512,0)+/-180,1)0/127,1)
CENTER(512,512)(0,0)0,0)

My previous code was this one: 

```
          int extra = 512;
	  x = analogRead(A0); // Joy X
	  y = analogRead(A1); // Joy Y
	  x = x - extra; // Position 0 (is actually 512)
	  y = y - extra; // Position 0 (is actually 512)
	  
	  
	  // Math to convert X,Y position to Angle and Distance from center position
	  
	  angleRAD = atan2 (y,x);
      angle = (180.0/M_PI)*angleRAD;
	  diversity = ((sqrt(pow(x,2) + pow(y,2))/extra))*1000; // times 1000 for mapping
	  
	  
	  // Mapping the values back to MIDI standard
	  
      AngleMapped = map(angle,-180,180,0,127);
      DiversityMapped = map(diversity,0,1000,0,127);
	  
	  
	  // Creating the CC MIDI messages (Both CC must be sent at the same time)
	  
	  controlChange(channelJoystick, 118, AngleMapped);
	  controlChange(channelJoystick, 119, DiversityMapped);
	  MidiUSB.flush();
```

tttapa, 2020-01-12:
You could use something like this:

```cpp
#include <Control_Surface.h>

class PanJoystick : public MIDIOutputElement {
  public:
    PanJoystick(PinList<2> pins, MIDICNChannel channel)
      : joyX(pins[0]), joyY(pins[1]), channel(channel) {}

    void begin() override {
      // Initialize if necessary
    }

    void update() override {
      bool changed = false;
      changed |= joyX.update();
      changed |= joyY.update();

      if (!changed)
        return;

      constexpr int bias = 512;
      int x = joyX.getValue() - bias; // Position 0 (is actually 512)
      int y = joyY.getValue() - bias; // Position 0 (is actually 512)

      // Math to convert X,Y position to Angle and Distance from the center position
      constexpr float PI_f = M_PI;
      float angleRad          = atan2(y, x);                  // [-π, +π]
      angleRad               += PI_f;                         // [0, +2π]
      float diversity         = hypot(x, y) / bias / sqrt(2); // [0, 1]
      //                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      // Check this formula, it might not do what you expect!
      
      // Map to MIDI CC range
      uint8_t angleMapped     = round(angleRad * 127 / (2 * PI_f)); // [0, 127]
      uint8_t diversityMapped = round(diversity * 127);             // [0, 127]

      // Sending the CC MIDI messages (Both CC must be sent at the same time)
      Control_Surface.MIDI().sendCC({118, channel}, angleMapped);
      Control_Surface.MIDI().sendCC({119, channel}, diversityMapped);
    }

  private:
    FilteredAnalog<> joyX, joyY;
    MIDICNChannel channel;
};

// ::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //

USBMIDI_Interface midi;

PanJoystick joystick = {{A0, A1}, CHANNEL_1};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

Note that there is a mathematical error in your diversity calculation: when both x and y are 512 (their maximum value), the result of `((sqrt(pow(512,2) + pow(512,2))/512))*1000` will be 1000 √2 ≃ 1414, and `DiversityMapped` will be larger than 127.  
This could be useful: https://stackoverflow.com/questions/13211595/how-can-i-convert-coordinates-on-a-circle-to-coordinates-on-a-square

kevinagnes, 2020-01-13:
tttapa, thanks for the quick response. 

The class is working, but you really caught a flaw in the math.
I am trying to solve the diversity problem, but still with no luck.
I implemented the square to circle position formula:

```
float u = x * sqrt(1.0 - y*y/2.0);
float v = y * sqrt(1.0 - x*x/2.0);
float diversity = hypot(u, v) / bias / sqrt(2); // [0, 1]
```

but somehow the mapping is still not the right one.

Besides the math problem, how can I activate the midi send with a pushbutton?
I would like to send CC messages  only when I am pressing a button, otherwise 
when I release the Joystick it would return to the centre position, therefore, resetting the
panning previously made.

I tried working with .disable() .enable() in the past, but with no great results.

tttapa, 2020-01-13:
You have to normalize x and y in [-1, +1] first, before computing u and v. They have to be floats. `y*y` overflows if `y` is an integer.

You can just add another if statement to the update method that checks if a button is pressed.

```cpp
class PanJoystick : public MIDIOutputElement {
  public:
    PanJoystick(PinList<2> pins, pin_t buttonPin, MIDICNChannel channel)
      : joyX(pins[0]), joyY(pins[1]), buttonPin(buttonPin), channel(channel) {}

    void begin() override {
        pinMode(buttonPin, INPUT_PULLUP);
    }

    void update() override {
        if (digitalRead(buttonPin) == HIGH) // if not pressed
            return;
        // ...
    }

  private:
    FilteredAnalog<> joyX, joyY;
    pin_t buttonPin;
    MIDICNChannel channel;
};
```

kevinagnes, 2020-01-13:
tttapa, although there was a mathematical error, this code actually works for a joystick module:

`float diversity         = hypot(x, y)/bias; `

I think this happens because the  X Y movement was already restricted to the circular shape.
So the maximum values of [0,0][1024,1024][1024,0][0,1024] are physically not achievable.

in

```
constexpr int bias = 512;
      int x = joyX.getValue() - bias; // Position 0 (is actually 512)
      int y = joyY.getValue() - bias; // Position 0 (is actually 512)

      // Math to convert X,Y position to Angle and Distance from the center position
      constexpr float PI_f    = M_PI;
      float angleRad          = atan2(y, x);                  // [-π, +π]
      angleRad               += PI_f;                         // [0, +2π]
   
      float diversity         = hypot(x, y)/bias;             // [0, 1]

      // Map to MIDI CC range
      uint8_t angleMapped     = round(angleRad * 127 / (2 * PI_f)); // [0, 127]
      uint8_t diversityMapped = round(diversity * 127);             // [0, 127]

      // Sending the CC MIDI messages (Both CC must be sent at the same time)
      Control_Surface.MIDI().sendCC({118, channel}, angleMapped);
      Control_Surface.MIDI().sendCC({119, channel}, diversityMapped);
```

Thanks for everything. :)

tttapa, 2020-01-13:
Good to hear!  
You might want to add a check to make sure that diversityMapped doesn't go above 127, otherwise it will probably wrap around to zero. If all you have is a physical constraint on the joystick, you might occasionally get a value of 128, for example, especially in the presence of floating point round-off errors etc.

```cpp
if (diversityMapped > 127)
  diversityMapped = 127;
```

closed 2020-01-13 by tttapa

kevinagnes, 2020-04-15:
Pieter, thank you again for this fantastic library.
I want to show you what I was able to achieve using your library so far:
https://youtu.be/OTbKde8f7XE

The small window on the top left of the video is the ILI Display. Although I am very new to programming, I found your documentation very easy to follow, and I was able to write the display code. :)

I hope this can bring you joy and motivation to keep the exemplary work you have done so far!

Cheers, and see you soon on future updates! :)

tttapa, 2020-04-15:
That's great!

The display is very nice, especially with the colors in EQ mode etc. I should definitely get one myself.

> I hope this can bring you joy and motivation

It sure does! Thanks a lot for sharing :)