tttapa/Control-Surface #87: Button doubleClick and longPress events

Rolbista, 2020-01-13:
Hello

I am experimenting with changing the controller's mode of operation (midi, select pot, select CC) via button presses and saving the values to EEPROM using a double click or a long press (2s). I tried using your `Button` class and while it worked for switching modes, I couldn't differentiate between a `Button::Falling` and `Button::Pressed` events for long press. Can you advise on how to approach this or maybe it is better to use another dedicated button library for what I want to do?

here's the link to some WIP:
https://github.com/Rolbista/MidiBoi/blob/master/switcher_one_pot.ino (master, single button press)
https://github.com/Rolbista/MidiBoi/pull/6/files (external library, single click to change mode, double to save to EEPROM)

tttapa, 2020-01-13:
For long presses, you can use this code as an example:
https://github.com/tttapa/Control-Surface/blob/49a942e2aa17092d653dcf5456b0e3909ecd3b82/src/AH/Hardware/IncrementButton.cpp#L7-L35

Basically, you check how long the button has been pressed for, and if it's longer than some threshold, you enter `LongPress` mode. You exit `LongPress` mode when the button is released.  
The code above also has a "repeat" feature (lines 23-26): if you keep on pressing the button, it also fires an event every `LONG_PRESS_REPEAT_DELAY` milliseconds, like a key on a computer keyboard.

Double presses are similar, just save the time it was pressed, and act if the button falls again within a specified window after the first time it was pressed.

You can use the [`previousBounceTime`](https://tttapa.github.io/Control-Surface-doc/Doxygen/de/d0d/classAH_1_1Button.html#a27d7b259b85b72a4e68fb86036e65b9c) and [`stableTime`](https://tttapa.github.io/Control-Surface-doc/Doxygen/de/d0d/classAH_1_1Button.html#a1cbdae7abfef90d54889e98b100f4d53) methods of the Button class. The former returns the `millis` timestamp of the last change in button state, the latter gives the number of milliseconds that have elapsed since the last state change.

santta, 2021-06-24:
hello, share your finished sketch with longPress events, for example.

closed 2022-02-10 by Rolbista