tttapa/Control-Surface #88: Any plans to implement a class to LCDs? Such as the common ILI9341?

kevinagnes, 2020-01-15:
hello tttapa!
Do you plan to add support for bigger screens?
I am trying to adapt the MCU OLED SSD1306 example to an ILI9341 LCD.

My idea is to display in to display on the top the surround master meter (VUs) and time display, and at the bottom the rough EQ Graph shape and surround panning.

By the way, this display also has touch capability, maybe it can even be used as midi input.

Cheers, Kevin.

tttapa, 2020-01-15:
I've only every used small OLED displays, and I don't have any ILI9341 displays, so I have no plans to implement it in the near future.

The main problem right now is that all displays are re-drawn completely on each `loop` iteration. This is fine for small OLEDs, but might be a problem for larger displays. This could be circumvented with a "dirty" flag for each display element, but I haven't really thought out the structure/architecture yet.

I think the easiest thing to do for now is to manage the display yourself in the `loop` function.

kevinagnes, 2020-02-07:
Sorry for the very long delay to reply.

I am working on that right now. The good thing is that it only updates the display when it needs to. This way, I was able to draw basic shapes and react to different encoder positions while also inputting text information.

But I am still confused about how to "get" the DAW values.
For example: 
[1] get value from the volume of the selected track.
[2] get value when recording.
[3] get X value from Y plug-in.

Cheers,
Kevin.

LukaszChrzanowski, 2020-02-07:
... and there goes "hell" for get this information. 
_In simpliest way - DAW must send this information via MIDI to hardware._
For ex. Propellerheads Reason there is "Remote" - so You can create Your's own codec for Your's hardware... and there You create mapping that literally takes values from controls, convert them to  MIDI messages and sends it to Your hardware.

Basically each DAW work the same way - althrough I'm not sure is others also uses some kind of "codec language" for custom hardware. (and really I'm eager to know this answer, someone? :))

AFAIK there is no other way to get values from DAW.

tttapa, 2020-02-07:
Lukasz is right, you cannot request data from the software.  
The way you get data from the software is by adding listeners for certain MIDI events in your code. These listeners are in the [MIDI Input Elements](https://tttapa.github.io/Control-Surface-doc/Doxygen/df/d8b/group__MIDIInputElements.html) module.

benwadub, 2020-02-07:
Hi Pieter do you think listener can work with my electron digitakt? It could avoid jumping values with faders?

tttapa, 2020-02-08:
You can only listen to MIDI events that the device/software sends, nothing else.  
If it doesn't send the messages you're interested in, you're out of luck, sadly.

You can hook your device up to a MIDI monitor to see what kinds of messages it sends.

LukaszChrzanowski, 2020-02-08:
benwadub - what DAW You are working with?

benwadub, 2020-02-08:
> benwadub - what DAW You are working with?

I tried with Logic Pro x

LukaszChrzanowski, 2020-02-08:
As I see, there is this a easy way to send sysex messages:
http://logicpro.skydocu.com/en/work-in-the-environment/environment-objects-reference/fader-objects/work-with-sysex-messages/
I supose if You dig more - You will find much more... like how to send values to Your's project.

best luck :)

benwadub, 2020-02-09:
It s for my elektron digitakt that I need that!

LukaszChrzanowski, 2020-02-09:
sorry, but I din't get it :) this library isn't for elektron digitakt, but for creating Your own hardware...

benwadub, 2020-02-09:
i m building an hardware controller for my digitakt!

LukaszChrzanowski, 2020-02-09:
ah.... so You need to use Uncle Google here :) I wish, but I could not help You - maybe tttapa

benwadub, 2020-02-09:
I looked in midi monitor my elektron seems to send data only when I turn it s knob so I ll have jumping values every time I change patterns :-(

LukaszChrzanowski, 2020-02-09:
not quite, becouse You can "hook up" on some defined midi messages.
This messages have their channel, CC and port (probably, becouse there is no notes as I saw in this device).

benwadub, 2020-02-09:
Sorry I didn’t understood your answer :-(

tttapa, 2020-02-10:
> I looked in midi monitor my elektron seems to send data only when I turn it s knob so I ll have jumping values every time I change patterns :-(

Does it send anything when you change patterns? Do the messages sent by the knob change when you change patterns?

kevinagnes, 2020-02-10:
Pieter, I can see (using MIDI Monitor app) that I get messages from my DAW everytime I rotate the encoders.

[image]

 In the picture you can see when I rotate the encoder mapped to a EQ freq parameter, I get the same CC number with values mapped from 0 - 127 (acording to the EQ parameter min and max).

My question is how can I listen to this information and convert it to a display?

LukaszChrzanowski, 2020-02-10:
Here You are: 
https://github.com/tttapa/Control-Surface/issues/99

tttapa, 2020-02-10:
You could try something like this:

```cpp
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Listener for CC #8
CCValue cc8 = {
  {8, CHANNEL_1},
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
  Serial.begin(115200);
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
  Serial.println(cc8.getValue()); // Print the value of CC #8
}
```