tttapa/Control-Surface #89: array out of bound for CD74HC4051 can brick Arduino

LukaszChrzanowski, 2020-01-20:
Hello,

I've found that if You create more potentiometers than 8 for CD74HC4051 if will brick Arduino (using reset pin and while that loading any working program from IDE eg. blinking while it is spotted after reset).

Code that creates this:
```
CD74HC4051 mux = {
  A0,       // Analog input pin
  {3, 4, 5} // Address pins S0, S1, S2
};
 
CCPotentiometer volumePotentiometers[] = {
  { mux.pin(0), { MIDI_CC::Channel_Volume, CHANNEL_1 } },
  { mux.pin(1), { MIDI_CC::Channel_Volume, CHANNEL_2 } },
  { mux.pin(2), { MIDI_CC::Channel_Volume, CHANNEL_3 } },
  { mux.pin(3), { MIDI_CC::Channel_Volume, CHANNEL_4 } },
  { mux.pin(4), { MIDI_CC::Channel_Volume, CHANNEL_5 } },
  { mux.pin(5), { MIDI_CC::Channel_Volume, CHANNEL_6 } },
  { mux.pin(6), { MIDI_CC::Channel_Volume, CHANNEL_7 } },
  { mux.pin(7), { MIDI_CC::Channel_Volume, CHANNEL_8 } },
  { mux.pin(8), { MIDI_CC::Channel_Volume, CHANNEL_9 } },
};
```

Could You add warning in text in example that it should have no more than it should? :) I've spend one day to bring back my Arduino from dead :)

(Arduino Pro Mini Clone, spotted as Arduino Leonardo ETH)

tttapa, 2020-01-21:
The problem is that `mux.pin(8)` is called before the program starts. [The `pin` function performs a bounds check](https://github.com/tttapa/Control-Surface/blob/fa2e4a49da50ffb0e7c659deff0420e2cfd288ae/src/AH/Hardware/ExtendedInputOutput/ExtendedIOElement.cpp#L29), but there's no way to do error handling if the platform isn't initialized yet (impossible to print to serial, for example, because the serial port is initialized after the error occurs).

I have exams right now, but I'll update the documentation when I find the time.

LukaszChrzanowski, 2020-01-22:
Thank You - this will probably help others.
I wish You luck! :)

4dvn, 2020-01-23:
> Thank You - this will probably help others.
> I wish You luck! :)

Can you upload your potentiometer wise? I tested on my arduino pro micro no poblem.

LukaszChrzanowski, 2020-01-24:
for my two arduino pro micro it was instant death