tttapa/Control-Surface #90: 4 toggle chanel changer buttons with led

DjGuuOmsk, 2020-01-21:
Hi. I used to use your old library and it was very clear and with many different examples. But making a new controller, I decided to try your new library, because it has MIDI_input in it. But I just can’t figure out how to make four buttons switch MIDI channels. In old library it was 
```
Digital button1[] = {
  {9, 0x01, 1, velocity},   //bankselector 1
  {13, 0x02, 1, velocity},  //bankselector 2
  {7, 0x03, 1, velocity},   //bankselector 3
  {11, 0x04, 1, velocity},  //bankselector 4 
};

Bank bank(1);
BankSelector bankselector1(bank, {9,13,7,11}, {8,12,6,10} ); // 8,12,6,10 is led pins

void setup() 
{
   bank.add(button1, Bank::CHANGE_CHANNEL);
 }

void loop()
{
    MIDI_Controller.refresh();
}

```
but how to make it in this library? This is the new code. What do I need to add to switch the midi channel ?
```
#include <Encoder.h>
#include <Control_Surface.h> 

USBDebugMIDI_Interface midi = {115200};
const int speedMultiplier = 1;

/*
chanel1 button 9, led pin 8
chanel2 button13, led pin12
chanel3 button7, led pin6
chanel4 button11, led pin10
*/

Bankable::CCRotaryEncoder enc[] = {
  {{bank, BankType::CHANGE_CHANNEL}, {36, 37}, 0x0A, speedMultiplier, 4,},
  {{bank, BankType::CHANGE_CHANNEL}, {46, 47}, 0x07, speedMultiplier, 4,},
  {{bank, BankType::CHANGE_CHANNEL}, {48, 49}, 0x06, speedMultiplier, 4,},
  {{bank, BankType::CHANGE_CHANNEL}, {50, 51}, 0x09, speedMultiplier, 4,},
  {{bank, BankType::CHANGE_CHANNEL}, {53, 52}, 0x08, speedMultiplier, 4,},
};

Bankable::NoteButton buttonencoders[] = {
  {{bank, BankType::CHANGE_CHANNEL}, 42, {0x08, CHANNEL_1} }, // 
  {{bank, BankType::CHANGE_CHANNEL}, 43, {0x09, CHANNEL_1} }, //
  {{bank, BankType::CHANGE_CHANNEL}, 44, {0x06, CHANNEL_1} }, // 
  {{bank, BankType::CHANGE_CHANNEL}, 45, {0x07, CHANNEL_1} }, //
};

CD74HC4067 mux2 = {A6,{A0,A1,A2,A3} };
Bankable::NoteButton button1[] = {
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(0), { 0x0E, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(1), { 0x0F, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(2), { 0x10, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(3), { 0x11, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(4), { 0x12, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(5), { 0x13, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(6), { 0x14, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(7), { 0x15, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(8), { 0x16, CHANNEL_1 } }, 
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(9), { 0x17, CHANNEL_1 } }, 
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(10), { 0x20, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(11), { 0x21, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(12), { 0x22, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(13), { 0x23, CHANNEL_1 } },
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(14), { 0x19, CHANNEL_1 } }, 
  {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(15), { 0x18, CHANNEL_1 } },
};

 void setup() {
  Control_Surface.begin(); // Initialize Control Surface
  }
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
  }
```
I hope you, or someone who is smarter than me, will help me with this task. Thanks in advance

tttapa, 2020-01-21:
You should be able to use something like this:

```cpp
// 4 banks with 1 track per bank
Bank<4> bank(1);

// Selector that manages the bank setting
ManyButtonsSelectorLEDs<4> bankselector = {
  bank,
  {9, 13, 7, 11}, // button pins
  {8, 12, 6, 10}, // led pins
};
```

Relevant documentation: [**Selectors**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html)

DjGuuOmsk, 2020-01-21:
> Relevant documentation: [**Selectors**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html)

Works perfectly!!! Biggest respect! The controller photo [https://www.instagram.com/p/B7aRA-OBCmJ/](https://www.instagram.com/p/B7aRA-OBCmJ/)

tttapa, 2020-01-21:
Looks good!

closed 2020-01-21 by DjGuuOmsk

benwadub, 2020-01-21:
Nice job!! Have you picture of the inside? I m really new to Arduino and I am on the way to solder for the first time, if I could seen how you wire things it would be nice!

4dvn, 2020-01-23:
> Nice job!! Have you picture of the inside? I m really new to Arduino and I am on the way to solder for the first time, if I could seen how you wire things it would be nice!

Check out the wiki of old midi controller library