tttapa/Control-Surface #91: Is it possible to add arduino joystick sensor module in Control_Surface?

panakronic, 2020-01-22:
Hi there,

Is it possible to add arduino joystick sensor module in Control_Surface?

tttapa, 2020-01-22:
Hi, a joystick just consists of two potentiometers (one for horizontal, one for vertical), so you can use two `CCPotentiometer`s, for example.

This issue might be useful as well: https://github.com/tttapa/Control-Surface/issues/86

panakronic, 2020-08-31:
Hi there,

I am working with a 4x4 matrix pad and I wish to get Bank Selector to work
with the Matrix pad.

I followed example from Bank-Button-Matrix..
https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d41/Bank-Button-Matrix_8ino-example.html

I got the keypad working with all 16 buttons sending unique note.  However,
when I pressed the select button from A0 or A1, the each button from the
matrix pad still sends same note.

This is what I have at the moment, which is very simple. Would you be able
to help me shine some light on this mystery? Thank you. Cheers

/**
 * @example Ex.11.Button-Matrix.ino
 *
 * This is an example of the ButtonMatrix class of the Control_Surface
library.
 *
 * Connect a 4 × 3 matrix of buttons with the rows to pins 2, 3, 4 and 5,
 * and the columns to pins 6, 7 and 8.
 * Pull-up resistors are not necessary, because the internal ones will be
used.
 *
 * If you want to be able to press multiple buttons at once, add a diode
 * in series with each button, as shown in the following schematic:
 *
 * @image html Button-matrix.svg
 *
 * The note numbers are specified in the `addresses` array.
 * Map accordingly in your DAW or DJ software.
 *
 * Written by tttapa, 24/09/2017
 * https://github.com/tttapa/Control-Surface
*/

#include <Control_Surface.h>

USBMIDI_Interface midi;
Bank<4> bank = {16}; // 4 banks, 6 addresse per banks
IncrementDecrementSelector<4> selector = {bank, {A0, A1}, Wrap::Wrap};

AddressMatrix<4, 4> notes = {{
  {1, 2, 3, 4},
  {5, 6, 7, 8},
  {9, 10, 11, 12},
  {13, 14, 15, 16}
}};

Bankable::NoteButtonMatrix<4, 4> buttons = {
  bank,
  {14, 15, 16, 17},    // row pins (outputs, driven low-Z low !)
  {18, 19, 20, 21}, // column pins (inputs, hi-Z)
 notes,     // address matrix
  CHANNEL_1, //
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}

tttapa, 2020-09-05:
What board are you using? 

You could use the [Button example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/da9/2_8Button_8ino-example.html) to check that the buttons are wired correctly. (Change the pin number to A0/A1 in that example.) The buttons should be wired between the Arduino input pin A0 or A1 and ground.