tttapa/Control-Surface #94: How to use "potentiometer" and "button" with multiplexers?

4dvn, 2020-01-23:
Hello petter, can you give me the example code of 16 pots and 16 buttons  2 multiplexer 74hc4067? Thank you!

tttapa, 2020-01-25:
What have you tried? It isn't any different from the "First Output" example. You just declare two muxes and use their pins to initialize the potentiometers and buttons. I don't have much time to write code right now.

Red1986z31, 2020-02-20:
AnalogMultiplex multiplexer2(25, { 2, 3, 4 } );
AnalogMultiplex multiplexer3(26, { 2, 3, 4 } );
AnalogMultiplex multiplexer4(27, { 2, 3, 4 } );
AnalogMultiplex multiplexer5(28, { 2, 3, 4 } );
AnalogMultiplex multiplexer6(A18, { 2, 3, 4 } );
AnalogMultiplex multiplexer7(A19, { 2, 3, 4 } );

Analog potentiometers3[] = {
{multiplexer8.pin(0), MIDI_CC::Pan, 1}, //PHONE VOLUME
{multiplexer8.pin(1), MIDI_CC::Pan, 2}, //PHONE MIX
{multiplexer8.pin(2), MIDI_CC::Pan, 3}, // RIGHT HIGH
{multiplexer8.pin(3), MIDI_CC::Pan, 4}, //RIGHT LOW
{multiplexer8.pin(4), MIDI_CC::Pan, 5}, //RIGHT GAIN
{multiplexer8.pin(5), MIDI_CC::Pan, 6}, //RIGHT MID
{multiplexer8.pin(6), MIDI_CC::Pan, 7}, // FX 1 A GAIN
{multiplexer8.pin(7), MIDI_CC::Pan, 8}, //RIGHT FADER

};

Digital Buttonslarge[] = {
{multiplexer1.pin(0), C2, 1, velocity}, //L MODE SELECT
{multiplexer1.pin(1), D2, 1, velocity}, //BUTTON 34
{multiplexer1.pin(2), E2, 1, velocity}, //FX 1 B SELECT
//{multiplexer1.pin(3), f2, 1, velocity}, //BUTTON 37
//{multiplexer1.pin(4), G2, 1, velocity}, // BUTTON 39
//{multiplexer1.pin(5), a2, 1, velocity}, OPEN
//{multiplexer1.pin(6), B2, 1, velocity}, // BUTTON 35
{multiplexer1.pin(7), C3, 1, velocity}, //BUTTON 36

I think this might help... sorry it’s just bits of my code, but i am multiplexing both pots and buttons
I just defined all the multiplexers as “AnalogMultiplex...” but then grouped them as “Analog potentiometers” and “Digital Buttons...”

4dvn, 2020-02-20:
> AnalogMultiplex multiplexer2(25, { 2, 3, 4 } );
> AnalogMultiplex multiplexer3(26, { 2, 3, 4 } );
> AnalogMultiplex multiplexer4(27, { 2, 3, 4 } );
> AnalogMultiplex multiplexer5(28, { 2, 3, 4 } ); [...]

I think i got this work: 

    // Instantiate a multiplexer
CD74HC4067 mux = {
  A0,              // analog pin
  {0, 1, 2, 3}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};

using namespace MIDI_Notes;

NoteButton buttons[] = {
  {mux.pin(0), {note(C, 2), CHANNEL_1}},
  {mux.pin(1), {note(Db, 2), CHANNEL_1}},
  {mux.pin(2), {note(D, 2), CHANNEL_1}},
  {mux.pin(3), {note(Eb, 2), CHANNEL_1}},
  {mux.pin(4), {note(E, 2), CHANNEL_1}},
  {mux.pin(5), {note(F, 2), CHANNEL_1}},
  {mux.pin(6), {note(Gb, 2), CHANNEL_1}},
  {mux.pin(7), {note(G, 2), CHANNEL_1}},
  {mux.pin(9), {note(Ab, 2), CHANNEL_1}},
  {mux.pin(10), {note(A, 2), CHANNEL_1}},
  {mux.pin(11), {note(Bb, 2), CHANNEL_1}},
  {mux.pin(12), {note(B, 2), CHANNEL_1}},
  {mux.pin(13), {note(C, 3), CHANNEL_1}},
  {mux.pin(14), {note(Db, 3), CHANNEL_1}},
  {mux.pin(15), {note(Eb, 3), CHANNEL_1}},
  {mux.pin(16), {note(E, 3), CHANNEL_1}},
};

tttapa, 2020-02-25:
> `{mux.pin(16), {note(E, 3), CHANNEL_1}},`

A 16-pin multiplexer doesn't have a `pin(16)`, the maximum pin number is `15`.