tttapa/Control-Surface #95: CCRotaryEncoder outputing only two values

laual, 2020-01-23:
Hi,

I'm trying to connect a rotary encoder to my Teensy 3.6 with the CCRotaryEncoder class but all I'm getting as output are two values like the DigitalCCSender.

```
#include <Encoder.h>
#include <Control_Surface.h>

USBMIDI_Interface midi;

const int speedMultiplier = 1;

CCRotaryEncoder enc1 = {
  {0,1},                                // pins
  {MIDI_CC::Pan, CHANNEL_1},
  speedMultiplier,                       // multiplier
  4,
};

void setup() {
  Control_Surface.begin();
}

void loop() { Control_Surface.loop(); }
```
Could anyone help with this problem?

Thanks!

tttapa, 2020-01-23:
This is the expected behavior. Rotary encoders send relative MIDI messages (1 = +1, 127 = -1). You can most likely configure relative mode in your DAW or other MIDI software.  
There are three main relative MIDI modes, you can set the correct mode for your software using [`RelativeCCSender::setMode(...)`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dd0/classRelativeCCSender.html#a0b9837f163967bb06206d3144de73eb9) ([example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/d1f/RotaryEncoder_8ino-example.html)).  
If your software doesn't support relative mode, you can use the [`CCAbsoluteEncoder`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d2f/classCCAbsoluteEncoder.html#details) class.

laual, 2020-01-23:
Thanks, I would like to control hardware synths though. `CCAbsoluteEncoder` works already better but implies that I can't go below the initial value. If I'm correct I would need to query the value from my synth and increment it with the `CCRoteryEncoder`? Or is there a better way?

tttapa, 2020-01-23:
>  I can't go below the initial value

Not sure what you mean, `CCAbsoluteEncoder` should go all the way from 0 up to 127, which is the full MIDI CC range.

> If I'm correct I would need to query the value from my synth and increment it with the CCRoteryEncoder? Or is there a better way?

There's an unmerged feature branch https://github.com/tttapa/Control-Surface/tree/share-encoder that has a `set` method for the `CCAbsoluteEncoder` class. You can then query the synth, and set the value of the encoder using this method:

https://github.com/tttapa/Control-Surface/blob/5972cf5148a4cfd38e60c2728938d4a86c9b80cc/src/MIDI_Outputs/Abstract/MIDIAbsoluteEncoder.hpp#L54-L57

You have to use my fork of the Encoder library in order to use the code on that branch:  
https://github.com/tttapa/Encoder  
(Keep in mind that you also have to remove the Encoder library that comes preinstalled with Teensyduino for this to work on a Teensy. On my installation, it's located in `arduino-1.8.9/hardware/teensy/avr/libraries/Encoder`.)