tttapa/Control-Surface #97: support for teensy 3.6 and button matrix?

skyriderone, 2020-01-25:
Hi Pete and control surface community.  I am new to arduino and have spent the past three months or so reading forum posts and experimenting with your midi controller library and now control surface.  THANK YOU!  

I am now trying to assemble my finished controller which will include a 5x6 bankable button matrix, and 16 pots on a mux.

I've built both aspects successfully on a breadboard with Leonardo, but I was hoping to use Teensy 3.6 for the final project because of it's additional digital I/O.  (I think I got the idea from a forum post from pieterp).

Testing my button matrix yesterday with Teensy 3.6, I found that all buttons triggered their appropriate midi note, but most also triggered the adjacent note in that column, but 1 row down.

For instance if I press (row 0, column 0), then (row 1, column 0) is also triggered sporadically.

I was wondering if this was a known issue with the Teensy 3.6?  I read an issue related to using a mux with 3.6, and the solution was to and 5 micro second delay to compensate for the high speed of the teensy. 

My wiring looks correct, so I am hoping this is something that can be corrected on the code side of things.  I am running the most recent version of the control surface library. 

any thoughts?

and thanks again for your inspiring work!

tttapa, 2020-01-25:
Glad to hear you find the library useful :)

> I was wondering if this was a known issue with the Teensy 3.6? I read an issue related to using a mux with 3.6, and the solution was to and 5 micro second delay to compensate for the high speed of the teensy.

Yes, that's a very plausible explanation. You could try adding `delayMicroseconds(5);` after this line:
https://github.com/tttapa/Control-Surface/blob/90d83bbf8fa9d05d74365510275921ed9e5c9c41/src/AH/Hardware/ButtonMatrix.ipp#L25

Please let me know if that fixes it, I haven't been able to try the button matrix on a Teensy 3.6 yet.

4dvn, 2020-01-26:
> Hi Pete and control surface community. I am new to arduino and have spent the past three months or so reading forum posts and experimenting with your midi controller library and now control surface. [...]

Try to adding diode zener

tttapa, 2020-01-26:
I don't think that's the problem here. It's true that you need _diodes_ (not zeners) if you want to be able to press multiple buttons at the same time, but this sounds more like a timing problem.

4dvn, 2020-01-26:
> I don't think that's the problem here. It's true that you need _diodes_ (not zeners) if you want to be able to press multiple buttons at the same time, but this sounds more like a timing problem.

Maybe i will pull request some image of my pcb and video why the diode zener required for multipress button at the same time.

tttapa, 2020-01-26:
> Maybe i will pull request some image of my pcb and video why the diode zener required for multipress button at the same time.

Diodes are required, not zener diodes.

skyriderone, 2020-01-26:
Thanks for the replies!  

For clarification, my matrix uses diodes (not zener).

I'm not completely clear on how to edit the file as you suggested, Pieter, but I will try to figure it out and report back.

skyriderone, 2020-01-27:
Adding 5 microseconds delay did the trick, and my button matrix appears to function perfectly now.  Thanks!

closed 2020-01-27 by skyriderone