tttapa/Control-Surface #100: NoteButton tied to variable rather than pin?

samlochner, 2020-01-27:
Hi, I'm wondering if it is possible to control a NoteButton by a variable rather than a pin. For instance, I have variable "A" and when it is set to 1 the NoteButton would be considered pressed and when it is 0 the NoteButton would be considered not pressed.

Thanks,
Sam

tttapa, 2020-01-27:
This is not possible, and it wouldn't be a good approach. It's better to just send the MIDI messages yourself.

For example: [Send-MIDI-Notes.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d8d/Send-MIDI-Notes_8ino-example.html)

Alternatively, you could look at the source code for NoteButton and write your own MIDI Output Element.

samlochner, 2020-01-27:
Understood, thanks for your speedy help, again, tttapa!

closed 2020-01-27 by samlochner