tttapa/Control-Surface #101: MIDI BLE adapter

ulrichbjerre, 2020-01-27:
Hi!
I joust found your project, at and it seems very promising!

I might have overlooked something?
But, I'm a musician and I've just recently bought a VoiceLive 3 Extreme.
I'm also using an iPad app called OnSong.

I've just tested the ability to have the OnSong app change programs (patches) using a USB to MIDI cable (by connecting it to my iPad using a camera-adapter).

Anyway - I'd looooove to get the connection wireless over BLE - exactly like the Yamaha MD-BT01 adapter.

As far as I can read - I need to use an ESP32 (no problem).
But, the examples seem to only introduce "sending commands"....opposed to what I need - I need to receive MIDI commands and bridge/send to my voicelive pedal through (preferable) MIDI connectors or USB....

I hope you can point me in the right direction.

Thank you!

/Ulrich

tttapa, 2020-01-27:
I've recently added MIDI routing options, you can find the full documentation [here](https://tttapa.github.io/Control-Surface-doc/Doxygen/df/ddc/group__MIDI__Routing.html) and an example [here](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dc4/MIDI_Pipes-Routing_8ino-example.html).

The one caveat is that the ESP32 is a dual core system. Incoming BLE MIDI is handled in a different thread than the main program, which could lead to all kinds of concurrency problems. The released code has mutexes etc. in the right places to prevent concurrent access to critical parts of the program, but I haven't looked at it into detail for the MIDI routing.  
I _think_ the default routing example linked above should work if you just replace one of the MIDI interfaces by a [BluetoothMIDI_Interface](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d89/classBluetoothMIDI__Interface.html), but you might run into problems if you start sending data to the same MIDI interface from your main program as well.

(Note: the MIDI pipes have locks, but these are not thread-safe mutexes.)

I'll address these problems in a couple of days when my exams are over.

ulrichbjerre, 2020-01-29:
Thank you very much for your response!
Just a little recap:
I use OnSong (an iOS app - I don't believe it's available no Android?), which supports MIDI devices through cable, BLE and virtual devices (software MIDI services). The device will in time be connected to my guitar pedal using a DIN 5-pin connection (in and out).

I'm giving you some feedback here, so you can see the progress of my little project - and I'd like to support your work with updates, that you hopefully find useful!
I don't know wether any changes need to be made in your BLE libraries for full support - or if it is OnSong, that need to "do something".

**Current progress**
I did a little testing with an ESP32 device.
I was quite challenged with the MIDI routing example you linked to.
The OnSong app could not discover the ESP32 device at all.

So, I decided to start with the ESP32 example.
I removed every code concerning the VU meter and OLED display - leaving me with only Bluetooth support...However, I was again not able to discover the device in the OnSong app.

Thinking a little "out of the box" I got the idea to test if, I could discover the ESP32 device with a **BLE scanner 4.0** by **Bluepixel Technologies LLP** (another iOS app) - lo and behold! The device was suddenly discoverable and connectable and did in fact show up under the list of connected Bluetooth device in the iPhone/iPad settings (but still not showing up in the OnSong app).

The next thing I did was downloading a MIDI connect application: **Connect bluetooth MIDI Connect** by **Korg Inc**. Here I was again able to discover and connect to the ESP32 device. And more surprisingly, this time I was able to find the ESP32 device inside the OnSong app (already connected - through the Korg app). And more promising - I could "out of the box" receive MIDI program change command which are sent from the OnSong app via BLE to the Serial Monitor in the Arduino IDE.

**So, what's the conclusion?**
The ESP32 device is discoverable and connectable in the Korg application.
Connecting first to the device from the Korg application makes the device connected in OnSong (otherwise it's not showing up).
And - it seems as if data from the OnSong app is indeed received by the ESP32 device and "routed" to the USB-Serial port (Serial 1 I believe?)

**Next steps**
I did check, that the UUID's in you library do match the official UUIDs (as far as I can tell) - off course. I will attempt to contact the OnSong developer crew to exchange information and perhaps enter a collaboration with them (and you?) to make it work.

For my own project I will also need to implement the DIN 5-pin (serial) output to my guitar pedalboard - which probably would be possible with the MIDI routing (PipeFactory) approach?

I hope you can use my current input for something.
Comments and advices are highly appreciated.

/Ulrich

Vahen1981, 2023-12-23:
Hello!
Do you know some app for a macbook, like BLE scanner 4.0?
I can not discover the device that I build with esp32 and Control Surface library in a mac with Ventura OS.
Any suggest?

ssk-dev, 2023-12-23:
@Vahen1981 Did you set the device up to act as a client instead server?
Maybe it make sense to share your code.

Vahen1981, 2023-12-23:
Hi @ssk-dev thanks for your response. I don't think is problem with the code, I can see the device in android and in windows just using the BluetoothMIDI_Interface. I also can see it in an ipad with the app BLE scanner 4.0.
The problem is that I can not see it in a Macbook