tttapa/Control-Surface #103: bank and mux

benwadub, 2020-01-28:
hi, I try to use bank for mux
i d like that all my potentiometer on a mux change the midi cc they send using 2 buttons
tried like this : I explain the problem after code
 //Instantiate second multiplexer for 8 potentiometer
CD74HC4051 mux2 = {
  A4,       // Analog input pin
  {2, 3, 4} // Address pins S0, S1, S2};
};
CCPotentiometer potentiometer[] = {
  {mux2.pin(0), {83, CHANNEL_1}},
  {mux2.pin(1), {83, CHANNEL_2}},
  {mux2.pin(2), {83, CHANNEL_3}},
  {mux2.pin(3), {83, CHANNEL_4}},
  {mux2.pin(4), {83, CHANNEL_5}},
  {mux2.pin(5), {83, CHANNEL_6}},
  {mux2.pin(6), {83, CHANNEL_7}},
  {mux2.pin(7), {83, CHANNEL_8}},
};
//Instantiate second multiplexer for 8 potentiometer
CD74HC4051 mux3 = {
  A3,       // Analog input pin
  {2, 3, 4} // Address pins S0, S1, S2};
};
CCPotentiometer potentiometer[] = {
  {mux3.pin(0), {83, CHANNEL_1}},
  {mux3.pin(1), {83, CHANNEL_2}},
  {mux3.pin(2), {83, CHANNEL_3}},
  {mux3.pin(3), {83, CHANNEL_4}},
  {mux3.pin(4), {83, CHANNEL_5}},
  {mux3.pin(5), {83, CHANNEL_6}},
  {mux3.pin(6), {83, CHANNEL_7}},
  {mux3.pin(7), {83, CHANNEL_8}},
};
// Create a new bank that has two tracks per bank
Bank bank(2); //create a bank
// Create a new bank selector that changes the bank setting of the bank we just created
// It has pushbuttons connected to pins 11 and 12 that increment or decrement the bank setting,
// and 4 LEDs to pins 4, 5, 6 and 7 that display the current bank setting.
BankSelector bankSelector(bank, { 11, 12 }, { 4, 5, 6, 7 } );
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
  
bank.add(mux1, Bank::CHANGE_ADDRESS); // When the bank setting is changed, change the address (note number) of the mute button
bank.add(mux3, Bank::CHANGE_ADDRESS);
}
 

}                           // the multiplexers sets it for all of them)
void loop() {

  Control_Surface.loop(); // Update the Control Surface
}

but I have a invalid use of template-name cs bank

tttapa, 2020-01-29:
Relevant documentation:
- [**Bank**](https://tttapa.github.io/Control-Surface-doc/Doxygen/db/dbd/classBank.html#details)
- [**Bankable MIDI Output Elements**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/d00/group__BankableMIDIOutputElements.html)
- [**Selectors**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html)

Note that the bank and the mux are completely separate. You can use a bank with or without a mux, and a mux with or without a bank.  
Most objects take a "pin" as a parameter. It doesn't matter if the pin is a real Arduino pin, or if it's the pin of a mux.  
You have to use the MIDI Elements in the `Bankable` namespace if you want to use them with a bank. They have an extra argument where you pass in the bank configuration.

By default, the bank changes the address of the MIDI Element:
```cpp
Bankable::CCPotentiometer potentiometer = {bank, mux2.pin(0), {83, CHANNEL_1}};
```
If you want to change the channel, you can use `BankType::CHANGE_CHANNEL`:
```cpp
Bankable::CCPotentiometer potentiometer = {{bank, BankType::CHANGE_CHANNEL}, mux2.pin(0), {83, CHANNEL_1}};
```

benwadub, 2020-01-29:
Thanks, I must admit that I don t really understand  the documentation document :-)
I try to understand how to use library with the examples I find in the library cause I really don t know how to write code :-)
But I begin to understand some things, thanks!
Do you have a link where is explain how to use and write the module of library’s? It could help me and stop asking noob question every days :-)

tttapa, 2020-01-29:
All you have to do to use the library is create the right objects. The documentation tells you how to do that and what parameters you have to use. The syntax is always the same, it's C++ uniform initialization (using `{ arguments }`).  
The groups/modules in the documentation just give an overview of the types of objects you can use, and are the best starting point.

There should probably be more examples for the Bank class, though.

> Do you have a link where is explain how to use and write the module of library’s?

What is a module of library's?

benwadub, 2020-01-29:
what I called Module is the objects in fact :-)
like I learn c++ with this project there is a lot of things that are not clear for me!
i ll try to understand the documentation!
i ll investigate this evening! again a big thanks to take time for me mr Pieter!

benwadub, 2020-01-29:
oh damn I can t figure out to make the selector button now
i tied this but
I got this :
could not convert '{bank, {A2, A3}}' from '<brace-enclosed initializer list>' to 'CS::Selector<2>'
here is what I wrote for selector
};
Bank<2> bank = {1}; // 2 banks, 8 addresse per banks

Bankable::CCPotentiometer potentiometer [] = {
  {bank, mux2.pin(0), {83, CHANNEL_1}},
  {bank, mux2.pin(1), {83, CHANNEL_1}},
  {bank, mux2.pin(2), {83, CHANNEL_1}},
  {bank, mux2.pin(3), {83, CHANNEL_1}},
  {bank, mux2.pin(4), {83, CHANNEL_1}},
  {bank, mux2.pin(5), {83, CHANNEL_1}},
  {bank, mux2.pin(6), {83, CHANNEL_1}},
  {bank, mux2.pin(7), {83, CHANNEL_1}},

};
Selector<2> bankselector = {
bank, {A2, A3},

};

tttapa, 2020-01-29:
`Selector` is a generic selector, it doesn't read input from any hardware. You'll see in the [documentation](https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/d54/classSelector.html#details)  that it only takes the selectable (the thing that can be selected, the bank, in this case) as an argument.  
Your code tries to pass pin numbers to the `Selector` constructors, so that won't work.

If you want a selector that reads input from push buttons, you'll have to pick a different selector from the [Selectors](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html) module.  
You could use the [`IncrementDecrementSelector`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/deb/classIncrementDecrementSelector.html#details), for example.  
If you look at the [documentation for the constructor](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/deb/classIncrementDecrementSelector.html#a75eabd0b6c91592f1448960cd7c42366), you'll see that the first argument is again the `selectable`, and the second argument are the `buttons`. The third argument has a default value, so we won't worry about it for now.

The type of the `buttons` argument is `IncrementDecrementButtons`. You can click on this type name in the documentation, and it will take you to the [documentation for `IncrementDecrementButtons`](https://tttapa.github.io/Control-Surface-doc/Doxygen/df/dc6/classAH_1_1IncrementDecrementButtons.html#a169fb832f90b7d02106cf895e3c13f13). Scroll down to the constructor again, and look at the arguments. The first argument is the increment button, and the second argument is the decrement button. You can then click on the `Button` class and look at the arguments. You'll see it has a single argument, the pin number.

Let's work our way back up now:

A `Button` can be constructed from a single pin number:

```cpp
pin_t myPin = A0;
Button myButton = {myPin};
```
Initialization is done using curly braces in C++, but if there's only one argument, they can be omitted:
```cpp
Button myButton = myPin;
```

Then you can use buttons to construct an `IncrementDecrementButtons` object:

```cpp
IncrementDecrementButtons myButtons = {
  Button{A0}, // increment button
  Button{A1}, // decrement button
};
```
You don't have to explicitly add the `Button` type here, because the compiler already knows the type of the arguments, you can also omit the braces again, because the `Button` constructor has only one argument:
```cpp
IncrementDecrementButtons myButtons = {
  A0, // increment button
  A1, // decrement button
};
```
Finally, we can use the `IncrementDecrementButtons` to initialize the `IncrementDecrementSelector`:
```cpp
IncrementDecrementSelector<2> mySelector = {
  bank, // selectable
  IncrementDecrementButtons{A0, A1}, // buttons
};
```
Again, there's no need to explicitly write the type `IncrementDecrementButtons`, because the compiler already knows the type of the argument. This time, you can't omit the curly braces, because the `IncrementDecrementButtons` constructor has two arguments:
```cpp
IncrementDecrementSelector<2> mySelector = {
  bank, // selectable
  {A0, A1}, // buttons
};
```

benwadub, 2020-01-30:
yessss thanks you are a king! I understand now how all the library works I ll try to not ask anymore boring question for  you!

benwadub, 2020-01-30:
if I understand well my other problem 
CCAbsoluteEncoder' is not a member of 'CS::Bankable'
it mean that I can't use  ccabsoluteencoder in a bank?

tttapa, 2020-01-30:
Indeed. I usually develop and test the non-bankable version first, and then later create a bankable version as well. I haven't had the time to create a bankable CCAbsoluteEncoder. 

Tomorrow is my last exam, so I'll see if I can add it tomorrow or the day after.

benwadub, 2020-01-30:
ok thanks! the cool thing is that I understood :-)
good luck for exam!

an other thing, I tried to make an array with my encoders but 
I can t find a way to write it without error!
 };
 CCAbsoluteEncoder encoder [] = {
  {5, 6}, {82, CHANNEL_1}, 16, 4,}},
  {7, 8}, {82, CHANNEL_1}, 16, 4,}},
  {9, 10},  {82, CHANNEL_1}, 16, 4,}},
  {11, 12}, {82, CHANNEL_1}, 16, 4,}},
  {A0, A1}, {82, CHANNEL_1}, 16, 4,}},
  {A2, A3}, {82, CHANNEL_1}, 16, 4,}},
  {A4, A5}, {82, CHANNEL_1}, 16, 4,}},
  {A6, A7}, {82, CHANNEL_1}, 16, 4,}},

};

tttapa, 2020-01-30:
Thanks!

Each element of the array is a `CCAbsoluteEncoder` object, so it needs a pair of braces around it.
```cpp
CCAbsoluteEncoder encoder [] = {
  {{5, 6}, {82, CHANNEL_1}, 16, 4,},
  {{7, 8}, {82, CHANNEL_1}, 16, 4,},
  // ...
};
```

benwadub, 2020-01-30:
oh my god it was so easy!! thanks again for all you share! you are a really nice guy!

benwadub, 2020-02-08:
hello piéter, i come back to my selector work! my controller is nearly soldered!
i don t understand how increment decrement selector work, I don t understand where I must write what is the destination of change.
I think I m not clear.
i have by example my mux for potentiometer that act on cc7 on channel 1 to 8 this work fine
where do I set the selector destination for by example when I push my increment button all my fader act on cc 8 on the same channel as before?

her is my mux and selector
Bankable::CCPotentiometer fader [] = {
  {bank3, mux3.pin(0), {7, CHANNEL_1}},
  {bank3, mux3.pin(1), {7, CHANNEL_2}},
  {bank3, mux3.pin(2), {7, CHANNEL_3}},
  {bank3, mux3.pin(3), {7, CHANNEL_4}},
  {bank3, mux3.pin(4), {7, CHANNEL_5}},
  {bank3, mux3.pin(5), {7, CHANNEL_6}},
  {bank3, mux3.pin(6), {7, CHANNEL_7}},
  {bank3, mux3.pin(7), {7, CHANNEL_8}},

};
 IncrementDecrementSelector<2> selector fader = {
  bank3, // selectable
  {A7, A8}, // buttons
 };

tttapa, 2020-02-08:
I've recently added a new [**Bank example**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d74/Bank_8ino-example.html) that might clear up some of the confusion.

The bank determines by how much the address changes.  
The bank configuration you pass to the Bankable::CCPotentiometer constructor determines what part of the address changes (the controller number, the channel, or the cable).  
The selector can control the bank to change the address.

tttapa, 2020-02-08:
If you want the address to change from controller 7 to controller 8, you need banks with 1 track per bank, and two banks in total:

```cpp
Bank<2> bank(1);
```

You need something to select which bank is active, you can use any [**Selector**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html), but since you can only switch between two banks, I think the [IncrementSelector](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d4c/classIncrementSelector.html) and the [SwitchSelector](https://tttapa.github.io/Control-Surface-doc/Doxygen/df/def/classSwitchSelector.html) are the best candidates.

```cpp
IncrementSelector<2> selector = {
  bank, // the selector selects the bank
  A7, // pin number of the push button
};
```

Finally, the bank has to be added to the potentiometers, after all, the goal of the bank is to change the address of the potentiometers.  
You have the choice to make the bank change the controller number, the channel, or the MIDI USB cable number.  
Changing the controller number is the default, the example I linked to in my previous reply demonstrates how to change the channel instead.

```cpp
Bankable::CCPotentiometer fader = {
  {bank, BankType::CHANGE_ADDRESS},
  pin, 
  address,
};
```

benwadub, 2020-02-08:
Thanks a lot again!!! I ll look at it tomorrow
I m making a great controller with you really really thank you! I m translating the explanation in the code in French and I ll make a pdf tutorial with all schematics for building I ll give it all that when it ll be finished if you have French follower that could help them!

benwadub, 2020-02-09:
hi, again me :-)
i want to put my 3 selector on a mux but I can t find how to write it, could you please help me?
here what I wrote:

CD74HC4051 mux4 = {
  A3,     
  {2, 3, 4} 

};

CCButtonLatched muxselector[] = { 
 IncrementSelector<2> selectorbutton = {bank2, {mux4.pin(0)}},
 IncrementSelector<2> selectorbutton2 = {bank1, {mux4.pin(1)}}, 
 IncrementSelector<2> selectorvolume = {bank3, {mux4.pin(2),}},

  
};

tttapa, 2020-02-10:
The selectors are completely separate from the CCButtons.

```cpp
IncrementSelector<2> selectorbutton = {bank2, {mux4.pin(0)}};
IncrementSelector<2> selectorbutton2 = {bank1, {mux4.pin(1)}};
IncrementSelector<2> selectorvolume = {bank3, {mux4.pin(2),}};
```

benwadub, 2020-02-12:
thanks, sorry for the delay of reply I have been busy at work!
so I just have to instantiate my selectors with the mux pin adress?

benwadub, 2020-02-12:
I did it!!! thank you very much!!
could you check my code and tell me if there are things I should remove?
maybe you can use it as an example, it s ok to use 16 potentiometer, 8 faders, 16 encoders and 19 buttons

#include <Encoder.h> // Inclu la library encoder a télécharger(en installant tinsy duino sur https://www.pjrc.com/teensy/td_libs_Encoder.html
// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
 #include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
 #include <Arduino_Helpers.h> // Inclu la library arduino helper
#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>// inclu la library pour les multiplexer
 //USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

auto &serial = Serial;// Selectionne le port série à utiliser

SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
};// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

 
//using namespace MIDI_Notes;  ces lignes midi servent à activer un bouton qui envoit une note midi enlever les // sur ces 5 lignes pour l'activer

// démarrer une touche midi
//NoteButton button = {
  //5,                       // numéro de l'entrée arduino sur laquelle le bouton est branché
  //{note(C, 2), CHANNEL_5}, // note C4 sur le canal 5 libre à vous de changer les numéro et d'en ajouter

// active un multiplexeur (une sorte de multiprise sur laquelle vous branchez 8 controleurs sur 1 seule entrée de l'arduino
CD74HC4051 mux1 = {
  A2,       // numéro de broche de l'arduino
  {2, 3, 4} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};
Bank<2> bank2 = {8}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched buttonmute[] = { //ces lignes déclarent le mux
  {{bank2,BankType::CHANGE_CHANNEL}, mux1.pin(0), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(1), {94, CHANNEL_2}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(2), {94, CHANNEL_3}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(3), {94, CHANNEL_4}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(4), {94, CHANNEL_5}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(5), {94, CHANNEL_6}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(6), {94, CHANNEL_7}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(7), {94, CHANNEL_8}},
};
Bank <2> bank3 = {8}; // 2 banks, 8 addresse per banks

Bank<2> bank1 = {-8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  A1,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(12), {82, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(13), {82, CHANNEL_6}},
[... 39 lines cut ...]
   mux4.pin(0), // pins sur lequel les boutons sont branchés
};
IncrementSelector<2> selectorbutton2 = {
  bank1, // banque sélectionnée
  mux4.pin(1),  // adresse du bouton
};
IncrementSelector<2> selectorvolume = {
  bank3, // banque sélectionnée
  mux4.pin(2),  
};
void setup() {
  Control_Surface.begin(); // initialise la library surface de control

 

}                           
void loop() {
  
    
    
  
  

  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2020-02-16:
Glad to hear you got it wokring!

You don't need these includes, they are included by the main `<Control_Surface>` header:
``cpp
#include <MIDI_Interfaces/SerialMIDI_Interface.hpp>
 #include <Arduino_Helpers.h> // Inclu la library arduino helper
#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>// inclu la library pour les multiplexer
```

benwadub, 2020-02-16:
Thanks to you! I ll have this nice controller!
Thanks thanks thanks!!!
[image]

tttapa, 2020-02-25:
Looks good!

closed 2020-04-05 by tttapa