tttapa/Control-Surface #105: Combo button & matrix

4dvn, 2020-01-31:
How can i add this combo push button Matrix code?

 void checkCombo(boolean s) {

  // from MIDI Fighter Pro documentation:

  // COMBOS
  //
  //   A          B           C           D           E
  // +--------+ +--------+  +--------+  +--------+  +--------+
  // |        | |        |  |        |  |        |  |        |
  // |        | |        |  |  3 4   |  |        |  |u       |
  // |        | |n n n n |  |  1 2   |  |a b c d |  |l r A B |
  // |1 2 3 4 | |        |  |        |  |        |  |d       |
  // +--------+ +--------+  +--------+  +--------+  +--------+
  //                                     a-b-c-c-d   uuddlrlrBA
  //
  // Combos retain a NoteOn while the final key is depressed and emit a NoteUp
  // when it is released.
  // Combo A               G#-2
  // Combo B               A-2
  // Combo C               A#-2
  // Combo D               B-2
  // Combo E	C-1

  // lets simplify things a bit :)
  //   A          B           C           D           E
  // +--------+ +--------+  +--------+  +--------+  +--------+
  // |        | |        |  |        |  |        |  |        |
  // |        | |        |  |  x x   |  |        |  |x       |
  // |        | |x x x x |  |  x x   |  |  x x x |  |    x x |
  // |x x x x | |        |  |        |  |        |  |x       |
  // +--------+ +--------+  +--------+  +--------+  +--------+

  //no combo active and a button is pressed
  if ((comboState==0)&&(s)) {
    // combo A
    if ((buttonState[0]->read()==LOW)
      &&(buttonState[1]->read()==LOW)
      &&(buttonState[2]->read()==LOW)
      &&(buttonState[3]->read()==LOW)) {
      comboState=1;
      usbMIDI.sendNoteOn(8, 127, midiChannel);
    }
    else
      //combo B
      if ((buttonState[4]->read()==LOW)
        &&(buttonState[5]->read()==LOW)
        &&(buttonState[6]->read()==LOW)
        &&(buttonState[7]->read()==LOW)) {
        comboState=2;
        usbMIDI.sendNoteOn(9, 127, midiChannel);
      }
      else
        //combo C
        if ((buttonState[5]->read()==LOW)
          &&(buttonState[6]->read()==LOW)
          &&(buttonState[9]->read()==LOW)
          &&(buttonState[10]->read()==LOW)) {
          comboState=3;
          usbMIDI.sendNoteOn(10, 127, midiChannel);
        }
        else
          //combo D
          if ((buttonState[5]->read()==LOW)
            &&(buttonState[6]->read()==LOW)
            &&(buttonState[7]->read()==LOW)) {
            comboState=4;
            usbMIDI.sendNoteOn(11, 127, midiChannel);
          }
          else
            //combo E
            if ((buttonState[0]->read()==LOW)
              &&(buttonState[8]->read()==LOW)
              &&(buttonState[6]->read()==LOW)
              &&(buttonState[7]->read()==LOW)) {
              comboState=5;
              usbMIDI.sendNoteOn(12, 127, midiChannel);
            }
  }
  else if ((comboState!=0)&&(!s)) {//combo was active and released
    switch(comboState) {
    case 1:
      usbMIDI.sendNoteOff(8, 0, midiChannel);
      break;
    case 2:
      usbMIDI.sendNoteOff(9, 0, midiChannel);
      break;  
    case 3:
      usbMIDI.sendNoteOff(10, 0, midiChannel);
      break;
    case 4:
      usbMIDI.sendNoteOff(11, 0, midiChannel);
      break;  
    case 5:
      usbMIDI.sendNoteOff(12, 0, midiChannel);
      break;
    }
    comboState=0;
  }
}
    When press the combo is change the note bank

tttapa, 2020-02-01:
What do you want to add it to? What does it have to do with banks?

4dvn, 2020-02-02:
> What do you want to add it to? What does it have to do with banks?

I think is the bank when push this combo button, can you adding this example: https://github.com/danielytics/Midifighter/blob/master/DJTT%20Latest/combo.c

tttapa, 2020-02-04:
I'm sorry, but I really have no idea what you're trying to do. Maybe communication will be easier if you post your questions in your native language instead of English.

closed 2020-02-15 by 4dvn