tttapa/Control-Surface #107: Control surface with XBEE

skyriderone, 2020-02-01:
I'm planning a wireless controller but am hoping for better range than Bluetooth.

Is there any reason that the library would not work using XBEE wireless?

Thanks

tttapa, 2020-02-01:
You should be able to get it to work. If the XBee library implements the `Stream` interface, you can use the `StreamMIDI_Interface`. If not, it shouldn't be too hard to adapt the `StreamMIDI_Interface` to your needs: <https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Interfaces/SerialMIDI_Interface.hpp>

skyriderone, 2020-02-04:
thanks for the advice.  I'll let you know how it goes.

radiservi, 2020-12-11:
hello Pieter can you explain to me how the stream works with the wire library because I cannot make it work