tttapa/Control-Surface #108: [Request] new mode for relativeCCmode

LukaszChrzanowski, 2020-02-01:
Hello,

there is mode settings for CCRotaryEncoder  -> RelativeCCSender::setMode(relativeCCmode::{MODE});
Could You add mode that will send +1 (eg.right) -1 (eg.left) for given range (eg. 1<->127 or 1 <->2)?

tttapa, 2020-02-01:
I'm not sure what you mean. MIDI data bytes are 7-bit unsigned integers, `-1` doesn't exist in this representation.  
The three available modes are the only ones I'm aware of, and the only ones that are supported by major DAWs.

LukaszChrzanowski, 2020-02-01:
hmm... maybe this is my lack of knowledge, but in mode there are it looks like only two fixed values (MIDI-OX output):
TRACKTION_RELATIVE - values 01 - 7F
REAPER_RELATIVE_3 - values 01 - 41
REAPER_RELATIVE_2 - values 3F - 41
REAPER_RELATIVE_1 - values 01 - 7F
MACKIE_CONTROL_RELATIVE - values 01 - 41

in some DAW You can write Your's own codecs, and assign hardware specific control to any control in DAW. So in example You can choose channel of whatever You will want...

maybe I should go to reverse engineering Mackie codecs provided for my DAW, but as far as I can see, encoders there have two patterns:
(channel CC value)
b? 01 xxxx
b? 01 <?y??>x

end then some crazy calculations on them: x*(1-2*y) o.O

and then used in some software controls in DAW, where range is specified (ew. 0-127)... so I thought easiest way will be to directly send this messages. Probably I'm wrong, and malybe I should sleep with this topic.

in my request range will tell that returned values will be not outside from range. In my example You will never go to -1 becouse range starts from 1, and never go above 127 (max value from range). But this will make library to hold state of last value of encoder... Maybe this isn't good idea.

tttapa, 2020-02-02:
The Mackie protocol for rotary encoders is as follows:
[image]
The Control Surface library follows the Mackie convention if you use the `MACKIE_CONTROL_RELATIVE` relative CC mode.

Do you mean you want to send the absolute position instead of the relative one? In that case, use [`CCAbsoluteEncoder`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d2f/classCCAbsoluteEncoder.html#details) instead of `CCRelativeEncoder`.

LukaszChrzanowski, 2020-02-02:
Wait, 

so when I set my encoder like this:

```
CCRotaryEncoder enc = {
  {9, 10},       // pins
  {40, CHANNEL_1}, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};
(..)
RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE); 
```

i should get output:

> B0 28 01 (clockwise)
> B0 28 11 (counter-clockwise)

but instead I'm receiving:

> B0 28 01 (clockwise)
> B0 28 7F (counter-clockwise)

tttapa, 2020-02-02:
No, you should get
> B0 28 01 (clockwise)
> B0 28 41 (counter-clockwise)

I just tested it, and this is indeed what it sends.

closed 2020-03-21 by tttapa