tttapa/Control-Surface #110: ESP32 Output Pin

ManoloCap, 2020-02-03:
Hello! I found your project and it is awesome!

I am making my own MIDI controller, I just need a matrix of potenciometers.

Now I have a NodeMCU ESP32, so for first I want to make this by wire.

and that was not work
I teste with BLE but that was not work too, maybe its de NodeMCU version?

I did this:

#include <Control_Surface.h>
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp>
BluetoothMIDI_Interface midi;
CCPotentiometer pot = { 13, MIDI_CC::General_Purpose_Controller_1 };

void setup() { Control_Surface.begin(); }
void loop() { Control_Surface.loop(); }

And I was monitoring it with MIDI-OX

tttapa, 2020-02-03:
The ESP32 doesn't have native USB support, it uses a separate, single-purpose USB-to-Serial chip.

You could either use Hairless MIDI, 5-pin DIN MIDI, or a Debug MIDI Interface: [**MIDI over USB**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html)  
[List of MIDI Interfaces](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/df0/group__MIDIInterfaces.html)

The Bluetooth interface should work, but I've only really tested it on Ubuntu 16.04 with a manually upgraded BlueZ driver, and on Android 9. On Android, I had to pair the device using the `MIDI+BTLE` app first, before it showed up as a MIDI device.  
I'll boot up a Windows 10 and Ubuntu 18.04 machine later to day to see if it works.

tttapa, 2020-02-04:
I've just tested it on Windows 10, and I had to install MIDIBerry for it to work. It seems like the 1809 update of Windows 10 doesn't support MIDI over BLE out of the box. I can't comment on the 1903 update.

On Ubuntu 18.04 and Ubuntu 16.04, I had to upgrade BlueZ to version 5.50 manually ([instructions](https://tttapa.github.io/Pages/Ubuntu/Software-Installation/BlueZ.html)).  
Ubuntu 18.10 and later ship with BlueZ 5.50 already, so they should work out of the box.

ManoloCap, 2020-02-04:
Thank you for the feedback! I will try it on Ubuntu 18 and on Windows 10 with MIDIBerry