tttapa/Control-Surface #111: CC Button Toggle

crazymaca69, 2020-02-03:
Hi, 

I'm very new to Teensy and am attempting to build a midi controller for my Boss ES8. I want to send CCs with a button but it needs to toggle with each full press rather than on the rising and falling edge (similar to a mute button in a DAW). is there a function for that?

I've tried some of your example code but can't see a way of solving it.

tttapa, 2020-02-03:
You can use [`CCButtonLatched`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da6/classCCButtonLatched.html#details).  
Have a look at the [**MIDI Output Elements** module](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/dcd/group__MIDIOutputElements.html) for a complete overview.

crazymaca69, 2020-02-03:
Excellent thanks is there any way to use CC numbers instead of defining the name of the controller?

I'm using your MIDI Controller example using banks.

I want to have 2 banks of 6.

Bank 1 will send out specific CC numbers from each button
Bank 2 will send out another set of specific CC numbers from each button.

All buttons will be on the same MIDI channel.

My bank button is just a normal momentary switch. Do I need to use a proper toggle switch? I'm having to hold this button down to switch banks but it doesn't seem to have the right functionality.

Thanks for your help on this, some of the documentation is just a little hard to understand for me which is why I'm going from some of your examples and deconstructing them.

tttapa, 2020-02-03:
> Excellent thanks is there any way to use CC numbers instead of defining the name of the controller?

You can just use the number instead of the name:
```cpp
CCButtonLatched button = {pin, {0x10, CHANNEL_1}};
```
You can replace `0x10` with the number you want (between 0 and 120).

I'm still working on the documentation, there's a lot to cover.  
I'll try to add a bank example later today.

You can use a momentary switch for the bank. The easiest way is to use an [`IncrementSelector<2>`](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d4c/classIncrementSelector.html#details).

crazymaca69, 2020-02-03:
Thanks got that working.

If I add a multi coloured LED, how do I get it to change colour relative to the bank I'm in? It currently flashes to indicate that the button has been pressed. It would be good to test this with the built in LED but I'll eventually insert a different one.

With the bank function:

When I toggle a button in mode A on midi channel 2, and then switch the bank button, the first message sent is still in midi channel 2. Is there a way of resetting this after the bank is changed?

Here's my code:

`#include <Encoder.h>
// Encoder has to be included before Control_Surface
#include <Control_Surface.h>

HardwareSerialMIDI_Interface midi = {Serial2, MIDI_BAUD};
 
Bank<2> bank(1); // A bank with 1 channel, and 2 bank settings

Bankable::CCButtonLatched TopRow[] = {
  {{bank, BankType::CHANGE_CHANNEL}, 3, {0x5E, CHANNEL_2}},
  {{bank, BankType::CHANGE_CHANNEL}, 4, {0x5F, CHANNEL_2}},
  {{bank, BankType::CHANGE_CHANNEL}, 5, {0x60, CHANNEL_2}},
};

CCButtonLatched BottomRow[] = {
  {0, {0x57, CHANNEL_2}},
  {1, {0x56, CHANNEL_2}},
  {2, {0x55, CHANNEL_2}},
};

IncrementSelector<2> selector = {bank, 6,};
 
void setup() {
  Control_Surface.begin();
  pinMode(LED_BUILTIN, OUTPUT);
}
 
void loop() { // Refresh all inputs
  Control_Surface.loop();
  digitalWrite(LED_BUILTIN, digitalRead(6));
}`

tttapa, 2020-02-03:
> If I add a multi coloured LED, how do I get it to change colour relative to the bank I'm in? It currently flashes to indicate that the button has been pressed. It would be good to test this with the built in LED but I'll eventually insert a different one.

You can add a custom callback to the selector, as shown in this example: <https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4b/Custom-Selector-Callback_8ino-example.html>

> When I toggle a button in mode A on midi channel 2, and then switch the bank button, the first message sent is still in midi channel 2. Is there a way of resetting this after the bank is changed?

This requires some extra work, I'll see what I can do.

tttapa, 2020-02-03:
I think it should be fixed now. 

You have to specify the size of the bank, because the compiler has to know beforehand how many button states it needs to remember.

This is the code I used to test it:

```cpp
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

Bank<8> bank(1);
IncrementDecrementSelector<8> selector = {
  bank,
  {A0, A1},
};

// Instantiate a bankable CCButtonLatched object with 8 banks
Bankable::CCButtonLatched<8> button = {
  bank,
  2,                 // Push button on pin 2
  {0x10, CHANNEL_1}, // MIDI address
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

crazymaca69, 2020-02-03:
when I try to run your code I get:

Test:13: error: 'CS::Bankable::CCButtonLatched' is not a template
 Bankable::CCButtonLatched<8> button = {
           ^

tttapa, 2020-02-03:
Did you upgrade the library to the latest `master` version?

crazymaca69, 2020-02-03:
ah sorry didn't really you'd done another version. 
Me = N00b

crazymaca69, 2020-02-03:
Ok I'm pretty sure I've updated and I'm still getting the same message

crazymaca69, 2020-02-03:
no my bad, it's worked. thanks

crazymaca69, 2020-02-03:
ok all working fine, I'm just struggling with the RGB LED now. Just cannot get the damn thing to illuminate using your code. 

> You can add a custom callback to the selector, as shown in this example: https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4b/Custom-Selector-Callback_8ino-example.html

Should this code work as printed in the example? 
I changed it a little to use a GenericIncrementSelector as I only want one button to scroll through the banks but neither seemed to work for me.

Here is my code now without any of the LED gubbins. I have the bank button on pin 8, any chance you could help me compile it? 

`#include <Encoder.h>
// Encoder has to be included before Control_Surface
#include <Control_Surface.h>

HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};
 
Bank<2> bank(1); // A bank with 1 channel, and 2 bank settings
IncrementSelector<2> selector = {bank, 8,};
const int speedMultiplier = 1;

Bankable::CCButtonLatched<2> TopRow[] = {
  {{bank, BankType::CHANGE_CHANNEL}, 5, {0x5E, CHANNEL_2}},
  {{bank, BankType::CHANGE_CHANNEL}, 6, {0x5F, CHANNEL_2}},
  {{bank, BankType::CHANGE_CHANNEL}, 7, {0x60, CHANNEL_2}},
};

CCButtonLatched BottomRow[] = {
  {2, {0x57, CHANNEL_2}},
  {3, {0x56, CHANNEL_2}},
  {4, {0x55, CHANNEL_2}},
};

CCPotentiometer fader[] = {
  {A0, {0x62, CHANNEL_2}},
};
 
void setup() {
  Control_Surface.begin();
  pinMode(LED_BUILTIN, OUTPUT);
}
 
void loop() { // Refresh all inputs
  Control_Surface.loop();
  digitalWrite(LED_BUILTIN, digitalRead(8));
}`

tttapa, 2020-02-03:
The example code requires a MIDI interface to be instantiated. I tested it using a newer version of the library I'm working on, where this is not required.

I just tested the example using the `master` version that you have installed, and it worked fine after adding the line `USBMIDI_Interface midi;` to the code.

crazymaca69, 2020-02-03:
Sorted, tested with my code. thanks a million!

closed 2020-02-03 by crazymaca69