tttapa/Control-Surface #112: how to delay a NoteButton?

skyriderone, 2020-02-04:
hello again and thanks for all the help.  

If you'd rather I ask my questions in the arduino forum, please let me know. 

So far my controller is coming together great.  I've got a 30 button matrix with 4 banks and led indicator lights.  I also have 12 pots which can be switched to ldr vactrol control from an input audio source.

I am now trying to implement 2 vactrol controlled NoteButtons.  They work great with the provided NoteButton example, but are, as expected, too reactive.  I want to add a delay of a few seconds between midi note off and on commands, so i don't overload the device I am controlling (a raspberry pi video sampler).

I think I need to use the "blink without delay" technique, but am unsure of which file I should edit, and want to make sure I won't negatively effect the other elements of my controller in doing so.

thanks again.

tttapa, 2020-02-04:
You can start from the MIDIButton class:  
https://github.com/tttapa/Control-Surface/blob/dbbfd17ba88ea6b20987ffe46fdbbcc260b53369/src/MIDI_Outputs/Abstract/MIDIButton.hpp#L35-L42

It has a generic [MIDI Sender](https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Senders/DigitalNoteSender.hpp), it doesn't have to be that way if you only want to use it with MIDI note events. 

I wrote an example on how to create a basic MIDI Output Element class:  
[Custom-MIDI-Output-Element.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html)

music59940, 2020-09-25:
Bonjour

J'aimerais aussi retarder de quelque ms les note on
avez vous trouvez une solution
merci

tttapa, 2020-09-25:
You'll have to be more specific. Do you want to delay the note off event as well, or just the note on event? What should happen when the button is released before the delay runs out? Should it still send the note on event, or should it be canceled?

As mentioned in my previous reply, you can easily implement this yourself using the Custom-MIDI-Output-Element example.

music59940, 2020-09-25:
Je voudrais que lorsque j'appuie sur le bouton la NoteOn doit être jouer après 50ms mais ne doit pas affecter la NoteOff

merci

music59940, 2020-09-28:
Bonjour

Pouvez vous me répondre 
merci beaucoup

tttapa, 2020-09-28:
I explained how you can implement this. What have you tried?  
You also didn't answer all of questions in my previous reply.