tttapa/Control-Surface #116: Using two arduino board to increase the number of in to add encoders

benwadub, 2020-02-09:
Hi Pieter, do you know if there is a way to use 2 arduino to add my 16 encoders to my existing project? If yes could you show me if there is a class for it?

LukaszChrzanowski, 2020-02-09:
why You didn't just simply use for example MC14067 multiplexer? You will "sacrifice" 5 pins to get 16 in exchange?

benwadub, 2020-02-09:
I use some for my buttons et pot but encoders can t be used on mux so 16 encoder will use 32 pins on the Arduino :-(

tttapa, 2020-02-09:
No, there's no class for it.

Using some kind of port expander is probably going to be the easiest solution, but unless you can figure out how to use I2C inside of an ISR (I couldn't), you'll have to resort to polling.  
An Arduino Uno/Nano only has 2 interrupt pins apart from the TX/RX pins, so using a second Arduino will not improve the performance of the encoder code, unless you use the pin change interrupts. I don't think the standard Encoder libraries support this.

You can relatively easily create your own class for it. Once you can read the position of the rotary encoder correctly, sending MIDI messages is trivial: <https://tttapa.github.io/Control-Surface-doc/Doxygen/db/d50/Abstract_2MIDIRotaryEncoder_8hpp_source.html>

This example demonstrates how to create your own MIDI Output Element:  
<https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html>

LukaszChrzanowski, 2020-02-09:
I think there is solution to this, but You will need two multiplexers:
https://forum.arduino.cc/index.php?topic=598973.0 

tttapa - could You add class for this? this would be great - I have this problem also (this is reason why I'm using simple rotary potentiometers instead encoders)... :)