tttapa/Control-Surface #117: Need help with shift button bank

4dvn, 2020-02-11:
How can i create the one shift button to change to 8 banks of button matrix midi like this video explain?
https://drive.google.com/file/d/1D5frl2iblFufQaTg56cpbIfHbpVd7cMv/view?usp=drivesdk

benwadub, 2020-02-11:
Use the bankable and bank classe! It work for button or potentiometer

4dvn, 2020-02-12:
> Use the bankable and bank classe! It work for button or potentiometer

Can you give me the example of it, here is my first code:
```
#include <Control_Surface.h>

USBMIDI_Interface midi;

// The note numbers corresponding to the buttons in the matrix
const AddressMatrix<4, 4> addresses = {{
  {51, 50, 49, 48}, // Ableton drumrack map
  {47, 46, 45, 44},
  {43, 42, 41, 40},
  {39, 38, 37, 36},
}};

NoteButtonMatrix<4, 4> buttonmatrix = {
  {2, 3, 4, 5}, // row pins
  {6, 7, 8, 9},    // column pins
  addresses,    // address matrix
  CHANNEL_1,    // channel and cable number
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

4dvn, 2020-02-15:
@tttapa I can'nt find how to make this :(

tttapa, 2020-02-15:
I don't have any experience with the software he's using. It seems to be using banks in the DAW, not on the Arduino.

The library has no out-of-the box solution, and I don't have time right now to write an entire program for you.

You can try asking a question on the Arduino forum, but you'll have to provide much more information about the exact behavior of the device, and about the MIDI implementation of the DAW you're using.

benwadub, 2020-02-15:
If I can help you he just switch + or - one octave to his buttons when he use his button.
So you should use bank for button and find a way for bank1 play note c1 to a#1 and second bank play c2 to a#2 by exemple