tttapa/Control-Surface #121: How to restrict value of midi cc message?

kentforth, 2020-02-15:
Hello Peiter!
Let's say I have a potentiometer. It sends  midi cc 56 on channel  1. When I turn a pot it sends values 0-127
**How can I restrict those values?**
For example: a pot should send values from 0 to 90, or values from 15 to 70?
**Can you give me a simple sketch?**

tttapa, 2020-02-15:
You can use the `map` function: <https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d26/CCPotentiometer-Map_8ino-example.html>

```cpp
// The value to send when the potentiometer is in the 0% position
constexpr analog_t minimumValue = 15 * 128;
// The value to send when the potentiometer is in the 100% position
constexpr analog_t maximumValue = 70 * 128;
 
analog_t mappingFunction(analog_t raw) {
    // map the value from [0, 16383] to [minimumValue, maximumValue]
    return map(raw, 0, 16383, minimumValue, maximumValue);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}
```

kentforth, 2020-02-16:
This code works but in a way that I don't want to

Let's say minimumValue is 0 and maximum Value is 70

This is what I get when I turn a pot:

When I turn a pot from 0 to 70% it shows midi cc value 0-70,  but I still can turn potentiometer to right side(to make it 100% right)

[image]

What I want to do: when I turn a pot to  another side( I mean 100% right) it should send midi cc value 70

[image]

tttapa, 2020-02-16:
Could you post the exact code you tried?

kentforth, 2020-02-16:
```

#include <Control_Surface.h>
 
USBMIDI_Interface midi;

CCPotentiometer potentiometer = {
  A0, {MIDI_CC::Channel_Volume, CHANNEL_1}
};
 

constexpr analog_t minimumValue = 1 * 128;
constexpr analog_t maximumValue = 70 * 128;
 

analog_t mappingFunction(analog_t raw) {
   
    raw = constrain(raw, minimumValue, maximumValue);  
    return map(raw, minimumValue, maximumValue, 0, 16383);
  
}
 
void setup() {
     
    potentiometer.map(mappingFunction);   
    Control_Surface.begin();
    Serial.begin(115200);
}
 
void loop() {
    Control_Surface.loop();
    Serial.println(potentiometer.getRawValue());
}
```

tttapa, 2020-02-16:
You have to use the mapping function in my first reply, not the one from the example.

kentforth, 2020-02-16:
This line of code:
potentiometer.map(mappingFunction);

Isn't it?

tttapa, 2020-02-16:
No, you have to change the actual function definition of `mappingFunction`

```cpp
analog_t mappingFunction(analog_t raw) {
 // ...
}
```

kentforth, 2020-02-16:
What should I write in this function definition?

tttapa, 2020-02-16:
The code I posted in my first reply.

kentforth, 2020-02-16:
But I used your code in this sketch O-o

tttapa, 2020-02-16:
No, the mapping function is different, the arguments to the `map` function are the other way around.

```cpp
// Include the Control Surface library
#include <Control_Surface.h>

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Create a new instance of the class `CCPotentiometer`, called `potentiometer`,
// on pin A0, that sends MIDI messages with controller 7 (channel volume)
// on channel 1.
CCPotentiometer potentiometer = {
  A0, {MIDI_CC::Channel_Volume, CHANNEL_1}
};

// The value to send when the potentiometer is in the 0% position
constexpr analog_t minimumValue = 15 * 128;
// The value to send when the potentiometer is in the 100% position
constexpr analog_t maximumValue = 70 * 128;
 
analog_t mappingFunction(analog_t raw) {
    // map the value from [0, 16383] to [minimumValue, maximumValue]
    return map(raw, 0, 16383, minimumValue, maximumValue);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

void setup() {
    // Add the mapping function to the potentiometer
    potentiometer.map(mappingFunction);
    // Initialize everything
    Control_Surface.begin();
    Serial.begin(115200);
}

void loop() {
    // Update the Control Surface (check whether the potentiometer's
    // input has changed since last time, if so, send the new value over MIDI).
    Control_Surface.loop();
    // Use this to find minimumValue and maximumValue: it prints the raw value
    // of the potentiometer, without the mapping function
    Serial.println(potentiometer.getRawValue());
}
```

closed 2020-02-16 by kentforth