tttapa/Control-Surface #125: MIDI CC number of channel limitation?

Red1986z31, 2020-02-18:
CD74HC4051 mux1={24, { 2, 3, 4 }};
CD74HC4051 mux2={25, { 2, 3, 4 }};
CD74HC4051 mux3={26, { 2, 3, 4 }};
CD74HC4051 mux4={27, { 2, 3, 4 }};
CD74HC4051 mux5={28, { 2, 3, 4 }};
CD74HC4051 mux6={A18, { 2, 3, 4 }};
CD74HC4051 mux7={A19, { 2, 3, 4 }};
CD74HC4051 mux8={A20, { 2, 3, 4 }};

CCPotentiometer volumePotentiometers[] = {
  {mux6.pin(0), {MIDI_CC::Channel_Volume, CHANNEL_1}},  //RIGHT PITCH
  {mux6.pin(1), {MIDI_CC::Channel_Volume, CHANNEL_2}}, //FX 2 A GAIN
  {mux6.pin(2), {MIDI_CC::Channel_Volume, CHANNEL_3}}, //FX 1 MIX
  {mux6.pin(3), {MIDI_CC::Channel_Volume, CHANNEL_4}}, //FX 1 B GAIN
  //{mux6.pin(4), {MIDI_CC::Channel_Volume, CHANNEL_5}}, //OPEN
  //{mux6.pin(5), {MIDI_CC::Channel_Volume, CHANNEL_6}},  //OPEN
  {mux6.pin(6), {MIDI_CC::Channel_Volume, CHANNEL_7}},  //LEFT MID
  //{mux6.pin(7), {MIDI_CC::Channel_Volume, CHANNEL_8}},  //OPEN

  {mux7.pin(0), {MIDI_CC::Channel_Volume, CHANNEL_9}},  //LEFT HIGH
  {mux7.pin(1), {MIDI_CC::Channel_Volume, CHANNEL_10}}, // FX 2 B GAIN
  {mux7.pin(2), {MIDI_CC::Channel_Volume, CHANNEL_11}}, //LEFT PITCH
  {mux7.pin(3), {MIDI_CC::Channel_Volume, CHANNEL_12}}, //X FADER
  {mux7.pin(4), {MIDI_CC::Channel_Volume, CHANNEL_13}}, //LEFT FADER
  {mux7.pin(5), {MIDI_CC::Channel_Volume, CHANNEL_14}},  // FX 2 MIX
  {mux7.pin(6), {MIDI_CC::Channel_Volume, CHANNEL_15}},  //LEFT GAIN
  {mux7.pin(7), {MIDI_CC::Channel_Volume, CHANNEL_16}}, //LEFT LOW

  {mux8.pin(0), {MIDI_CC::Channel_Volume, CHANNEL_17}}, //PHONE VOLUME
  {mux8.pin(1), {MIDI_CC::Channel_Volume, CHANNEL_18}}, //PHONE MIX
  {mux8.pin(2), {MIDI_CC::Channel_Volume, CHANNEL_19}}, // RIGHT HIGH
  {mux8.pin(3), {MIDI_CC::Channel_Volume, CHANNEL_20}}, //RIGHT LOW
  {mux8.pin(4), {MIDI_CC::Channel_Volume, CHANNEL_21}},  //RIGHT GAIN
  {mux8.pin(5), {MIDI_CC::Channel_Volume, CHANNEL_22}},  //RIGHT MID
  {mux8.pin(6), {MIDI_CC::Channel_Volume, CHANNEL_23}},  // FX 1 A GAIN
  {mux8.pin(7), {MIDI_CC::Channel_Volume, CHANNEL_24}},  //RIGHT FADER
};

This is just part of my code, but channels 17-24 do not seem to work properly.  
I was working with the Midi_Controller library and ran into the issue where channels 17-24 were just copying 1-8 or 9-16 if i tried to number them higher. 
I thought i would upgrade to the Control_surface library anyway to see if the problem would correct itself, but it did not. 
I should note that i am also having issues with some buttons as well. Most of them work, but a few of them just copy another channel. When i say copy i mean they output the same midi note in the midi debug. I also tried to run it on some other software and confirmed that it's a duplicate note output. 
I also tested the circuit side and  there are no problems there, i also tried swapping the actual CD74HC4051 chip around to see if the problem would follow it, but it did not. 

In total i am running 8 multiplexers, 3 analog and 5 digital. 21 potentiometers and 36 buttons, 8 encoders, two touch inputs. Everything mostly works, except for these duplicate notes being produced. 
I tested my circuit multiple times and it's not a hardware issue. 

```

leshadave, 2020-02-18:
If you had even studied a little MIDI, you would have understood that, in fact, nothing exists from Channel 17. Since midi has only 16 channels.

Red1986z31, 2020-02-18:
Is it not possible to have more than 16 potentiometers then?

benwadub, 2020-02-18:
You can but you ll have to use different midi cc note different midi channel only

leshadave, 2020-02-18:
> Is it not possible to have more than 16 potentiometers then?

you can have more than 16 potentiometers, in one channel 127 midi СС, respectively, you can install 127 channels in one channel somewhere in this area, you have 16 channels and consider what you need and how much. just need to distribute everything on the channels.

Red1986z31, 2020-02-19:
Got it! thank you very much for your help!

Red1986z31, 2020-02-19:
just wanted to add that i called a few of these "Pan" instead of "Channel_Volume", fixed it right up

closed 2020-02-19 by Red1986z31

tttapa, 2020-02-25:
You don't have to use the predefined names, you can also use numbers. An overview of the different MIDI addresses can be found [**here**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/dc6/md_pages_Basics.html).