tttapa/Control-Surface #127: Midi Schematic for making pcb

benwadub, 2020-02-20:
hi, do you know where I can find a clear schematic of midi in out and thru to write it in eagle please?

IsThisNameGoodEnough, 2020-02-25:
You can access the MIDI hardware standard for free here: https://www.midi.org/specifications-old/item/midi-din-electrical-specification

The standard has all the info you need to create the schematic.

benwadub, 2020-02-28:
> You can access the MIDI hardware standard for free here: https://www.midi.org/specifications-old/item/midi-din-electrical-specification
> 
> The standard has all the info you need to create the schematic.

thanks I lll look at it, hope I ll find the value of resistor I need for teensy 3.5 midi in out thru

benwadub, 2020-02-28:
[midi schematic.pdf](https://github.com/tttapa/Control-Surface/files/4266725/midi.schematic.pdf)
could you confirm that the symbol on schematic can be a 74hc14E?
and for teensy 3.5 with 3.3v
that Re is 33 ohm
and rf 10 ohm?
Rd and RB are 220 ohm
electronic is really new for me and English not my mother language! thanks for help!

tttapa, 2020-02-28:
I don't see anything on the pdf you posted, but I presume you're talking about the triangular symbol. It's a buffer. 

A 74HC14 contains six inverters, you could connect two inverters in series to get a non-inverting buffer.  
You don't really need it, as the Arduino's UART has an output buffer, but it might not be a bad idea to use an external buffer, especially at 3.3V, since Rc = 10Ω isn't going to limit the current enough to save the Arduino when you accidentally short the output, by using the wrong cable, for instance.

You could also decide to use a 5V buffer, and use Rc = 220Ω to protect the output (the current will be limited to ~23 mA when you short the output to ground).

benwadub, 2020-03-17:
hi i havent seen you replied here sorry! so you think I could remove the 74hc14? and with the teensy 3.5 would it work without it? I have to build a second MIDI Thru interface and doesn't have a 74hc14 here. do you have a buffer ref that could be used in place of it? I have some component laying around

tttapa, 2020-03-17:
The MIDI baud rate is just 31250 symbols/s, so almost any buffer will do. You can even use an NPN transistor for the second inverter, as proposed by the MIDI standard:

[image]

Leaving the buffer out completely is at your own risk. The MIDI cable goes out to the outside world, so it's not impossible for it to receive a static discharge, get shorted out, etc.

IsThisNameGoodEnough, 2020-03-17:
Hi benwadub.  As ttapa mentioned you don't actually need the buffer for it to work with the teensy, but if you accidentally short the DIN TX pin to ground then you'll probably burn out the pin on the teensy you're using for the DIN-MIDI output.  The buffer will give you some insurance on not damaging your board in that instance.

If you do decide to include a buffer you have three options:

- Use a non-inverting buffer like a 74LVC1G07 ($0.26 [HERE](https://www.digikey.com/product-detail/en/nexperia-usa-inc/74LVC1G07GW-125/1727-3099-1-ND/946756))
- Stack two inverting buffers like tttapa mentioned with the 74hc14
- Use an op-amp in a buffer configuration like [this example](https://www.eecs.tufts.edu/~dsculley/tutorial/opamps/opamps5.html).  Just make sure the gain bandwidth of the op-amp is fast enough that it doesn't start to round over the MIDI signal (2+ MHz should be fine).

If it's helpful I attached a snapshot of a schematic I put together for a project using both a DIN-IN and DIN-OUT.  Each part name can be searched on digi-key.

[image]

benwadub, 2020-03-17:
thank you! I have nothing of this here, and don t know how to solder cms component, I ll order again a 74hc14 that worked the first time, really thank you for reply guys!

benwadub, 2020-03-18:
does an mc74hc04 would do the job? I have it in an old Thomson Mo6 that I could desolder.

tttapa, 2020-03-18:
You can probably use it, but I wouldn't sacrifice a vintage computer to get a single chip that costs literally a couple of cents new (on Conrad, for example).

IsThisNameGoodEnough, 2020-04-16:
Just in case anyone reads through this thread looking for a simple schematic, here's an updated version that I've confirmed works well.  The ICs called out in the image are for surface mount variants, but they also come in through hole as well.  As for the diode, a good through hole candidate is P/N BAW76-TR, although any general purpose diode should work.  Hope it helps someone.

[image]