tttapa/Control-Surface #128: Out of Sync Midi Clock

ransalman, 2020-02-24:
```cpp
#include <Control_Surface.h> // Include the Control Surface library.
using namespace MIDI_Notes;
SoftwareSerial OpenLCD(34, 33); //RX, TX
#include <Adafruit_NeoPixel.h>

bool realTimeMessageCallback(uint8_t message) {
    if (logging) Serial << "Real-Time message: " << hex << message << dec << endl;
    RealTimeSystem(message); // Parse and Handle Realtime Message
    return true;  // Return true to indicate that handling is done, 
                  // and Control_Surface shouldn't handle it anymore.
                  // If you want Control_Surface to handle it as well, 
                  // return false.
}
void RealTimeSystem(byte realtimebyte) {
  if (realtimebyte == CLOCK) { // If message == clock
    breath = false;
    stepSeq();
  }
  if (realtimebyte == START || realtimebyte == CONTINUE) {
    breath = false;
    seqCount = 0;
    currentStep = 0;
    turnOffAllSEQLeds(); // Turn Off Previous Sequencer LEDs
  }

  if (realtimebyte == STOP) {
    breath = true;
    seqCount = 0;
    currentStep = 0;
    turnOffAllSEQLeds;  
  }
}

```
I have a step sequencer I built using the library old commit where everything was fine (cd08dea14772fa4c57ee0d39d764ae19b8c44f2c) 
I'm getting a midi clock from Ableton live to Teensy 3.6 and sending midi notes from the Teensy to Ableton in time...
Everything was perfect till I updated to the current master, it really breaks the timing it is drifting and can't be used.

I even checked older commits such as "6eff8a55389ed0b8788b06373fcc27d6031c48d5".
and it also breaks the sync...

Can you please advise how to fix this?
Code is very big so I only attached some of it.

ransalman, 2020-02-24:
it looks like the board is getting a delay sometimes as I logged the midi clock beat interval in the serial monitor.
also noticed that tweaking certain knobs also affects heavily the time

tttapa, 2020-02-25:
cd08dea was a long time ago, I don't know which change could have caused it.

Could you try the most basic MIDI Real-Time input sketch possible, and then incrementally add the other parts of the sequencer to isolate what part of the library causes the drift?

Could you also be more specific about the drift? How did you measure it, how does it manifest itself? Is it just a delay, or does it actually drift and increase over time?

ransalman, 2020-02-26:
I will try  incrementally adding the code.
As for the drift - You can hear it and I also printed to serial the beat interval in ms.
But you definitely feel it getting out of tempo with the DAW.. this is the problem.
Most of the time it just getting out of sync after a few minutes or less like something is delaying it in the code (but no delay in my code).
Sometimes it's just breaking the sync like there is a big gap and all the sync is messed up.
I tested the old library (cd08dea) a lot and it was super stable - I even let it run for an entire day and it was right on time.

Also - a big problem is when tweaking some knobs really fast, the whole clock is messed up.
I use 2 x 4 CD74HC4067 MUXes chained but for sure this didn't happen in the old commit.... 

So I think it something changed in the library or a breaking change that I need to change in my code but I reviewed and didn't find anything I need to change.

Hope this description makes it brighter for you about the issue, I will try incrementally adding my code and report..

Thanks

tttapa, 2020-02-26:
I did some digging, and one significant thing that has changed is the following:

In the old version you were using, the all available MIDI input was read at once:
https://github.com/tttapa/Control-Surface/blob/cd08dea14772fa4c57ee0d39d764ae19b8c44f2c/src/MIDI_Interfaces/MIDI_Interface.cpp#L23-L27

In the new version, it reads just one message on each iteration.

https://github.com/tttapa/Control-Surface/blob/c434456be9c91b1a229c558f214be5abceebf966/src/MIDI_Interfaces/MIDI_Interface.cpp#L154-L160

That being said, this didn't change in the second commit you mentioned, so it might not be the only reason.

ransalman, 2020-03-13:
I've done some cleanup in my code, changed the way I'm talking to the lcd and now I'm on of the latests commits 61ccbbe2  with no problems.
Tried to review the older code but I can't find what was causing the issue
You can close it, thanks for the help :-)

tttapa, 2020-03-13:
Glad to hear!

closed 2020-03-13 by tttapa