tttapa/Control-Surface #129: Possible to instantiate inputs based on external triggers?

IsThisNameGoodEnough, 2020-02-25:
First off, thank you for all the work you've done.  The more I dig into your software the more impressive it is.

I have a question about instantiating inputs.  I can see two scenarios that could have a lot of value:

- Having the software pull input configuration info from non-volatile memory when instantiating said inputs.  An example would be using an ESP32 and storing the following info for each I/O pin in flash: type of I/O (e.g. button, pots, etc), mapping, channel.
- Having the software instantiate an I/O pin based on the state of another pin.  An example would be a toggle switch which would determine whether a bank of I/Os are instantiated as buttons or pots.

I believe each one of these examples require instantiating I/Os after entering the setup function.  I tried doing this but couldn't get it working.  Is there a way to do this, or another approach I'm not seeing?  Thanks for the help!

tttapa, 2020-02-25:
The design philosophy of the library is to specify the hardware configuration in your sketch, and then upload your sketch to the board.  
The hardware itself usually doesn't change at run-time, so neither should the hardware configuration in you program.

That being said, the majority of the library's components is self-registering. This means that when you create a new MIDI Output Element, for example, the library automatically picks it up, and starts updating it.

If you create a new Element before you call `Control_Surface.begin()` in your setup, the library will pick it up and initialize it for you as well.  
If you create a new Element _after_ the call to `Control_Surface.begin()`, you have to initialize it yourself by calling the `element.begin()` method.

For example:

```cpp
#include <Control_Surface.h> // Include the Control Surface library

#include <AH/STL/memory>

// Instantiate a MIDI over USB interface.
USBDebugMIDI_Interface midi;

constexpr size_t NUM_ELEM = 4;
std::unique_ptr<Updatable<>> elements[4];
Button buttons[NUM_ELEM] = {2, 3, 5, 7};
pin_t analogPins[NUM_ELEM] = {A0, A1, A2, A3};
MIDIAddress addresses[NUM_ELEM] = {0, 1, 2, 3};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
  for (Button &button : buttons)
    button.begin();
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface

  for (size_t i = 0; i < NUM_ELEM; ++i) {
    auto buttonState = buttons[i].update();
    if (buttonState == Button::Falling) {
      // Create a new CCPotentiometer
      elements[i] = make_unique<CCPotentiometer>(analogPins[i], addresses[i]);
      elements[i]->begin(); // initialize
    } else if (buttonState == Button::Rising) {
      // Delete the CCPotentiometer
      elements[i].reset();
    }
  }
}
```

IsThisNameGoodEnough, 2020-02-28:
Thanks for the response.  I tried making the simplest test case to see if it would work and ran into an issue.  When I run the following it works as intended, and I see the CC message on the receiving side:

`#include <Control_Surface.h>
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp>

BluetoothMIDI_Interface midi;

CCButton button = { 0, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}, };

void setup() { 
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}`

When I move instantiating the button to within the setup function but before Control_Surface is called I do not receive a CC message on the receiving side when the button is pressed:

`#include <Control_Surface.h>
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp>

BluetoothMIDI_Interface midi;

void setup() { 
  CCButton button = { 0, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}, };
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}`

Software isn't my background, so any tips would be greatly appreciated.

tttapa, 2020-02-28:
You cannot declare the button inside of the setup function, it's a local variable with automatic storage duration, its lifetime ends at the end of the setup function. 

You have to make the variable global if it has to live through both the setup and loop functions, like in the example I posted previously. By using smart pointers and allocating the button on the heap, you can dynamically allocate, create and initialize them.

tttapa, 2020-02-29:
Alternatively, you could in this case make the button variable static:

```cpp
void setup() { 
  static CCButton button = { 0, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1} };
  Control_Surface.begin();
}
```

IsThisNameGoodEnough, 2020-03-01:
Thanks so much for the explanation.  I should have known the issue was variable scope.  Appreciate the help!

IsThisNameGoodEnough, 2020-04-01:
One more question for you if you don't mind.  I have an ESP32 and am planning to store info on whether inputs are buttons or encoders.  My plan was to initialize the element arrays for encoders and buttons, then pass the arguments to the arrays by reading them from the EEPROM.  This is being done in the setup before running Control_Surface.begin().  After trying a few things, it looks like I need to pass the arguments to the arrays at the time of creating them.  Since the length of the arrays are variable (read from the EEPROM during startup), any suggestions on how to initialize the arrays while passing the necessary arguments?

Trying to do this before Control_Surface.begin() is called as I'm trying to get away from having to update elements in the main loop (I'm not a programmer and so I'm trying to keep it as simple as possible).  The relevant code for my initial attempt is shown below.  When compiling I get an "no matching function for call to 'CS::CCRotaryEncoder::CCRotaryEncoder()'" error.

```cpp
// Determine how many encoders there are so we can define the array size of objects to initialize
int qtyEncoders;
int qtyButtons;

for(int i=0; i<32; i++) {
	if(EEPROM.readBool(address)) {
	address += sizeof(bool);
	qtyEncoders++;
	}
}

qtyButtons = 64 - (2 * qtyEncoders);

// Reset EEPROM address so we can read through the encoder configuration again
address += initString.length() + 1;

// Initialize encoder and button object arrays
if( qtyEncoders =! 0 ) {
	static CCRotaryEncoder encoderArray[qtyEncoders];
}
if( qtyButtons != 0 ){
	static CCButton buttonArray[qtyButtons];
}

// Initialize every digital input based on configuration stored in EEPROM
int encoderArrayIndex = 0;
int buttonArrayIndex = 0;

for(int i=0; i<32; i++){
	int t = i*2;
	if(EEPROM.readBool(address)) {
		address += sizeof(bool);
		encoderArray[encoderArrayIndex] = { { digitalInputPairs[i][1], digitalInputPairs[i][2] }, { t, CHANNEL_1 }, 1, };
		encoderArrayIndex++;
	}
	else {
		address += sizeof(bool);
		buttonArray[buttonArrayIndex] = { digitalInputPairs[i][1], { t, CHANNEL_1}, };
		buttonArrayIndex++;
		buttonArray[buttonArrayIndex] = { digitalInputPairs[i][2], { (t+1), CHANNEL_1}, };
		buttonArrayIndex++;
	}
}
````

Thanks for the help!

tttapa, 2020-04-01:
C++ has no VLAs (variable length arrays). If you want to change the size of an array at runtime, you have to allocate it on the heap.

As you've found, you cannot create arrays of MIDI elements without initializing them. This is normal, because they don't have a default constructor (and such a constructor would semantically ill-defined).

Both of these things mean that dynamically allocating the MIDI elements is the only option. The easiest way to do this is to use `std::vector`:

```cpp
#include <Control_Surface.h>
#include <vector>

std::vector<CCButton> buttons;

USBDebugMIDI_Interface midi;

void setup() {
  size_t num_buttons = 5; // read this from EEPROM
  buttons.reserve(num_buttons);
  for (size_t i = 0; i < num_buttons; ++i) {
    pin_t pin = 0; // read this from EEPROM
    MIDIAddress address = 7; // read this from EEPROM
    buttons.emplace_back(pin, address); // cfr. CCButton button = {pin, address};
  }
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

IsThisNameGoodEnough, 2020-04-01:
Yet again, thanks for pointing me in the right direction.  Very much appreciated.

IsThisNameGoodEnough, 2020-04-02:
Ok, I've got the buttons working based on your example, but I'm running into problems trying to generate the code for dynamically allocating the encoders.  Below is a variant of your example code that I modified for encoders.  When I try running it I get the following error:

> no matching function for call to 'std::vector<CS::CCRotaryEncoder>::emplace_back(<brace-enclosed initializer list>, CS::MIDIAddress&)'

```cpp
#include <Encoder.h>
#include <Control_Surface.h>
#include <vector>

std::vector<CCRotaryEncoder> encoders;

USBDebugMIDI_Interface midi;

void setup() {
  size_t num_encoders = 5;
  encoders.reserve(num_encoders);
  for (size_t i = 0; i < num_encoders; ++i) {
    pin_t pin1 = 0;
    pin_t pin2 = 1;
    MIDIAddress address = 7;
    encoders.emplace_back({pin1, pin2}, address);
  }
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

Any thoughts on what I'm missing?

tttapa, 2020-04-02:
The easiest solution is to just prepare all arguments beforehand, without using brace-initializer lists in the call to `emplace_back`.

```cpp
  size_t num_encoders = 1; // read this from EEPROM
  buttons.reserve(num_encoders);
  for (size_t i = 0; i < num_encoders; ++i) {
    EncoderPinList pins = {0, 1}; // read this from EEPROM
    MIDIAddress address = 7; // read this from EEPROM
    encoders.emplace_back(pins, address);
  }
```

You can find the arguments you need in the [documentation](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d62/classCCRotaryEncoder.html#aceb83976568f994c61b95324cdc4aadc).

IsThisNameGoodEnough, 2020-04-02:
Ah, I figured I could just pass the list directly.  Easy enough change, and it looks like everything's working now.  Thanks again for all the help!

closed 2020-04-02 by IsThisNameGoodEnough

IsThisNameGoodEnough, 2020-05-08:
Hope you don't mind, but I've got another question for you.  Using `std::vector` for dynamically allocating arrays works well on the ESP32.  I've started playing with the Arduino Leonardo in order to gain MIDI USB support, and it looks like AVR doesn't natively support the vector library.  Do you have any suggestions for implementing dynamic arrays using a Leonardo?

Another option could be to create the array of buttons and encoders with the maximum number that either could have, but this would result in a portion of each array being left blank.  I haven't tried this yet, but are there any issues you'd foresee with this approach?

tttapa, 2020-05-08:
Control Surface includes an `std::vector` implementation, you can just `#include <AH/STL/vector>`. On boards that include the STL, it uses the toolchain's implementation, on AVR, it uses its own implementation.

You have to be very careful to prevent heap fragmentation. Constantly resizing vectors can be disastrous. If at all possible, allocate memory once in the setup using `std::vector::reserve`. 

You cannot create blank arrays of objects without a default constructor. This includes most of Control Surface's classes.

IsThisNameGoodEnough, 2020-05-08:
Wow, that makes my life a lot easier than I was expecting.  Thanks for the very thorough implementation.

Yep, I was reading up on the issues with heap fragmentation (especially with the Leonardo's 2kb of ram).  I've been using 'vector::reserve' and don't touch the array after I do the initial population, and it seems to work well on the ESP32.  Guess I'll find out soon enough on the Leonardo.