tttapa/Control-Surface #130: Bank explanation please

Red1986z31, 2020-02-28:
Not an issue per say, just curious how the Bank function works. If you have four push buttons, and you have all four of them in a 4 way bank ( 16 total midi notes), How does the Bank re assign the notes? 
Just curious which way it shifts the notes around... I have a lot of buttons and banks and am running into an issue where some notes are being duplicated. Not too big of a deal as i just go back in there and give the original input a different note to start on, seems to fix things.

tttapa, 2020-02-28:
Most of it should be explained in the [Bank.ino example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d74/Bank_8ino-example.html).  
If you have any further questions, just let me know.

Red1986z31, 2020-02-29:
I guess I’m more curious if a particular button was set to A0 for example, and it’s in a bank what note does the bank push it up to?

tttapa, 2020-02-29:
To `A0 + bank_setting * tracks_per_bank`.

Red1986z31, 2020-03-05:
Thank you for the better explication.
I’m still having some duplicate note issues that I am trying to overcome and the transposer example looks like it may fix things. 
Thanks again!

closed 2020-03-05 by Red1986z31