tttapa/Control-Surface #131: teensy 3.5 send values in debug midi usb without anything plugged on board

benwadub, 2020-02-29:
hi, i have a problem with my teensy, can t make it work with midi din so I tied debug usb and the teensy seems to send random cc, do you know where the problem can come from?
```cpp

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
 USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

//auto &serial = Serial1;// Selectionne le port série à utiliser

//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
//};// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

 
//using namespace MIDI_Notes;  ces lignes midi servent à activer un bouton qui envoit une note midi enlever les // sur ces 5 lignes pour l'activer

// démarrer une touche midi
//NoteButton button = {
  //5,                       // numéro de l'entrée arduino sur laquelle le bouton est branché
  //{note(C, 2), CHANNEL_5}, // note C4 sur le canal 5 libre à vous de changer les numéro et d'en ajouter

// active un multiplexeur (une sorte de multiprise sur laquelle vous branchez 8 controleurs sur 1 seule entrée de l'arduino
//};
CD74HC4067 mux1 = {
  7,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};
Bank<2> bank2 = {8}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched buttonmute[] = { //ces lignes déclarent le mux
  {{bank2,BankType::CHANGE_CHANNEL}, mux1.pin(0), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(1), {94, CHANNEL_2}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(2), {94, CHANNEL_3}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(3), {94, CHANNEL_4}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(4), {94, CHANNEL_5}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(5), {94, CHANNEL_6}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(6), {94, CHANNEL_7}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(7), {94, CHANNEL_8}},
};
Bank <2> bank3 = {8}; // 2 banks, 8 addresse per banks

Bank<2> bank1 = {-8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  8,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(12), {82, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(13), {82, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(14), {82, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(15), {82, CHANNEL_8}},
};

  //Instantiate second multiplexer for 8 potentiometer
CD74HC4067 mux3 = {
  9,       // Analog input pin
  {2, 3, 4, 5} // Address pins S0, S1, S2};

};
Bankable::CCPotentiometer fader [] = { //ici on déclare les faders avec comme vu plus haut le numéro de CC et le canal
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(0), {7, CHANNEL_1}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(1), {7, CHANNEL_2}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(2), {7, CHANNEL_3}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(3), {7, CHANNEL_4}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(4), {7, CHANNEL_5}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(5), {7, CHANNEL_6}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(6), {7, CHANNEL_7}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(7), {7, CHANNEL_8}},

};
 
//CD74HC4067 mux4 = {
  //A3,       // numéro de broche de l'arduino
  //{2, 3, 4} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
//};

//CCAbsoluteEncoder encoder [16] = {
  //{{6, 7}, {85, CHANNEL_9},16, 8,},
  //{{8, 9}, {86, CHANNEL_9}, 16, 4,},
  //{{10, 11}, {87, CHANNEL_9}, 16, 4,},
  //{{12, 13}, {88, CHANNEL_9}, 1, 4,},
  //{{14, 15}, {89, CHANNEL_9}, 1, 4,},
  //{{16, 17}, {90, CHANNEL_9}, 1, 4,},
  //{{18, 19}, {91, CHANNEL_9}, 16, 4,},
  
  //{{20, 21}, {92, CHANNEL_9}, 1, 4,},
  //{{22, 23}, {24, CHANNEL_9}, 1, 4,},
  //{{24, 25}, {25, CHANNEL_9}, 1, 4,},
  //{{26, 27}, {26, CHANNEL_9}, 1, 4,},
  //{{28, 29}, {27, CHANNEL_9}, 1, 4,},
  //{{30, 31}, {28, CHANNEL_9}, 1, 4,},
  //{{32, 33}, {29, CHANNEL_9}, 1, 4,},
  //{{34, 35}, {30, CHANNEL_9}, 1, 4,},
//  {{36, 37}, {31, CHANNEL_9}, 1, 4,},

//};
//IncrementSelector<2> selectorbutton = { //déclare les boutons de changement de bank
  //bank2, // banque sélectionnée
//   mux4.pin(0), // pins sur lequel les boutons sont branchés
//};
//IncrementSelector<2> selectorbutton2 = {
  //bank1, // banque sélectionnée
  //mux4.pin(1),  // adresse du bouton
//};
//IncrementSelector<2> selectorvolume = {
  //bank3, // banque sélectionnée
  //mux4.pin(2),  
  
//};
void setup() {
  Control_Surface.begin(); // initialise la library surface de control

 

}                           
void loop() {
  

  Control_Surface.loop(); // Update the Control Surface

}

```

tttapa, 2020-02-29:
It's probably just a loose wire, or an incorrect connection.

What messages does it send?

benwadub, 2020-02-29:
it send midi message even if nothing is plugged to the teensy!!!
i have this kind of message:
Control Change  	Channel: 3	Data 1: 0x07	Data 2: 0x16	Cable: 0
Control Change  	Channel: 5	Data 1: 0x07	Data 2: 0x15	Cable: 0
Control Change  	Channel: 6	Data 1: 0x07	Data 2: 0x17	Cable: 0
Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x13	Cable: 0
Control Change  	Channel: 8	Data 1: 0x07	Data 2: 0x14	Cable: 0
Control Change  	Channel: 1	Data 1: 0x07	Data 2: 0x1d	Cable: 0
Control Change  	Channel: 2	Data 1: 0x07	Data 2: 0x1c	Cable: 0
Control Change  	Channel: 3	Data 1: 0x07	Data 2: 0x13	Cable: 0
Control Change  	Channel: 4	Data 1: 0x07	Data 2: 0x1a	Cable: 0
Control Change  	Channel: 5	Data 1: 0x07	Data 2: 0x13	Cable: 0
Control Change  	Channel: 6	Data 1: 0x07	Data 2: 0x13	Cable: 0
Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x0e	Cable: 0
Control Change  	Channel: 8	Data 1: 0x07	Data 2: 0x0f	Cable: 0
Control Change  	Channel: 1	Data 1: 0x07	Data 2: 0x1f	Cable: 0

tttapa, 2020-02-29:
> it send midi message even if nothing is plugged to the teensy!!!

There's your problem. If you leave the analog input pins floating, you'll pick up random noise.

benwadub, 2020-02-29:
I didnthad this problem yesterday evening when you told me to change to serial1 in the midi interface, floating mean unused? is there a way to stop them with code?

tttapa, 2020-02-29:
Floating means that the input is not held at a specific voltage, because nothing is connected to it.

You have to remove it from the code if you don't connect anything to a given pin.

benwadub, 2020-02-29:
thanks I ll try it but for the moment nothing is working :-( no midi output at all! really don t understand what happen!

benwadub, 2020-02-29:
It seems that I have a bigger problem, my Mac send serial things on all my arduino boards, Rex is flashing when I plug any board! And my mega doesn’t show anymore in port in ide! Don t know how to fix all this 🤯

benwadub, 2020-03-02:
hi Pieter, hope you re fine?
i have been able to make work my controller again, now when all is wired all work fine except my faders that seems to have noise and send cc like this:
21:18:49.157 -> Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x28	Cable: 0
21:18:49.157 -> Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x26	Cable: 0
21:18:49.191 -> Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x25	Cable: 0
21:18:49.191 -> Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x24	Cable: 0
21:18:49.191 -> Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x23	Cable: 0
21:18:49.227 -> Control Change  	Channel: 3	Data 1: 0x07	Data 2: 0x21	Cable: 0
21:18:49.227 -> Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x22	Cable: 0
21:18:49.264 -> Control Change  	Channel: 7	Data 1: 0x07	Data 2: 0x21	Cable: 0
21:18:49.302 -> Control Change  	Channel: 2	Data 1: 0x53	Data 2: 0x25	Cable: 0
21:18:49.613 -> Control Change  	Channel: 5	Data 1: 0x52	Data 2: 0x1e	Cable: 0
i doubled check my wires and soldering, all seems to be fine, do you know where I could check now?
'''

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
 USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

//auto &serial = Serial;// Selectionne le port série à utiliser

//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD
//};// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

 
//using namespace MIDI_Notes;  ces lignes midi servent à activer un bouton qui envoit une note midi enlever les // sur ces 5 lignes pour l'activer

// démarrer une touche midi
//NoteButton button = {
  //5,                       // numéro de l'entrée arduino sur laquelle le bouton est branché
  //{note(C, 2), CHANNEL_5}, // note C4 sur le canal 5 libre à vous de changer les numéro et d'en ajouter
//};
CD74HC4067 mux1 = {
  A2,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};
Bank<2> bank2 = {8}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched buttonmute[] = { //ces lignes déclarent le mux
  {{bank2,BankType::CHANGE_CHANNEL}, mux1.pin(0), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(1), {94, CHANNEL_2}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(2), {94, CHANNEL_3}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(3), {94, CHANNEL_4}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(4), {94, CHANNEL_5}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(5), {94, CHANNEL_6}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(6), {94, CHANNEL_7}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(7), {94, CHANNEL_8}},
};

Bank<2> bank1 = {-8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  A1,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(12), {82, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(13), {82, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(14), {82, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(15), {82, CHANNEL_8}},
};

Bank <2> bank3 = {8}; // 2 banks, 8 addresse per banks

  //Instantiate second multiplexer for 8 potentiometer
CD74HC4067 mux3 = {
  A0
  ,       // Analog input pin
  {2, 3, 4, 5} // Address pins S0, S1, S2};

};
Bankable::CCPotentiometer fader [] = { //ici on déclare les faders avec comme vu plus haut le numéro de CC et le canal
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(0), {7, CHANNEL_1}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(1), {7, CHANNEL_2}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(2), {7, CHANNEL_3}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(3), {7, CHANNEL_4}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(4), {7, CHANNEL_5}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(5), {7, CHANNEL_6}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(6), {7, CHANNEL_7}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(7), {7, CHANNEL_8}},

 
  
};

void setup() {
  Control_Surface.begin(); // initialise la library surface de control

 

}                           
void loop() {
  

  Control_Surface.loop(); // Update the Control Surface

}
'''
thanks

benwadub, 2020-03-02:
Found the problem! A bad solder on a mux caused all this! Now all work fine on arduino uno!
Now I have to make it work on teensy to add my 16 encoders but I can’t send midi, I used serial1 in code and changed the resistors, can I still call the pin A0 A1 and A2 for the teensy?
And still use pins 2,3,4 and 5 for mux?

tttapa, 2020-03-03:
Does the debug MIDI interface work?  
How did you connect the MIDI port?  
You should have received a pinout card with your Teensy, it tells you which pins are A0, A1, etc. Why wouldn't you be able to use pins 2, 3, 4 and 5 for the mux?

benwadub, 2020-03-03:
No debug midi doesn’t work :-( I connected the midi port by wiring midi pin 5 to 3,3v through 47ohm resistor.
Midi pin 2 to ground and midi pin 4 to pin 1 on teensy (tx)
I have the pin out card for teensy. A0 is pin 14 A2 is pin 15, i didn’t knew if in the sketch I have to name them A0 or 14.
Which usb mode must I use for teensy? Serial? Serial and midi...?

tttapa, 2020-03-03:
Please post your code.

A0 is equivalent to pin 14.

The USB mode doesn't matter if you're not using MIDI over USB.

benwadub, 2020-03-03:
here is the code thanks
``

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
 //USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

auto &serial = Serial1;// Selectionne le port série à utiliser
USBMIDI_Interface usbmidi;
HardwareSerialMIDI_Interface serialmidi = {Serial1, MIDI_BAUD};
MultiMIDI_Interface<2> midi = {{&usbmidi, &serialmidi}};
//};// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

 
//using namespace MIDI_Notes;  ces lignes midi servent à activer un bouton qui envoit une note midi enlever les // sur ces 5 lignes pour l'activer

// démarrer une touche midi
//NoteButton button = {
  //5,                       // numéro de l'entrée arduino sur laquelle le bouton est branché
  //{note(C, 2), CHANNEL_5}, // note C4 sur le canal 5 libre à vous de changer les numéro et d'en ajouter
//};
CD74HC4067 mux1 = {
  A2,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};
Bank<2> bank2 = {8}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched buttonmute[] = { //ces lignes déclarent le mux
  {{bank2,BankType::CHANGE_CHANNEL}, mux1.pin(0), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(1), {94, CHANNEL_2}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(2), {94, CHANNEL_3}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(3), {94, CHANNEL_4}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(4), {94, CHANNEL_5}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(5), {94, CHANNEL_6}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(6), {94, CHANNEL_7}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(7), {94, CHANNEL_8}},
};

Bank<2> bank1 = {-8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  A1,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(12), {82, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(13), {82, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(14), {82, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(15), {82, CHANNEL_8}},
};

Bank <2> bank3 = {8}; // 2 banks, 8 addresse per banks

  //Instantiate second multiplexer for 8 potentiometer
CD74HC4067 mux3 = {
  A0
  ,       // Analog input pin
  {2, 3, 4, 5} // Address pins S0, S1, S2};

};
Bankable::CCPotentiometer fader [] = { //ici on déclare les faders avec comme vu plus haut le numéro de CC et le canal
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(0), {7, CHANNEL_1}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(1), {7, CHANNEL_2}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(2), {7, CHANNEL_3}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(3), {7, CHANNEL_4}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(4), {7, CHANNEL_5}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(5), {7, CHANNEL_6}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(6), {7, CHANNEL_7}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(7), {7, CHANNEL_8}},

 
  
};

void setup() {
  Control_Surface.begin(); // initialise la library surface de control

 

}                           
void loop() {
  

  Control_Surface.loop(); // Update the Control Surface

}
``

benwadub, 2020-03-03:
do I need to use the vin pin to power something in my project? in my wiring all is driven by 3,3v, maybe something need more?

tttapa, 2020-03-03:
I don't think MultiMIDI_Interface is supported on the current master branch. It should work in the latest stable release.

Try removing all other MIDI interfaces, just keep the `USBDebugMIDI_Interface midi = 115200;` and try opening the serial monitor again.

tttapa, 2020-03-03:
No, you don't need the Vin pin, everything should be 3.3V.

benwadub, 2020-03-03:
Merci ´ si now I get activity in the usb debug midi interface, but lot of noise, I recheck my soldering, I think I burned a pin on a mux cause I soldered and de soldered it few times :-(

benwadub, 2020-03-03:
It s strange I have noise when in the debug usb and nothing run to the din out when I use serial interface and no noise at all with my UNO.
I checked all my wiring in midi module, rebuild another one on breadboard but still nothing :-(

tttapa, 2020-03-03:
You could try connecting the potentiometer grounds to AGND instead of GND.

benwadub, 2020-03-03:
[image]
oh my god on the PJRC site the din is in the wrong way!!! I just tried to invert pin 4 and 5 and it work just fine!!! sorry for all this!!

tttapa, 2020-03-10:
Those DIN schematics can be very confusing indeed, male and female connectors are mirror images of each other, glad to hear you figured it out!

benwadub, 2020-03-10:
And I can tell you that your code run nicely with 16 encoder 32 button and 32 pot on teensy 3.5!
Again a big thank to you!
Now I ll try to add some drum Controler with piezo!

tttapa, 2020-03-11:
Good to know! I've never tried using this many encoders on a single Arduino.

Be careful not to blow up the analog inputs of the Teensy, piezos produce a very high current, which can result in a very high voltage if you're not careful.

benwadub, 2020-03-11:
Thanks, I ll try to find a tutorial to make it or maybe go with sanwa arcade button that seems to be nice!

benwadub, 2020-03-11:
For my drum controller do you think I could use my arduino uno add to the teensy and how can I wire them together? I thought maybe using midi thru on the teensy and midi out on the uno maybe there is another better solution?

tttapa, 2020-03-12:
You could use MIDI through, have a look at the [MIDI Routing example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dc4/MIDI_Pipes-Routing_8ino-example.html).

benwadub, 2020-03-12:
oh my god you are a real midikiller :-) !! thanks!!!!

benwadub, 2020-03-12:
and do you think there is a way that Logic Pro recognize the midi controller as an MCU in usb midi interface?

benwadub, 2020-03-13:
hi I again need your help, I can t find how to write my array to midi note I tried this and lot of other things but impossible find the good way!
```
using namespace MIDI_Notes;

NoteButton button key [8] = {
                        
  {{5, note(C, 2), CHANNEL_1}},
  {{6, note(C#, 2), CHANNEL_2}},
  {{7, note(D, 2), CHANNEL_3}},
  {{8, note(D#, 2), CHANNEL_4}},
  {{9, note(E, 2), CHANNEL_5}},
  {{10, note(F, 2), CHANNEL_6}},
  {{11, note(F#, 2), CHANNEL_7}},
  {{12, note(G, 2), CHANNEL_8}},
};
```

tttapa, 2020-03-13:
Most DAWs I've tried have a menu where you can tell it what kind of MIDI devices are connected. 

If that doesn't work, it could be detecting it automatically using either the USB VID/PID, or maybe a MIDI SysEx challenge/response, but I seriously doubt that.

Remember that a single NoteButton is initialized like this:
```cpp
// Instantiate a NoteButton object
NoteButton button = {
  5,                       // Push button on pin 5
  {note(C, 4), CHANNEL_1}, // Note C4 on MIDI channel 1
};
```
To make it an array, just add another pair of braces around it:
```cpp
// Instantiate an array of NoteButton objects
NoteButton buttons[] = {
  {
    5,                       // Push button on pin 5
    {note(C, 4), CHANNEL_1}, // Note C4 on MIDI channel 1
  },
  {
    6,                       // Push button on pin 6
    {note(Db, 4), CHANNEL_1}, // Note C#4 on MIDI channel 1
  },
};
```

The note number and the channel are grouped together as the address.

Note that the `#` sign is not a valid character in a C++ identifier, you have to use `Db` instead of `C#`.

benwadub, 2020-03-13:
thanks for all this again!

benwadub, 2020-03-14:
hi, I begin to try the midi routing, if I want to chain an Arduino and a teensy I have to wire my Arduino tx to the teensy rx and my teensy tx to my hardware midi interface I m right?
for the sketch I thought something like this:
HardwareSerialMIDI_Interface midiser = Serial1;
HardwareSerialMIDI_Interface midiser2 = Serial2;
 
MIDI_PipeFactory<3> pipes;
 
void setup() {
  midiser2 >> pipes >> midiser; // all incoming midi from USB is sent to serial
  midiser2 << pipes << midiser; // all incoming midi from Serial is sent to USB
midiser >> pipes >> midiser; // all incoming midi from USB is looped back
 
  midiusb.begin();
  midiser.begin();
}
 
void loop() {
  midiusb.update();
  midiser.update();
}

tttapa, 2020-03-14:
The MIDI routing stuff is all pretty new, so I can't guarantee that everything will work straight out of the box, but I'd try something like this:

```cpp
#include <Control_Surface.h>

HardwareSerialMIDI_Interface hwmidi = Serial1;
HardwareSerialMIDI_Interface unomidi = Serial2;
 
MIDI_Pipe pipe;
 
void setup() {
  // All incoming midi messages from UNO are forwarded to MIDI device:
  unomidi >> pipe >> hwmidi; 
  // Control Surface will use this MIDI interface as the default one:
  hwmidi.setAsDefault();
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
}
```

Connect TX1 of the Teensy to the MIDI input of your MIDI device.  
Connext TX of the Arduino UNO to RX2 of the Teensy.

On the UNO, just use a single `HardwareSerialMIDI_Interface hwmidi = Serial;` without any explicit routing.

***

Since all data is flowing in one direction (i.e. RX1 and TX2 are unused), you could get this working with just a single MIDI Interface by looping it back in software. However, this also means that Control Surface sees all MIDI data from the UNO, which is probably not what you want.

benwadub, 2020-03-14:
I ll try it and tell you if this work!
Thanks!
I begin to have a big project with 16 encoders, 16 pots, 16 buttons and banks, with serial I t’efface, maybe you d line to have it to put in your examples?

benwadub, 2020-03-14:
I get MIDI_pipe' does not name a type, any idea what is going on?

tttapa, 2020-03-14:
Either you didn't spell it correctly (it needs a capital P), or you haven't got the latest version installed.

benwadub, 2020-03-14:
i made a copy/past I certainly need to install the last update! I ll check how to update!

tttapa, 2020-03-14:
If you installed the library using Git, you can just do `git pull`. Otherwise, the easiest way is to delete `~/Arduino/libraries/Control-Surface` and download and install the ZIP again.

benwadub, 2020-03-14:
I tried to install with git but I made a mistake, git pull didn't t work!

tttapa, 2020-03-14:
Were you in the `libraries/Control-Surface` directory when executing `git pull` ?

benwadub, 2020-03-14:
yes but not a problem I deleted control surface and downloaded the new one and it s fine! but now my code doesn't t compile because of button latched, I got this
invalid use of template-name 'CS::Bankable::CCButtonLatched' without an argument list
```

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
 //USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

auto &serial = Serial1;// Selectionne le port série à utiliser
//USBMIDI_Interface usbmidi;
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD

};// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

 
//using namespace MIDI_Notes;  ces lignes midi servent à activer un bouton qui envoit une note midi enlever les // sur ces 5 lignes pour l'activer

// démarrer une touche midi
//NoteButton button = {
  //5,                       // numéro de l'entrée arduino sur laquelle le bouton est branché
  //{note(C, 2), CHANNEL_5}, // note C4 sur le canal 5 libre à vous de changer les numéro et d'en ajouter
//};
CD74HC4067 mux1 = {
  A2,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};
Bank<2> bank2 = {8}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched buttonmute[] = { //ces lignes déclarent le mux
  {{bank2,BankType::CHANGE_CHANNEL}, mux1.pin(0), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(1), {94, CHANNEL_2}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(2), {94, CHANNEL_3}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(3), {94, CHANNEL_4}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(4), {94, CHANNEL_5}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(5), {94, CHANNEL_6}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(6), {94, CHANNEL_7}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(7), {94, CHANNEL_8}},
};

Bank<2> bank1 = {-8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  A0,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(12), {82, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(13), {82, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(14), {82, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(15), {82, CHANNEL_8}},
};
[... 54 lines cut ...]
};
using namespace MIDI_Notes;//utilise le fichier nommé midi note dans la bibliothèque afin de ne pas à devoir écrire les note en hexadecimal

NoteButton pads[] = {
  {5, {note(C, 2), CHANNEL_1}},// note c2 sur le canal midi1
  {6, {note(Db, 2), CHANNEL_2}},//ne pas écrire de# mais utilisé les bémols le #ferait planter la compilation du code
  {7, {note(D, 2), CHANNEL_3}},
  {8, {note(Eb, 2), CHANNEL_4}},
  {9, {note(E, 2), CHANNEL_5}},
  {10, {note(F, 2), CHANNEL_6}},
  {11, {note(Gb, 2), CHANNEL_7}},
  {12, {note(G, 2), CHANNEL_8}},// Note C4 on MIDI channel 1
};
void setup() {
  Control_Surface.begin(); // initialise la library surface de control

 

}                           
void loop() {
  

  Control_Surface.loop(); // Update the Control Surface

}```

tttapa, 2020-03-14:
This is because of a bug that was fixed in #111. You now have to specify the size of the bank when creating a `CCButtonLatched` object:

```cpp
Bankable::CCButtonLatched<2> buttonmute[] = { 
  // ... 
};
```

benwadub, 2020-03-15:
do you think it s possible to make on my drum controller button for changing the midi channel and also a transposer button?
I tried this but can t find again how to write it properly
```
};
using namespace MIDI_Notes;//utilise le fichier nommé midi note dans la bibliothèque afin de ne pas à devoir écrire les note en hexadecimal
// Transposer to go +1 to -1 octave
Transposer<-1, +1> transposer(12);
IncrementDecrementSelector<transposer.getNumberOfBanks()> transposeSelector = {
  transposer,
  {A0, A1}, // Press push button A0 → +1 octave, A1 → -1 octave, A0+A1 → reset
  Wrap::Clamp,
};
IncrementSelector<1> selection canal = {
  bank4, // banque sélectionnée
  {mux3.pin(12), mux3.pin(13)}, 
Bank <2> bank4 = {12};
Bankable::NoteButton pads[] = {
  {{bank4, transposer, 5, {note(C, 2), CHANNEL_1}},// note c2 sur le canal midi1
  {{bank4, transposer, 6, {note(Db, 2), CHANNEL_1}},//ne pas écrire de# mais utilisé les bémols le #ferait planter la compilation du code
  {{bank4, transposer, 7, {note(D, 2), CHANNEL_1}},
  {{bank4, transposer, 8, {note(Eb, 2), CHANNEL_1}},
  {{bank4, transposer, 9, {note(E, 2), CHANNEL_1}},
  {{bank4, transposer, 10, {note(F, 2), CHANNEL_1}},
  {{bank4, transposer, 11, {note(Gb, 2), CHANNEL_1}},
  {{bank4, transposer, 12, {note(G, 2), CHANNEL_1}},
  {{bank4, transposer, 13, {note(Ab, 2), CHANNEL_1}},
  {{bank4, transposer, 14, {note(A, 2), CHANNEL_1}},
  {{bank4, transposer, 15, {note(B, 2), CHANNEL_1}},
  {{bank4, transposer, 16, {note(Bb, 2), CHANNEL_1}},// 
};```

tttapa, 2020-03-15:
You cannot just add extra arguments to in the braces, the argument list is well-defined, and can be found in the [**Bankable::NoteButton documentation**](https://tttapa.github.io/Control-Surface-doc/Doxygen/de/d36/classBankable_1_1NoteButton.html#a8d9831ac818aaf36d7e2869cd3a79807). The `Bankable::NoteButton` constructor only has a single `OutputBankConfig` argument, so adding a second bank will result in a compiler error. The braces are also unbalanced, there are more `{` than `}`.

Having multiple banks control a single element is not supported out of the box, but most of it is there, so you just need a couple of lines of boilerplate code to glue everything together:

```cpp
#include <Encoder.h>
#include <Control_Surface.h>

namespace Custom {

template <uint8_t N>
class SingleAddressMultipleBanks {
  public:
    SingleAddressMultipleBanks(const Array<BankableMIDIOutput, N> &banks,
                               MIDIAddress address)
        : banks{banks}, address{address} {}

    MIDIAddress getBaseAddress() const { return address; }

    MIDIAddress getActiveAddress() const {
        MIDIAddress address = getBaseAddress();
        for (const BankableMIDIOutput &bank : banks)
            address += bank.getAddressOffset();
        return address;
    }

    void lock() {
        for (BankableMIDIOutput &bank : banks)
            bank.lock();
    }

    void unlock() {
        for (BankableMIDIOutput &bank : banks)
            bank.unlock();
    }

  private:
    Array<BankableMIDIOutput, N> banks;
    MIDIAddress address;
};

using Bankable::MIDIButton;
class DualBankNoteButton : public MIDIButton<SingleAddressMultipleBanks<2>, 
                                             DigitalNoteSender> {
  public:
    DualBankNoteButton(const Array<BankableMIDIOutput, 2> &banks, 
                       pin_t pin, MIDIAddress address, uint8_t velocity = 0x7F)
        : MIDIButton{{banks, address}, pin, {velocity}} {}
};

} // namespace Custom

// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //

USBDebugMIDI_Interface midi;

Bank<2> bank(8);
Transposer<-1, +1> transposer(12);

SwitchSelector banksel = {bank, 11};
EncoderSelector<3> transpsel = {transposer, {0, 1}};

using namespace MIDI_Notes;
Custom::DualBankNoteButton button = {
    {{
        {bank, BankType::CHANGE_CHANNEL},
        {transposer, BankType::CHANGE_ADDRESS},
    }}, 
    5, 
    {note(C, 2), CHANNEL_1},
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

benwadub, 2020-03-15:
I think I nearly got it but get a problem at the end : 
could not convert '{{{{bank4, CHANGE_CHANNEL}, {transposer, CHANGE_ADDRESS}}}, 5, {CS::MIDI_Notes::note(0, 2), CS::CHANNEL_1}, 6, {CS::MIDI_Notes::note(1, 2), CS::CHANNEL_1}, 7, {CS::MIDI_Notes::note(2, 2), CS::CHANNEL_1}, 8, {CS::MIDI_Notes::note(3, 2), CS::CHANNEL_1}, 9, {CS::MIDI_Notes::note(4, 2), CS::CHANNEL_1}, 10, {CS::MIDI_Notes::note(5, 2), CS::CHANNEL_1}, 11, {CS::MIDI_Notes::note(6, 2), CS::CHANNEL_1}, 12, {CS::MIDI_Notes::note(7, 2), CS::CHANNEL_1}, 13, {CS::MIDI_Notes::note(8, 2), CS::CHANNEL_1}, 14, {CS::MIDI_Notes::note(9, 2), CS::CHANNEL_1}, 15, {CS::MIDI_Notes::note(11, 2), CS::CHANNEL_1}, 16, {CS::MIDI_Notes::note(10, 2), CS::CHANNEL_1}}' from '<brace-enclosed initializer list>' to 'Custom::DualBankNoteButton'

here is my full sketch
'''cpp

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
 //USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

auto &serial = Serial1;// Selectionne le port série à utiliser
//USBMIDI_Interface usbmidi;
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};
HardwareSerialMIDI_Interface hwmidi = Serial1;
HardwareSerialMIDI_Interface unomidi = Serial2;
MIDI_Pipe pipe;
//};// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

 
//using namespace MIDI_Notes;  ces lignes midi servent à activer un bouton qui envoit une note midi enlever les // sur ces 5 lignes pour l'activer

// démarrer une touche midi
//NoteButton button = {
  //5,                       // numéro de l'entrée arduino sur laquelle le bouton est branché
  //{note(C, 2), CHANNEL_5}, // note C4 sur le canal 5 libre à vous de changer les numéro et d'en ajouter
//};
CD74HC4067 mux1 = {
  A2,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};
Bank<2> bank2 = {8}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched<2> buttonmute [] = { //ces lignes déclarent le mux
  {{bank2,BankType::CHANGE_CHANNEL}, mux1.pin(0), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(1), {94, CHANNEL_2}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(2), {94, CHANNEL_3}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(3), {94, CHANNEL_4}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(4), {94, CHANNEL_5}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(5), {94, CHANNEL_6}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(6), {94, CHANNEL_7}},
  {{bank2,BankType::CHANGE_CHANNEL},mux1.pin(7), {94, CHANNEL_8}},
};

Bank<2> bank1 = {-8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  A0,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(12), {82, CHANNEL_5}},
[... 132 lines cut ...]
    15, {note(B, 2), CHANNEL_1},
    16, {note(Bb, 2), CHANNEL_1}
    
};

void setup() {
  // All incoming midi messages from UNO are forwarded to MIDI device:
  unomidi >> pipe >> hwmidi; 
   // Control Surface will use this MIDI interface as the default one:
  hwmidi.setAsDefault();
  Control_Surface.begin(); // initialise la library surface de control

 /*Connect TX1 of the Teensy to the MIDI input of your MIDI device.
Connext TX of the Arduino UNO to RX2 of the Teensy.

On the UNO, just use a single HardwareSerialMIDI_Interface hwmidi = Serial; without any explicit routing.
*/
}                           
void loop() {
  

  Control_Surface.loop(); // Update the Control Surface

}
'''

tttapa, 2020-03-15:
```cpp
Custom::DualBankNoteButton button = {
    {{
        {bank4, BankType::CHANGE_CHANNEL},
        {transposer, BankType::CHANGE_ADDRESS},
    }}, 
    5, {note(C, 2), CHANNEL_1},
    6, {note(Db, 2), CHANNEL_1},
    7, {note(D, 2), CHANNEL_1},
    8, {note(Eb, 2), CHANNEL_1},
    9, {note(E, 2), CHANNEL_1},
    10, {note(F, 2), CHANNEL_1},
    11, {note(Gb, 2), CHANNEL_1},
    12, {note(G, 2), CHANNEL_1},
    13, {note(Ab, 2), CHANNEL_1},
    14, {note(A, 2), CHANNEL_1},
    15, {note(B, 2), CHANNEL_1},
    16, {note(Bb, 2), CHANNEL_1}
};
```
You're trying to initialize a single button with way too many parameters.  
The code I posted initializes just one object, not an array.

To initialize an array, you take the initializer list for a single objecte, i.e.
```cpp
{
    {{
        {bank, BankType::CHANGE_CHANNEL},
        {transposer, BankType::CHANGE_ADDRESS},
    }}, 
    5, 
    {note(C, 2), CHANNEL_1},
}
```
And you create a list of those, separated by commas, with another pair of braces around it:

```cpp
Custom::DualBankNoteButton array_of_buttons[] = {
  {
      {{
          {bank, BankType::CHANGE_CHANNEL},
          {transposer, BankType::CHANGE_ADDRESS},
      }}, 
      5, 
      {note(C, 2), CHANNEL_1},
  },
  {
      {{
          {bank, BankType::CHANGE_CHANNEL},
          {transposer, BankType::CHANGE_ADDRESS},
      }}, 
      6, 
      {note(Db, 2), CHANNEL_1},
  },
  // etc.
};
```

You have to think about what each level of braces actually _means_. In the last snippet, the outer braces are the initializer for the array. An array is initialized using a list of elements. Each element is initialized using the `Custom::DualBankNuteButon` constructor, as defined in the code above, the arguments to this constructor are well-defined and can be found in the documentation.

Always check the constructor declaration and documentation:
```cpp
DualBankNoteButton(const Array<BankableMIDIOutput, 2> &banks, 
                       pin_t pin, MIDIAddress address, uint8_t velocity = 0x7F)
```
We'll ignore the velocity here, since it has a default value. The constructor signature then becomes:
```cpp
DualBankNoteButton(const Array<BankableMIDIOutput, 2> &banks, pin_t pin, MIDIAddress address)
```
Study the following snippet very carefully:
```cpp
Custom::DualBankNoteButton array_of_buttons[] = { // start of array initialization
  { // start of initialization of the first array element
      {{ // start of initialization of the first constructor argument: const Array<BankableMIDIOutput, 2> &banks
          {bank, BankType::CHANGE_CHANNEL}, // first element of the `banks` Array
          {transposer, BankType::CHANGE_ADDRESS}, // second element of the `banks` Array
      }}, // end of initialization of first constructor argument
      5, // second constructor argument: pin_t pin
      { // start of initialization of third constructor argument: MIDIAddress address
        note(C, 2), // first argument to MIDIAddress constructor
        CHANNEL_1, // second argument to MIDIAddress constructor
      }, // end of initialization of third constructor argument
  }, // end of initialization of first array element
  // ...
}; // end of array initialization
```

benwadub, 2020-03-15:
thanks for all this detailed explanation!

benwadub, 2020-03-15:
ok thanks so I did the array thanks to you! but when I compile now I get: expected '}' at end of input
the problem seems to be in the custom dual bank button part but can t find where the } must be added 
```cpp

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
//USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

auto &serial = Serial1;// Selectionne le port série à utiliser
//USBMIDI_Interface usbmidi;
//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};
HardwareSerialMIDI_Interface hwmidi = Serial1;
HardwareSerialMIDI_Interface unomidi = Serial2;
MIDI_Pipe pipe;
//};// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

//using namespace MIDI_Notes;  ces lignes midi servent à activer un bouton qui envoit une note midi enlever les // sur ces 5 lignes pour l'activer

// démarrer une touche midi
//NoteButton button = {
//5,                       // numéro de l'entrée arduino sur laquelle le bouton est branché
//{note(C, 2), CHANNEL_5}, // note C4 sur le canal 5 libre à vous de changer les numéro et d'en ajouter
//};
CD74HC4067 mux1 = {
  A2,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};
Bank<2> bank2 = {8}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched<2> buttonmute [] = { //ces lignes déclarent le mux
  {{bank2, BankType::CHANGE_CHANNEL}, mux1.pin(0), {94, CHANNEL_1}}, //numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2, BankType::CHANGE_CHANNEL}, mux1.pin(1), {94, CHANNEL_2}},
  {{bank2, BankType::CHANGE_CHANNEL}, mux1.pin(2), {94, CHANNEL_3}},
  {{bank2, BankType::CHANGE_CHANNEL}, mux1.pin(3), {94, CHANNEL_4}},
  {{bank2, BankType::CHANGE_CHANNEL}, mux1.pin(4), {94, CHANNEL_5}},
  {{bank2, BankType::CHANGE_CHANNEL}, mux1.pin(5), {94, CHANNEL_6}},
  {{bank2, BankType::CHANGE_CHANNEL}, mux1.pin(6), {94, CHANNEL_7}},
  {{bank2, BankType::CHANGE_CHANNEL}, mux1.pin(7), {94, CHANNEL_8}},
};

Bank<2> bank1 = { -8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  A0,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(12), {82, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(13), {82, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(14), {82, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS}, mux2.pin(15), {82, CHANNEL_8}},
[... 125 lines cut ...]
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 13, {note(Ab, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 14, {note(A, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 15, {note(B, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 16, {note(Bb, 2), CHANNEL_1},},

};

void setup() {
  // All incoming midi messages from UNO are forwarded to MIDI device:
  unomidi >> pipe >> hwmidi;
  // Control Surface will use this MIDI interface as the default one:
  hwmidi.setAsDefault();
  Control_Surface.begin(); // initialise la library surface de control

  /*Connect TX1 of the Teensy to the MIDI input of your MIDI device.
    Connext TX of the Arduino UNO to RX2 of the Teensy.

    On the UNO, just use a single HardwareSerialMIDI_Interface hwmidi = Serial; without any explicit routing.
  */
}
void loop() {

  Control_Surface.loop(); // Update the Control Surface

}```

tttapa, 2020-03-15:
You forgot to close the `Custom` namespace.

benwadub, 2020-03-16:
all is working fine!! thanks! here is the full sketch for midi keyboard with up down octave button and up down button for channel selection if you want to add as an example
```cpp

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
//USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

auto &serial = Serial;// Selectionne le port série à utiliser
//USBMIDI_Interface usbmidi;
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};
//HardwareSerialMIDI_Interface hwmidi = Serial;
//HardwareSerialMIDI_Interface unomidi = Serial2;
//MIDI_Pipe pipe;
//};// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

//using namespace MIDI_Notes;  ces lignes midi servent à activer un bouton qui envoit une note midi enlever les // sur ces 5 lignes pour l'activer

// démarrer une touche midi
//NoteButton button = {
//5,                       // numéro de l'entrée arduino sur laquelle le bouton est branché
//{note(C, 2), CHANNEL_5}, // note C4 sur le canal 5 libre à vous de changer les numéro et d'en ajouter
//};
//};
Bank <1> bank4 = {12};
IncrementDecrementSelector<1> selectioncanal = {
  bank4, // banque sélectionnée
  {A2, A3},
  Wrap::Wrap,
};

using namespace MIDI_Notes;//utilise le fichier nommé midi note dans la bibliothèque afin de ne pas à devoir écrire les note en hexadecimal
// Transposer to go +1 to -1 octave
Transposer < -1, +1 > transposer(12);
IncrementDecrementSelector<transposer.getNumberOfBanks()> transposeSelector = {
  transposer,
  {A0, A1}, // Press push button A0 → +1 octave, A1 → -1 octave, A0+A1 → reset
  Wrap::Wrap,
};

namespace Custom {
template <uint8_t N>
class SingleAddressMultipleBanks {
  public:
    SingleAddressMultipleBanks(const Array<BankableMIDIOutput, N> &banks,
                               MIDIAddress address)
      : banks{banks}, address{address} {}

    MIDIAddress getBaseAddress() const {
      return address;
    }

    MIDIAddress getActiveAddress() const {
      MIDIAddress address = getBaseAddress();
      for (const BankableMIDIOutput &bank : banks)
        address += bank.getAddressOffset();
      return address;
    }

    void lock() {
      for (BankableMIDIOutput &bank : banks)
        bank.lock();
    }

    void unlock() {
      for (BankableMIDIOutput &bank : banks)
        bank.unlock();
    }

  private:
    Array<BankableMIDIOutput, N> banks;
    MIDIAddress address;

};

using Bankable::MIDIButton;
class DualBankNoteButton : public MIDIButton<SingleAddressMultipleBanks<2>,
  DigitalNoteSender> {
  public:
    DualBankNoteButton(const Array<BankableMIDIOutput, 2> &banks,
                       pin_t pin, MIDIAddress address, uint8_t velocity = 0x7F)
      : MIDIButton{{banks, address}, pin, {velocity}} {}
};
}
using namespace MIDI_Notes;
Custom::DualBankNoteButton button [] = {
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 5, {note(C, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 6, {note(Db, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 7, {note(D, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 8, {note(Eb, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 9, {note(E, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 10, {note(F, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 11, {note(Gb, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 12, {note(G, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 13, {note(Ab, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 4, {note(A, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 15, {note(B, 2), CHANNEL_1},},
  {{{{bank4, BankType::CHANGE_CHANNEL}, {transposer, BankType::CHANGE_ADDRESS},}}, 16, {note(Bb, 2), CHANNEL_1},},

};

void setup() {
  Control_Surface.begin(); 
}

  Control_Surface.loop(); // Update the Control Surface

}```

closed 2020-04-05 by tttapa