tttapa/Control-Surface #132: Weird latency (?) problem with Pro Micro clone

reip161, 2020-03-09:
Hey there, first of all thanks for the work you've done - it's amazing how painless programming your circuit might be with this library. Disclaimer: I'm neither a programmer or heavy MIDI user.

Few years back, I made a controller based on Arduino Pro Micro clone. It consists of four potentiometers connected to analog inputs and a 4x4 arcade pushbutton matrix to be used as a poor man's MPC drum machine. Somebody helped me with the code back then and it worked well except the notes  assigned to buttons weren't in proper order, so I needed to rewrite the code and fix it. I decided to use your library due to it's simplicity and lots of features. I'm quite certain that this is not a hardware issue, this is why I'm opening it here.

To start, potentiometers work just fine, exactly as expected. The device is recognised as a midi one without a hassle. All of the pushbuttons send midi signal I wanted them to send - this might be considered a success. The problem starts when I try to push tem quickly or not in regular intervals - it acts like it has the limit of hits per second, so I'm not able to play freely and create more complex patterns. When I hit several buttons in the same time, the signal for some of them comes later, which distorts the whole game and ruins the fun. 

I'm running KDE Neon (based on Ubuntu 18.04), using Jack Server and my internal Intel soundcard I tested it with Bitwig and vmpk - they both returned the same problem - I can't play dynamically with either of them. I am happy to post any additional information, if you need it.
Here is the code I'm using in my controller (I combined it from examples I found on the library page):

```cpp

#include <Control_Surface.h>
#include <MIDIUSB.h> 

USBMIDI_Interface midi;
CCPotentiometer pot = { A0, MIDI_CC::General_Purpose_Controller_1 };
CCPotentiometer pot1 = { A1, MIDI_CC::General_Purpose_Controller_2 };
CCPotentiometer pot2 = { A2, MIDI_CC::General_Purpose_Controller_3 };
CCPotentiometer pot3 = { A3, MIDI_CC::General_Purpose_Controller_4 };

// The note numbers corresponding to the buttons in the matrix
const AddressMatrix<4, 4> addresses = {{
  {1, 2, 3, 4},
  {5, 6, 7, 8},
  {9, 10, 11, 12},
  {13, 14, 15, 16},
}};
 
NoteButtonMatrix<4, 4> buttonmatrix = {
  {5, 4, 3, 2}, // row pins
  {6, 7, 8, 9},    // column pins
  addresses,    // address matrix
  CHANNEL_1,    // channel and cable number
};
void setup() { Control_Surface.begin(); }
void loop() { Control_Surface.loop(); }

```

tttapa, 2020-03-10:
The ButtonMatrix classes limit the number of button changes per second, this is important to eliminate switch contact bounce when using them as toggle controls (e.g. mute and solo buttons on a mixing console).

As you've found, this can be quite problematic when you press many buttons in a short period of time, when playing notes, for example.

AFAIK, there are two alternatives: 
1. remove the debouncing code
2. debounce each switch in the matrix separately

The first option is definitely the easiest, you can just comment out this if statement:

https://github.com/tttapa/Control-Surface/blob/61ccbbe27664e40192f4ad26b33e18afec021a15/src/AH/Hardware/ButtonMatrix.ipp#L18-L22

The second option requires a lot of memory, which is why the Control Surface library doesn't do it out of the box.

reip161, 2020-03-11:
I commented out lines 21 and 22 - it solved everything. Thank you very much!

closed 2020-03-11 by reip161