tttapa/Control-Surface #133: How to send multiple midi cc from one potentiometer?

kentforth, 2020-03-11:
I have  a joystick. Y Position of joystick is like a potentiometer. that send one midi cc. What I want to do: 
 When I press up(Y position of joystick) - joystick send midi cc 55, if I press down(Y position of joystick) joystick send midi cc 102.

**Is it possible to devide potentiometer on multiple midi cc?**
 Like one pot can send two midi cc according to position of pot

**Can you give me a sample sketch?**

tttapa, 2020-03-11:
You can easily create your own MIDI Output Element for this purpose:  
https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html

You can find some inspiration in the implementation of the [MIDIFilteredAnalog](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/de4/Abstract_2MIDIFilteredAnalog_8hpp_source.html) class.

kentforth, 2020-03-15:
Thank you for answer!

closed 2020-03-15 by kentforth