tttapa/Control-Surface #134: Problem with HairlessMIDI_Interface - seems to freeze on Control_Surface.begin()

LiamInJapan, 2020-03-14:
I'm having problems getting HairlessMIDI_Interface to work. 
I am trying to bridge from AbletonLive with HairlessMIDI with this:

```
#include <Control_Surface.h>

HairlessMIDI_Interface midi;

void sendIR(uint32_t cmd) {
  Serial << "Send IR 0x" << hex << cmd << dec << endl;
  // Implement this
  Serial.print("sendIR");
}

void setup()
{
  Serial.begin(115200);
  Serial.print("setup");
  Control_Surface.begin();
}

struct MyCallback final : SimpleNoteCCValueCallback {
  // Function is called when note event for given range of notes is received.
  void update(const INoteCCValue &noteInput, uint8_t index) override {
    Serial.print("MyCallback");
  }
};
 
// Listen to a range of 2 notes, and use MyCallback.
GenericNoteRange<2, MyCallback> noteInput = {
  0,            // first note number
  MyCallback(), // callback to use
};

void loop() {
  Control_Surface.loop();
}
```

I expect to see "MyCallback" print to the serial monitor on playing the events in Ableton. I can see the events enter Hairless correctly (although they say "SystemMessage #8") and I've set up all baud rates to 115200. But what I get is "se" come through the monitor... Then nothing.

It's like everything freezes on Control_Surface.begin();... Any advice would be appreciated!!

tttapa, 2020-03-14:
Hi, I posted a reply on (what I believe to be) your [thread on the Arduino Forum](https://forum.arduino.cc/index.php?topic=653004):

> 
> The problem is that you are calling Serial.print() before Control_Surface.begin().
> 
> Control_Surface.begin() will initialize the HairlessMIDI_Interface, which will in turn call Serial.begin() again. Calling Serial.begin() multiple times is not an issue, but if you've previously added data to the buffer, this can corrupt it.
> 
> Simple example to reproduce:
> ```cpp
> void setup() {
>   Serial.begin(115200);
>   Serial.println("A first line");
>   Serial.begin(115200);
>   delay(2000);
>   Serial.println("Another line");
> }
> 
> void loop() {}
> ```
> 
> The solution: Either print after Control_Surface.begin(), or if you need to print before it, call Serial.flush() right before Control_Surface.begin.
> 
> The reason you're not seeing MyCallback printed is because you're not getting the right messages. The callback will only fire when note messages for notes 0 and 1 are received on MIDI channel 1, the code I posted doesn't react to "SystemMessage \#8".
> 
> I've just tried the original sketch I posted in reply \#1, and it works correctly if I send note events like "90 00 7F" (note on \#0, channel 1, velocity 0x7F) and "90 01 10" (note on \#1, channel 1, velocity 0x10).
>

LiamInJapan, 2020-03-15:
Thanks! 

And just one final question if I may, how do I modify the channel from 1 to say 10 (e.g. so the callback will call on different channels)

tttapa, 2020-03-15:
You can use 

```cpp
// Listen to a range of 2 notes, and use MyCallback.
GenericNoteRange<2, MyCallback> noteInput = {
  {0, CHANNEL_10}, // first note number, MIDI channel 10
  MyCallback(), // callback to use
};
```

More information about addresses [**here**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/dc6/md_pages_Basics.html).

LiamInJapan, 2020-03-16:
Ok that's really useful, thanks.

But despite all this, and trying a lot of different things, I just can't get my Roland TD-3 to talk to the Arduino environment via a UX16 and Hairless_MIDI. Here are the permutations I've tried:

- CHANNEL_9 10 11 in case of some "by 1" misunderstanding that I'm not aware of
- USBMIDI_Interface vs. HairlessMIDI_Interface
- The notes are coming through from the drum to HairlessMIDI and I can see events such as MIDIIN:  CH10 Note 38 velocity 40 each time I hit a drum (I have just one drum plugged in at the moment). To be safe I have tried Note 37/38/39 again in case there is a by-one error that I am not aware of.

But despite this, nothing on the Arduino-side, the callback just doesn't want to call :/

Here is my latest sketch:

```
#include <IRremote.h>
#include <Control_Surface.h>

//USBMIDI_Interface midi;
HairlessMIDI_Interface midi;

IRsend irsend;

void sendIR(uint32_t cmd) {
  // Serial << "Send IR 0x" << hex << cmd << dec << endl;
  // Implement this
  Serial.print("sendIR");
  //irsend.sendNEC(0xFFE01F,32); // ON
  //delay(2000);
  //irsend.sendNEC(0xFF609F,32); // OFF
  
}

void setup()
{
  Control_Surface.begin();
  Serial.print("setup");
}

struct MyCallback final : SimpleNoteCCValueCallback {
  // Function is called when note event for given range of notes is received.
  void update(const INoteCCValue &noteInput, uint8_t index) override {
    Serial.print("MyCallback");
    if (noteInput.getValue(index) == 0) // check that velocity > 0
      return;
    switch (index) { // index in the range of notes
      case 0: sendIR(0xFFE01F); break;
      case 1: sendIR(0xFF609F); break;
      case 2: sendIR(0xFF609F); break;
      case 3: sendIR(0xFF609F); break;
      case 4: sendIR(0xFF609F); break;
      case 5: sendIR(0xFF609F); break;
      case 6: sendIR(0xFF609F); break;
      case 7: sendIR(0xFF609F); break;
      case 8: sendIR(0xFF609F); break;
      case 9: sendIR(0xFF609F); break;
      
    }
  }
};
  
// Listen to a range of 2 notes, and use MyCallback.
GenericNoteRange<2, MyCallback> noteInput = {
  {38, CHANNEL_10},            // first note number
  MyCallback(), // callback to use
};

void loop() {
  Control_Surface.loop();
}
```

Any advice would be great. I am losing my hair with this one!

tttapa, 2020-03-16:
Is Hairless able to open the port of the Arduino? You cannot use Hairless while you have the Serial monitor open. Using Serial.print() might also confuse Hairless. You only have a single serial port, it's used for MIDI as well.

Your Note Range has a length of two, so the index in your callback can never be higher than 1.

tttapa, 2020-03-16:
I've removed all `Serial.print` statements from your code, and I replaced them with `digitalWrite` statements to turn on/off some LEDs, and it works fine with Hairless for me. If I connect my DAW with Hairless and play some MIDI notes, the LEDs turn on/off accordingly.

LiamInJapan, 2020-03-16:
Still no luck I'm afraid. 

You are right I looked a bit further and it turns out the serial monitor was fighting with hairless. However if I remove all Serial statements from the sketch and turn the serial monitor off, then I see hairless correctly blinking on both the serial port side and the MIDI IN side. 

I have it connected to a remote controlled "disco light" that I have tested I can communicate with. However I try and get it to turn off the light on the hit of the TD-3 and it just doesn't work. Here's my latest:

```
#include <IRremote.h>
#include <Control_Surface.h>

//USBMIDI_Interface midi;
HairlessMIDI_Interface midi; // I have tried both of these
IRsend irsend;

boolean lightOn = false;

void Off()
{
  irsend.sendNEC(0xFF609F,32); // these have been tested independently and work
  lightOn = false;
}

void On()
{
  irsend.sendNEC(0xFFE01F,32); // these have been tested independently and work
  lightOn = true;
}

void ToggleLight() {
  if(lightOn == true)
  {
    Off();
  }
  else
  {
    On(); 
  }
}

void setup()
{
  On();
  delay(2000);
  Off();
  delay(2000);
  On();
  delay(2000);
  Off();
  delay(2000); // These on/off statements work - I see the bulb flashing shortly after upload
  On();
  Control_Surface.begin();
}

struct MyCallback final : SimpleNoteCCValueCallback {
  // Function is called when note event for given range of notes is received.
  void update(const INoteCCValue &noteInput, uint8_t index) override {
// I also tried Off() here but got no result
    if (noteInput.getValue(index) == 0) // check that velocity > 0
      return;
    switch (index) { // index in the range of notes
      case 0: Off(); break;   // This however never seems to get called
    }
  }
};
  
// Listen to a range of 2 notes, and use MyCallback.
GenericNoteRange<1, MyCallback> noteInput = {
  {38, CHANNEL_10},            // first note number
  MyCallback(), // callback to use
};

void loop() {
  Control_Surface.loop();
}
```

Some other things I've tried:
1) Swapping between 38400 and 115200 baud
2) Different timings of turning on the board and hairless

The only thing that I can see that looks suspicious now is that my output in hairless MIDI gets spammed with messages like
```
+709.269 - MIDI In: System Message #14
```
regardless of whether I interact with the pad or not...

tttapa, 2020-03-17:
Hmmm, I don't really know what could be wrong, your code looks alright.

System Message # 14 is Active Sense, it's normal for some MIDI devices to be sending it continuously. One possible route to explore is to check if the device expects that the Arduino responds with active sense messages.

Does it send anything else apart from the System Message # 14? The fact that it's sending Active Sense indicates that the device hasn't sent any other messages for ~300ms.  
If Hairless doesn't show any MIDI Note messages, this is an indication that the MIDI device isn't actually sending any.

***

Let's take a step back and check the connections first.

The following sketch should loop back all MIDI data it receives:

```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;
MIDI_Pipe pipe;

void setup() {
  midi >> pipe >> midi; // loop back all MIDI messages
  midi.begin();
}

void loop() {
  midi.update();
}
```
If the compiler can't find `MIDI_Pipe`, please upgrade the library to the `master` version. You can download it using the green button in the top right of the homepage of this repository.

Next, open Hairless, and connect the Arduino to its serial port. The baud rate should be 115200.  
Connect Hairless's "MIDI In" to something that sends MIDI, such as a MIDI keyboard.  
Connect Hairless's "MIDI Out" to something that can play or display MIDI messages, such as a (software) synth or MIDI monitor.  
Enable "Debug MIDI messages" in Hairless.

If you press a key on the keyboard, you should see it appear twice in the Hairless debug log: once coming from "MIDI In", and once coming from "Serial In", i.e. the response of the Arduino. It should also be played on the synth/monitor connected to "MIDI Out".

I just tested it on an Arduino UNO, with the Hairless MIDI In connected to a hardware USB MIDI device that sends MIDI notes, and MIDI Out connected to a MIDI monitor. These are the results:
[image]

tttapa, 2020-03-17:
> 1. Swapping between 38400 and 115200 baud

By default, Control Surface uses 115200 baud. The baud rate doesn't really matter as long as both Hairless and the Arduino use the same rate. (And as long as it's not too slow, like 9600.)

> 2. Different timings of turning on the board and hairless

This shouldn't matter, as long as Hairless can connect to the board. The board should be connected and the USB device should be enumerated before selecting the port from the drop-down menu in Hairless.

> 3. Running the range from different notes, 37/38/39/40 (the TD3's drum snare comes through as 40, the main hitting area as 38)

The note numbers used by the library start at 0. Note number zero is the first note, and corresponds to pitch C-1. I'm pretty sure most other MIDI devices use the same convention, but it doesn't hurt to check. Can you verify that these notes are in fact received in Hairless?

> 4. Channel 9/10/11 in case of by-1 errors somewhere

Channel 10 should be the correct channel (it's the standard percussion channel).

LiamInJapan, 2020-03-17:
> If Hairless doesn't show any MIDI Note messages

Hairless shows the messages, and I managed to rig it up with your case above, so it'd seem that the messages _do_ get through to the board and back out again. So I'm at a bit of a loss.

Here's my latest:

```
#include <Control_Surface.h>
#include <IRremote.h>

IRsend irsend;

HairlessMIDI_Interface midi;
//MIDI_Pipe pipe;

void on() {
  irsend.sendNEC(0xFFE01F,32); // ON
}

void off() {
  irsend.sendNEC(0xFF609F,32); // OFF
}

void setup() {
  on();
  delay(2000);
  off();
  delay(1000);
  on();
  delay(1000);
  off();
  delay(1000); // this works and I see the bulb flashing a few times at the beginning
  //midi >> pipe >> midi; // loop back all MIDI messages - this works with Hairless and MIDI Monitor
  midi.begin();    
  //Control_Surface.begin();
}

struct MyCallback final : SimpleNoteCCValueCallback {
  // Function is called when note event for given range of notes is received.
  void update(const INoteCCValue &noteInput, uint8_t index) override {
    
    if (noteInput.getValue(index) == 0) // check that velocity > 0
    {  
      on();
      return;
    }
      
    switch (index) { // index in the range of notes
      case 0: on(); break;
      case 1: on(); break;
      case 2: on(); break;
      case 3: on(); break;
      case 4: on(); break;
      case 5: on(); break;
      case 6: on(); break;
      case 7: on(); break;
         
    }
  }
};

GenericNoteRange<8, MyCallback> noteInput = {
  {37, CHANNEL_10},            // first note number
  MyCallback(), // callback to use
};

void loop() {
  midi.update();
  //Control_Surface.loop();
}
[... 559 lines cut ...]

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "/var/folders/tn/srvgdx6s03sdtlzg82x065v40000gn/T/arduino_build_360062/MidiPipeTest.ino.hex"
avrdude: writing flash (10150 bytes):

Writing | ################################################## | 100% 1.64s

avrdude: 10150 bytes of flash written
avrdude: verifying flash memory against /var/folders/tn/srvgdx6s03sdtlzg82x065v40000gn/T/arduino_build_360062/MidiPipeTest.ino.hex:
avrdude: load data flash data from input file /var/folders/tn/srvgdx6s03sdtlzg82x065v40000gn/T/arduino_build_360062/MidiPipeTest.ino.hex:
avrdude: input file /var/folders/tn/srvgdx6s03sdtlzg82x065v40000gn/T/arduino_build_360062/MidiPipeTest.ino.hex contains 10150 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.31s

avrdude: verifying ...
avrdude: 10150 bytes of flash verified

avrdude done.  Thank you.

```

tttapa, 2020-03-17:
The errors like 
```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
```
are most likely because of Hairless claiming the Serial port.

If the Arduino receives the messages but doesn't do anything, the only reason I can think of is that the messages you're sending don't match those that the Arduino is listening for.  
I'll post a simplified sketch in a couple of minutes.

If you only use the MIDI interface (like in the loopback example), using `midi.begin()` and `midi.update()` is fine. If you use other parts of Control Surface, you need `Control_Surface.begin()` and `Control_Surface.loop()`. These functions will call (among other things) call `midi.begin()` and `midi.update()` for you.

tttapa, 2020-03-17:
> Hairless shows the messages

Could you post the exact messages you're getting in Hairless?

***

The following simple sketch works without problems for me. It uses the built-in LED of the Arduino instead of the IR output.

```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;

void on() {
  digitalWrite(LED_BUILTIN, HIGH); // ON
}

void off() {
  digitalWrite(LED_BUILTIN, LOW); // OFF
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  on();
  delay(1000);
  off();
  delay(1000);
  on();
  delay(1000);
  off();
  delay(1000); // this works and I see the bulb flashing a few times at the beginning
  Control_Surface.begin();
}

struct MyCallback final : SimpleNoteCCValueCallback {
  // Function is called when note event for given range of notes is received.
  void update(const INoteCCValue &noteInput, uint8_t index) override {
    
    if (noteInput.getValue(index) == 0) { // check that velocity > 0  
      return;
    }
      
    switch (index) { // index in the range of notes
      case 0: on(); break;
      case 1: off(); break;
    }
  }
};

GenericNoteRange<2, MyCallback> noteInput = {
  {37, CHANNEL_10}, // first note number
  MyCallback(), // callback to use
};

void loop() {
  Control_Surface.loop();
}
```
The base note is 37 (C#2) on MIDI Channel 10, and the Arduino listens to a range of 2 notes, so 37 (C#2) and 38 (D2). If the index in the range is 0 (this corresponds to note 37+0 = 37), the built-in LED of the Arduino is turned on. If the index in the range is 1 (this corresponds to note 37+1 = 38), the LED turns off.

I uploaded the sketch, opened my DAW software, opened Hairless, connected it to the DAW and to the Arduino. Then I created a MIDI clip, set the MIDI channel to channel 10, added some C#2 and D2 notes, and routed the output to Hairless MIDI->Serial.

The LED turns on when note C#2 plays, and turns off when note D2 plays, as expected.

[image]
[image]

Please try out this exact sketch with your setup. It works for me, so if it doesn't work for you, we know something else is wrong.

LiamInJapan, 2020-03-18:
Progress! I can make the internal light flicker to my drum! :) Thanks

It begs the question why does the irSend work outside the drumkit context but not within it :/

I will look into that more here. Thanks for all your help

LiamInJapan, 2020-03-18:
It works! Turns out that Hairless has a bit of an issue where you have to turn the serial port on and off. I noticed this while looking at the internal LED test (that the RX light sometimes wasn't flashing). 

This probably explains the intermittent non-connectivity stuff too.

Thanks @tttapa !

[image]

closed 2020-03-18 by LiamInJapan