tttapa/Control-Surface #135: Button that changes Potentiometer's CC number

kentforth, 2020-03-15:
Hello Pieter!

I have a potentiometer and I have a button
Let's say  my pot send midi cc 25. If I press a button(toggle) I want a pot send midi cc 32. If I press a button again  a pot will send midi cc 25 .
**How can I do that? Can you give me a sample sketch?**

benwadub, 2020-03-15:
hi, use the bank change
'''
Bank<2> bank1 = {-8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  A0,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(12), {82, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(13), {82, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(14), {82, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(15), {82, CHANNEL_8}},
};
'''
here is a part of my sketch

kentforth, 2020-03-15:
@benwadub 

``Bank<2> bank1 = {-8}; ``
What -8 means?

Where is the button pin that changes banks?

benwadub, 2020-03-15:
oops sorry it on a mix declared before in the sketch
'''
IncrementSelector<2> selectorbutton = { //déclare les boutons de changement de bank
  bank1, // banque sélectionnée
   mux3.pin(8), // pins sur lequel les boutons sont branchés
};
'''

benwadub, 2020-03-15:
-8 mean that when I push the button cc number pass from 83 to 75 in my exemple

kentforth, 2020-03-15:
@benwadub 
Thank you for your answer, but is it possible to make change every button's midi cc on particular midi cc in bank ?
Example: button1 with midi cc 25 - > 28
button2 with midi cc 33 -> 40
button3 with midi cc 45 -> 46

tttapa, 2020-03-15:
If you use the normal `Bankable` elements, all addresses are changed by the same offset, depending on the number of tracks per bank (the -8  some posts back).

You can also use the ManyAddresses elements, this allows you to switch between arbitrary addresses:

```cpp
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Two banks
Bank<2> bank;

// Selector to change the active bank
SwitchSelector selector = {
  bank, // bank to change the setting of
  11,   // switch pin
};

// Instantiate a CCPotentiometer object with two addresses, depending on
// which bank is active
Bankable::ManyAddresses::CCPotentiometer<2> potentiometer = {
  bank, // Bank that controls this element
  A0,   // Analog pin connected to potentiometer
  {{
    {25, CHANNEL_1}, // Address of bank 1
    {28, CHANNEL_6}, // Address of bank 2
  }},
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```
See these documentation pages for more information:
- [**Bank example**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d74/Bank_8ino-example.html)
- [**ManyAddresses elements**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dc0/group__ManyAddressesMIDIOutputElements.html)
- [**Selectors**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html)

closed 2020-03-15 by kentforth