tttapa/Control-Surface #136: Indicate each bank with led

kentforth, 2020-03-15:
I have 4 buttons for banks
I have 6 buttons that send midi cc
I have 4 leds.
Leds are connected via SPI through shift register
Each bank should have it's own led. If I press button bank1 the led for that bank should light up and all other leds should turn off . The same for all other button banks. Only one led can be active at a time
**How can I do that? Can anyone give an example?**

What I have now:

```
#include <Control_Surface.h>  // Include the library

USBMIDI_Interface midi; // Instantiate a MIDI Interface to use

// Instantiate a multiplexer
 CD74HC4067 mux = {
  2,               // input pin
  {3, 4, 5, 6},    // Address pins S0, S1, S2, S3
  // 7,             // Optionally, specify the enable pin
};
\
// Instantiate a shift register as output for the LEDs
SPIShiftRegisterOut<24> sreg = {
  10,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

// Instantiate a CCButtonLatched object
CCButtonLatched buttons[] = {
  {mux.pin(0), 5},
  {mux.pin(1), 6},
  {mux.pin(2), 7}  
};

void setup() {
  Control_Surface.begin();   // Initialize Control Surface
}

void loop() {
  Control_Surface.loop();   // Update the Control Surface
  uint8_t i = 0;
  for (CCButtonLatched &button : buttons)   
    sreg.digitalWrite(i++, button.getState());  //check all leds in shift register
                                                //and light up led if button is pressed
                                                //press a button again to turn  a led off
}
```

tttapa, 2020-03-15:
I think you want to use the [**ManyButtonsSelectorLEDs class**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d68/classManyButtonsSelectorLEDs.html#details).

kentforth, 2020-03-19:
How can I use it? can you give me a sample sketch?

tttapa, 2020-03-19:
Start from the [Bank example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d74/Bank_8ino-example.html) and replace the `IncrementDecrementSelector` with a `ManyButtonsSelectorLEDs` object. 

Have a look at the documentation to see what parameters to pass to the `ManyButtonsSelectorLEDs` constructor. You'll have to pass the bank to use, a list of button pins, and a list of LED pins.

kentforth, 2020-03-19:
What if I want to pass just toggle buttons not banks. 
Can you you write examples for leds reacting on toggle buttons and leds reacrting on toggle buttons as banks?

tttapa, 2020-03-19:
> What if I want to pass just toggle buttons not banks.
> Can you you write examples for leds reacting on toggle buttons and leds reacrting on toggle buttons as banks?

I don't understand what you mean.

kentforth, 2020-03-19:
I mean in ManyButtonsSelectorLeds I have to pass the bank name as first parameter. How can I pass just toggle buttons pins and leds?

tttapa, 2020-03-19:
Then what would it select? It's a selector, it controls the setting of some `Selectable`, in this case a Bank.

kentforth, 2020-03-19:
I have an array of toggle buttons:

```
CCButtonLatched buttons[] = {
  {mux.pin(0), 5},
  {mux.pin(1), 6},
  {mux.pin(2), 7}  
};
```

What I want to do is that I can click the only one toggle button at a time and the sorresponding led will light up. If I click on other button a current led should turn off and other led will light up. So only one toggle button with corresponding led can be used at a time
Can you show me an example?

tttapa, 2020-03-19:
Wouldn't the LEDs go out of sync with the MIDI messages sent by the buttons? What MIDI messages do you expect in what cases?

kentforth, 2020-03-19:
Leds should not react on midi messages, just on buttons state. Buttons will send midi cc

kentforth, 2020-03-19:
I really don't care  how leds will react on any type of input I just need behavior of leds according to button pressed

Let's say I have an array of buttons like this:
```

CCButtonLatched buttons3[] = {
  {mux4.pin(15), {0x7C, CHANNEL_1}}, 
  {mux4.pin(12), {0x7D, CHANNEL_1}},
  {mux4.pin(14), {0x64, CHANNEL_1}}
  {mux4.pin(13), {0x65, CHANNEL_1}}, 
 };
```

tttapa, 2020-03-19:
No, I mean the MIDI output, not MIDI input.

`CCButtonLatched` objects have a state. If you press it once, it turns on and sends one CC message, if you press it again, it turns off and sends another CC message.  
The state of one single button is independent of the other buttons.

What you are describing breaks the independence between the LEDs, but not the independence between buttons, so the relationship between each LED and the corresponding button becomes completely meaningless. 

Please give a clear example of what you want ("if I press this button in this case, I expect this MIDI message to be sent, and I expect the LED to do this").

Petrikland, 2020-03-20:
Good afternoon! I want to join your conversation.
For example: there are 4 buttons (but1, but2, but3, but4) connected to the multiplexer
            there are 4 LEDs (led1, led2, led3, led4) connected to the shift register
            and there is 1 button that we’ll call “selector” - (Sbut1) connected to an Arduino pin.
Needed:
 when you press the but1 button, it sends, for example, midiCC: Note C4 on MIDI channel-1 on
          and the led1 lights up. when you release the button, it continues to light.
           pressing the but1 button again sends midiCC: Note C4 on MIDI channel-1 off
           and led1 goes out. and so with all the other buttons but1, but2, but3, but4 and LEDs led1, 
            led2, led3, led4.
           that is: the buttons are independent of each other, send their own command and turn their 
           LEDs on and off.
           Moreover, they are grouped into one group (bank) and do not respond to incoming MIDI 
           messages.
The button (Sbut1) when pressed should change the commands of this group of buttons (but1, 
          but2, but3, but4) and controlLED operation (led1, led2, led3, led4).
For example: but1 now sends midiCC: Note C4 on MIDI channel-2 on, led1 lights up
                     pressing the but1 button again sends midiCC: Note C4 on MIDI channel-2 off,
                     led1 goes out.

The button (Sbut1) is a selector and switches (controls) the operation of this group of buttons and LEDs. It should have 3 positions, change in turn the channel number of the message midiCC CHANNEL_1, CHANNEL_2, CHANNEL_3.

tttapa, 2020-03-20:
Nice description, it can be implemented as follows:

```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midi;

Bank<3> bank(1); // Change track offset by one, three banks in total
IncrementSelector<3> selector = {
  bank,       // bank to manage
  2,          // button pin
  Wrap::Wrap, // wrap around when max/min setting reached
};

Bankable::CCButtonLatched<3> buttons[] = {
  {{bank, BankType::CHANGE_CHANNEL}, 3, 0x3C}, // {bank config, button pin, CC#}
  {{bank, BankType::CHANGE_CHANNEL}, 4, 0x3D},
  {{bank, BankType::CHANGE_CHANNEL}, 5, 0x3E},
  {{bank, BankType::CHANGE_CHANNEL}, 6, 0x3F},
};

const pin_t ledPins[] = {
  10, 
  11,
  12,
  13,
};

void setup() {
  for (pin_t ledPin : ledPins)
    ExtIO::pinMode(ledPin, OUTPUT);
  Control_Surface.begin();
}

template <class T, size_t N> 
constexpr size_t len(const T (&)[N]) { return N; }

void loop() {
  Control_Surface.loop();
  static_assert(len(ledPins) == len(buttons), "Use the same number of LEDs as the number of buttons!");
  for (size_t i = 0; i < len(buttons); ++i)
    ExtIO::digitalWrite(ledPins[i], buttons[i].getState());  
}
```

(The button pins are Arduino pins for simplicity, but you can simply replace them with `mux.pin(#)` as shown in the multiplexer examples.)

Petrikland, 2020-03-21:
OK, thanks for answer. I'll try to implement in the near future.

Petrikland, 2020-03-21:
Many thanks to the author of the library for the opportunity to implement a long-abandoned project. Thank you for your comprehensive answers. Everything worked out for me! Thanks also to "Kentforth" for his support of my small project. 
                                                          May the force come with you!

kentforth, 2020-03-22:
@tttapa let's say I have 3 buttons and 3 leds
When I press button1  it sends midi cc 25 and the led 1 will light up. If I press button3, it sends midi cc 30,  the  led1 will turn off and the led3 will light up. So if I press any button the corresponding led for that button will light up and other led will be switched off

[VIdeo](https://imgur.com/OsIKBLc)

tttapa, 2020-03-22:
In that case, `CCButtonLatched` is not the right choice.

```cpp
#include <Control_Surface.h>

template <setting_t N>
class CCSelection : public Selectable<N> {
  public:
    CCSelection(const Array<MIDIAddress, N> &addresses)
      : addresses{addresses} {}

    void select(setting_t setting) override {
      setting = Selectable<N>::validateSetting(setting);
      Control_Surface.sendCC(addresses[setting], 0x7F);
    }

  private:
    Array<MIDIAddress, N> addresses;
};

USBMIDI_Interface midi;

CCSelection<3> cc = {{25, 27, 30}};

ManyButtonsSelectorLEDs<3> selector = {
  cc,           // what to select
  {2, 3, 5},    // button pins
  {11, 12, 13}, // led pins
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

kentforth, 2020-03-23:
I tried to apply your code to my needs. Only two buttons work with leds instead of 4. And it works only If I press button on pin mux4.pin(15). If I press anmother button first time those two buttons doesn't work. 

In this case I have 4 buttons on mux and 4 leds on shift register
I checked wiring. It is ok.

**Do you have an Idea why is it so?**

```
#include <Control_Surface.h> // ??????????? ??????????
USBMIDI_Interface midi; // ??????????? USB ??????????
using namespace ExtIO; // Bring the ExtIO pin functions into your sketch
using namespace MIDI_Notes;

template <setting_t N>
class CCSelection : public Selectable<N> {
  public:
    CCSelection(const Array<MIDIAddress, N> &addresses)
      : addresses{addresses} {}

    void select(setting_t setting) override {
      setting = Selectable<N>::validateSetting(setting);
      Control_Surface.sendCC(addresses[setting], 0x7F);
    }

  private:
    Array<MIDIAddress, N> addresses;
};

CCSelection<4> cc = {{0x7C, 0x7D, 0x64, 0x65}};

 CD74HC4067 mux4 = {
  16,               // input pin
  {2, 3, 4, 5},    // Address pins S0, S1, S2, S3
  // 7,             // Optionally, specify the enable pin
};

ManyButtonsSelectorLEDs<4> selector = {
cc, // what to select
{mux4.pin(12), mux4.pin(13), mux4.pin(14), mux4.pin(15)}, // button pins
{sreg.pin(0), sreg.pin(1), sreg.pin(2), sreg.pin(3)}, // led pins
};

void setup() {
  Control_Surface.begin();        // Initialize Control Surface
  sreg.begin();                   // Initialize the shift registers
  mux4.begin();                    //Initialize multiplexer
  mux4.pinMode(0, INPUT_PULLUP);   // Set the pin mode (setting it for one pin of
                                  // the multiplexers sets it for all of them)  
} 

void loop() {
Control_Surface.loop(); // Update the Control Surface
}
```

tttapa, 2020-03-23:
I've just tried it here on a breadboard, and it seems to work just fine. There used to be a problem where the first LED wouldn't be activated upon initialization. This has been fixed recently. Could you try pulling the latest version of the library?

kentforth, 2020-03-23:
I'm using momentary push buttons. I downloaded the library yesterday

tttapa, 2020-03-23:
Could you download it again just to be sure? I committed the change yesterday, but pushed it later.

I'll set up a more extensive test now.

kentforth, 2020-03-23:
Can you create an example for this situation where you can click on only one of the 4 buttons at once  and appropriate led will light up?

tttapa, 2020-03-23:
I've tried your code, and it seems to work just fine for me. The only thing I changed were the pin numbers, and I removed some unnecessary statements from your code. I used an 74HC595 shift register and a 4067 multiplexer.

I did fix a problem with the initialization order when using Hardware UART MIDI interfaces (e.g. on an Arduino UNO). This shouldn't affect you if you're using `USBMIDI_Interface` on a Leonardo, Due or a Teensy. If you're using an UNO or MEGA, you can try updating to the version I just pushed a couple of minutes ago (7793644a90b3ee73c2da9ef6683bf444c801dc07).

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

template <setting_t N>
class CCSelection : public Selectable<N> {
  public:
    CCSelection(const Array<MIDIAddress, N> &addresses)
      : addresses{addresses} {}

    void select(setting_t setting) override {
      setting = Selectable<N>::validateSetting(setting);
      Control_Surface.sendCC(addresses[setting], 0x7F);
    }

  private:
    Array<MIDIAddress, N> addresses;
};

CCSelection<4> cc = {{0x7C, 0x7D, 0x64, 0x65}};

CD74HC4067 mux4 = {
  A4,               // input pin
  {A0, A1, A2, A3}, // Address pins S0, S1, S2, S3
  A5,               // Optionally, specify the enable pin
};

SPIShiftRegisterOut<24> sreg = {
  10,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

ManyButtonsSelectorLEDs<4> selector = {
  cc, // what to select
  {mux4.pin(12), mux4.pin(13), mux4.pin(14), mux4.pin(15)}, // button pins
  {sreg.red(0), sreg.red(1), sreg.red(2), sreg.red(3)}, // led pins
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```
[**Video**](https://youtu.be/cv43IngU-M8)

kentforth, 2020-03-23:
What ontrol_Surface.sendCC(addresses[setting], 0x7F); means?
I mean why do you use particualr number 0x7f?

Why do you use analog pins for control pins(S0-S3) in code if you connect momentary buttons? Does it depent on what type of elemt you connect to control pins? I mean analog or digital

tttapa, 2020-03-23:
0x7F is the value for the control change message. You didn't specify it, so I just used the default "on" value, i.e. the maximum possible value. It's the same value that's used by the CCButton classes if you don't specify it.

Analog pins are just digital GPIO pins that have an extra connection to the ADC. I'm just using them as digital pins here, it doesn't matter (except if it's A6 or A7 on an Arduino Nano).

How did you wire up your buttons?

kentforth, 2020-03-23:
I noticed that you used A5 pin on multiplexer. What is this pin for?

tttapa, 2020-03-23:
It's the enable pin. You only need it if you didn't enable the multiplexer in hardware.

kentforth, 2020-03-23:
Buttons wiring:
one leg -> Arduino GND
another leg -> Arduino pin

tttapa, 2020-03-23:
That's correct. Did you try it without the mux? Have you tested your LEDs separately (there should be examples in File > Examples > Control Surface > Arduino Helpers > Hardware > ExtIO > Output). In the Input-Output folder, there's a mux/shift register toggle example, IIRC.

Have you tried the latest version of the library?

santta, 2020-04-02:
Windows 10 + arduino leonardo + Control Surface (7793644) + 4067 + 74HC595 

Good afternoon, I exactly repeated your components wiring diagram, sketch, installed the library version (644). But I get a wrong job:

- When the board is turned on, led # 1 (sreg pin 0) turn on.

- When press the button # 1 (mux pin 12),switches the LED # 4 (sreg pin 3) , and CC 7D is sent.

- When press the button # 2 (mux pin 13), switches the LED # 1 (sreg pin 0) , and CC 7C is sent.

- Buttons # 3, # 4 are pressed, any the LEDs do not switch, and the CC are not sent.

P.S. i tried to use other pins, multiplexer and shift register. But the wrong principle of work remains.

[image]

Sketch:

#include <Control_Surface.h>

USBMIDI_Interface midi;

template <setting_t N>
class CCSelection : public Selectable<N> {
  public:
    CCSelection(const Array<MIDIAddress, N> &addresses)
      : addresses{addresses} {}

    void select(setting_t setting) override {
      setting = Selectable<N>::validateSetting(setting);
      Control_Surface.sendCC(addresses[setting], 0x78);
    }

  private:
    Array<MIDIAddress, N> addresses;
};

CD74HC4067 mux4 = {
  A4,               // input pin
  {A0, A1, A2, A3}, // Address pins S0, S1, S2, S3
  A5,               // Optionally, specify the enable pin
};

SPIShiftRegisterOut<24> sreg = {
  10,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

CCSelection<4> cc = {{0x7C, 0x7D, 0x64, 0x65}}; 

ManyButtonsSelectorLEDs<4> selector = {
  cc, // what to select
  {mux4.pin(12), mux4.pin(13), mux4.pin(14), mux4.pin(15)}, // button pins
  {sreg.red(0), sreg.red(1), sreg.red(2), sreg.red(3)}, // led pins
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2020-04-02:
>  `{sreg.red(0), sreg.red(1), sreg.red(2), sreg.red(3)}, // led pins`

Your picture shows the LEDs connected to pins 0, 1, 2, 3 of the shift register. In that case, you have to use `sreg.pin(x)`, not `sreg.red(x)`.

It still sounds to me like a hardware issue, since the exact same code works perfectly fine for me. 

Please double-check your wiring, and try the DigitalReadSerial example to ensure that your mux is working correctly: https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d42/2_8DigitalReadSerial_8ino-example.html

santta, 2020-04-02:
In the image, I indicated "sreg pin x" only to visually indicate which pin in the shift register the LED is connected to.  I used in sketch "sreg.red (x)".  

Test sketch for checking mux, displayed in Serial Monitor: 

- mux pin 0-11 always sends "1". 
- mux pin 12-15 sends a "0" when button rising , and sends a "1" when  button falling  

I experimented a lot and found out that only 2 buttons and 2 LEDs can work.
Tried all the pins of the MUX and the shift register. All of them work only in the first two positions of ManyButtonsSelectorLEDs
[image]

tttapa, 2020-04-02:
> Test sketch for checking mux, displayed in Serial Monitor:
> -mux pin 0-11 always sends "1".
> -mux pin 12-15 sends a "0" when button rising , and sends a "1" when button falling
> MUX seems to be fixed

That sounds right.
Does it work now?

As you can see, there's nothing in the ManyButtonsSelector code that could explain why only the two first buttons work:

https://github.com/tttapa/Control-Surface/blob/935ebe9ad17a82981d0b561c8fa823bb10f54004/src/Selectors/ManyButtonsSelector.hpp#L26-L36

santta, 2020-04-02:
not only the FIRST two connected buttons work - any two buttons work.

If I connect buttons 2 and 3, they will switch led 2 and 3.
And the connected buttons 1 and 4 will switch leds 1 and 4

tttapa, 2020-04-02:
How does connecting buttons change anything?  
Are you using momentary or latching switches?

santta, 2020-04-02:
And one more incorrect fulfillment discovered by me:

Here is part of the Selection code for an example:

      CCSelection<8> cc = {{0x7A, 0x7B, 0x7C, 0x7D}}; 

      ManyButtonsSelectorLEDs<8> selector = {
      cc, // what to select
      {mux4.pin(1), mux4.pin(2), mux4.pin(3), mux4.pin(4)}, // button pins
      {sreg.pin(1), sreg.pin(2), sreg.pin(3), sreg.pin(4)}, // led pins

MUX pins invert turn on SREG pins:

Since buttons and leds work only in pairs, i get the interaction scheme:
 

MUX | SREG
-- | --
1 | 2
2 | 1
  |  
3 | 4
4 | 3
  |  
1 | 4
4 | 1
  |  
2 | 3
3 | 4

santta, 2020-04-02:
> How does connecting buttons change anything?
> Are you using momentary or latching switches?

I use momentary buttons

tttapa, 2020-04-02:
Ah, I see, I think you've wired your buttons incorrectly.  
The push buttons should be connected between the mux pins and ground.

santta, 2020-04-02:
I now double-checked everything. I definitely use momentary buttons and I have them between the ground and mux.
What could be the problem?

santta, 2020-04-02:
I understood the pattern.
In my scheme, everything works correctly, if  disconnect all the button jumpers from mux, and then simply connecting any button jumper with any mux pin, the LED will switch correctly

tttapa, 2020-04-02:
Yes, your buttons seem normally closed. That means they conduct when the button is _not_ pressed. You can invert all digitalRead results in the ManyButtonsSelector source code.

santta, 2020-04-02:
thanks for the help

closed 2020-04-03 by kentforth