tttapa/Control-Surface #137: enviar y recibir  midi sobre ble, esp32

katakm, 2020-03-16:
hola Peter, gracias por compartir tu libreria, es excelente, felicitaciones.

quisiera preguntarte si la libreria es 100 funcional en esp32, es decir todo funciona o hay limitaciones como por ejemplo que solo permite enviar midi, y tambin quisiera saber si hay que instalarr algun programa extra en windows para que funcione, como se hace con el arduino uno y hairless. muchas gracias.

tttapa, 2020-03-17:
You can both send and receive MIDI over BLE using the ESP32.

The main limitation is that all packets longer than the BLE MTU are truncated. This shouldn't be an issue, as long as you're not sending hundreds of bytes worth of SysEx messages in a very short time.

Another thing that is a bit more tricky is PWM output for LEDs etc. This is a design decision of the ESP32 Core, not a limitation of Control Surface.

IIRC, one of the ADCs could not be used while WiFi/BLE were in use. I don't know if this has been fixed in the ESP Core.

All basic features of the library should work fine on the ESP32 (potentiometers, push buttons, LEDs, ...) and I've successfully used SSD1306 OLED displays as well.

The MIDI BLE support in Windows is very lacking, especially when compared to OSX, iOS and Linux. I'm a Linux user myself, but I've tested the ESP32 examples in the library with the MidiBerry application from the Microsoft store, and it seemed to work fine.

katakm, 2020-03-17:
Gracias Peter por tu pronta respuesta, el controlador que tengo planeado sera con el protocolo mackie, dices que solo se puede congelar si se envian cientos de bytes, dato que desconozco, pero para darte una idea requiero recibir los datos de los primeros 4 canales vumetros y los 4 vpot-rings, no usare pantallas, serian unos 128 leds, no todos estaran encendidos, creo que maximo encenderan 64,  y por supueto los leds feedback de solo, mute, rec, play y stop. Eso seria el controlador. 
Tambien quisiera preguntarte si has usado el esp32 en windows con algun daw. o si para usarlo en windows hay que usar un puente entre el esp32 y el daw. Muchas gracias.

tttapa, 2020-03-19:
> serian unos 128 leds, no todos estaran encendidos, creo que maximo encenderan 64

Still, that's a lot of LEDs. You'll have to do some serious planning for the power supply, you cannot power 64 LEDs from a USB port (64 * 0.020 A = 1.280 A).  
Do you just need on/off control for the LEDs, or do you need PWM control?  
Eiher way, be sure to order decent LED drivers. Many people (including myself) use the 74HC595, but it was never designed to drive LEDs. [TI](http://www.ti.com/power-management/led-drivers/signage-led-display-drivers/overview.html) has many LED driver chips with constant current outputs, that can be controlled by the Arduino just like a standard 74HC595.

> Tambien quisiera preguntarte si has usado el esp32 en windows con algun daw. o si para usarlo en windows hay que usar un puente entre el esp32 y el daw.

I'm not a Windows user, so I can't really tell you. IIRC, I've tried the ESP32 with Reaper, and it wasn't supported out of the box, so I had to use the MidiBerry application as a bridge for it to work. I read somewhere MIDI BLE support in Windows was improved recently (last year or so), so hopefully the major DAWs will implement it in the near future.