tttapa/Control-Surface #139: Toggle Led for every button

kentforth, 2020-03-19:
Hello Pieter!
If I press a button the corrensponding led should light up, and If i press the buton again a led should turn off
Button on pin 2 -> led 8
Button on pin 3 -> led 9
Button on pin 4 -> led 10
Button on pin 5 -> led 11
Button on pin 6 -> led 12
Button on pin 7 -> led 13

**How to toggle led for every button?**

I tried this code but it doesn't work:

```
#include <Control_Surface.h> // Include the Control Surface library

//leds
const int ledPin1 =  8;
const int ledPin2 =  9;
const int ledPin3 =  10;
const int ledPin4 =  11;
const int ledPin5 =  12;
const int ledPin6 =  13;

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

//массив кнопок
CCButtonLatched buttons[] = {
  {2, {MIDI_CC::Channel_Volume, CHANNEL_1}}, //номера пина кнопки, midi cc под названием channel_volume, номер канала
  {3, {MIDI_CC::Channel_Volume, CHANNEL_2}},
  {4, {MIDI_CC::Channel_Volume, CHANNEL_3}},
  {5, {MIDI_CC::Channel_Volume, CHANNEL_4}},
  {6, {MIDI_CC::Channel_Volume, CHANNEL_5}},
  {7, {MIDI_CC::Channel_Volume, CHANNEL_6}},
 };

void setup() {  
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();   // Update the Control Surface
  uint8_t i = 0;
  for (CCButtonLatched &button : buttons)   
    digitalWrite(i++, button.getState());  

}
```

tttapa, 2020-03-19:
You never set the pin mode of the LED pins to `OUTPUT`, and the pins you're using `digitalWrite` on (`i++`) don't correspond to the LED pin numbers at all.

You could try something like this (untested):
```cpp
#include <Control_Surface.h> // Include the Control Surface library

const pin_t ledPins[] =  {8, 9, 10, 11, 12, 13};

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

//массив кнопок
CCButtonLatched buttons[] = {
  {2, {MIDI_CC::Channel_Volume, CHANNEL_1}}, //номера пина кнопки, midi cc под названием channel_volume, номер канала
  {3, {MIDI_CC::Channel_Volume, CHANNEL_2}},
  {4, {MIDI_CC::Channel_Volume, CHANNEL_3}},
  {5, {MIDI_CC::Channel_Volume, CHANNEL_4}},
  {6, {MIDI_CC::Channel_Volume, CHANNEL_5}},
  {7, {MIDI_CC::Channel_Volume, CHANNEL_6}},
};

void setup() {
  for (pin_t ledPin : ledPins)
    pinMode(ledPin, OUTPUT);
  Control_Surface.begin();
}

template <class T, size_t N> 
constexpr size_t len(const T (&)[N]) { return N; }

void loop() {
  Control_Surface.loop();   // Update the Control Surface
  static_assert(len(ledPins) == len(buttons), "Use the same number of LEDs as the number of buttons!");
  for (size_t i = 0; i < len(buttons); ++i)
    digitalWrite(ledPins[i], buttons[i].getState());  
}
```

kentforth, 2020-03-19:
It works!
Thank you grandmaster!

closed 2020-03-19 by kentforth

forsakenrecordingstudios, 2020-03-30:
is there a way for the LEDS to update from when when I go to 1 DAW channel to another if they differ?

forsakenrecordingstudios, 2020-03-30:
As in have the Plugin in the DAW to mirror the Arduino LEDS, I am making a EQ Plugin MIDI Controller, and have the LEDS showing when a FREQ is engaged on the EQ which operates with a button, But if i have say the low frequency ON with one channel, jump to the next with the same plugin and its off, i have to double press the button for it to match back up with the LED. 

Thank you for all that you do with this, it is incredible!

tttapa, 2020-03-30:
The code in this thread doesn't listen for MIDI input. If you want the LEDs to follow the settings in your DAW, you could use the [`NoteLED`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d92/1_8Note-LED_8ino-example.html) class, for example.

This only works if your DAW actually sends the necessary information to the Arduino.