tttapa/Control-Surface #141: Strange behavior of VULEDs.ino example

shlammod, 2020-03-21:
Dear Peter,

I tried example VULEDs.ino but without the shift register. I’ve connected 12 LEDs to my Teensy 3.2, first LED (green) connects to pin 2 and so on, push button connects to A0.
The code compiled and I can upload it to my Teensy, and after reboot all LEDs come on (shouldn’t they be all off?) and when I connect it to my DAW, it kind of works but instead of building up starting at the green LED, all go off and depending on the signal, only the yellow or red LEDs go on. When there’s no sound, all LEDs will eventually light up. Also, when I press the button, nothing seems to change.
Did I do something wrong?

Kind regards,
Rob

tttapa, 2020-03-21:
Could you post the exact code you used? 

How did you connect the LEDs? Each LED should have its own current limiting resistor, and the cathodes should all be connected to ground.    
They should all be off at boot, no matter what. I have a unit test that checks this, so I believe the code for it is correct, but I'll do an on-target test tomorrow.

How did you configure your DAW? Did you select a Mackie Control Universal control surface in the settings?

shlammod, 2020-03-21:
Good morning,

First of all, you've done some very nice work with this library dear sir, keep it up! You are very much appreciated.

Below you'll find my used code. It turns out it's better to do some debugging when you're awake and not almost asleep. I attached the LEDs and some wiring the wrong way round, so when they were supposed to be lit, they went out :-/
Everything seems to be working now. I used pin 15 instead of 13 so my internal LED was not used but the one I put on my breadboard instead, which shows in the code below.

```cpp
#include <Control_Surface.h>
 
USBMIDI_Interface midi;

MCU::VULEDs<12> vu = {
 {{2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 15}}, // Arduino pin numbers
 1,                      // track number [1, 8]
 MCU::VUDecay::Hold,  // how long does it take for the meter to decay
};

Button dotBarModeButton = A0; // Push button to switch between dot and bar mode.
 
void setup() {
  Control_Surface.begin();
  dotBarModeButton.begin();
}
 
void loop() {
  Control_Surface.loop();
  if (dotBarModeButton.update() == Button::Falling) // when button is pressed
    vu.toggleMode(); // toggle between dot and bar mode
}
```

Is there any way to hold the peak LED like you do in your VU-Meter-Bridge.ino video?

Rob

tttapa, 2020-03-21:
Glad to hear you got it working!

> Is there any way to hold the peak LED like you do in your VU-Meter-Bridge.ino video?

The default VULEDs handler is rather simplistic:
https://github.com/tttapa/Control-Surface/blob/c2232d31690620e4aef51e63c2de5ee6191f50bf/src/MIDI_Inputs/LEDs/MCU/VULEDs.hpp#L21-L25

So it's not supported out of the box, but you can relatively easily add support for it by writing your own callback (just like the `VULEDsCallback` in the file linked to above). You can find inspiration in the `VUDisplay` class, it's the code used in the demo with the OLED displays:
https://github.com/tttapa/Control-Surface/blob/c2232d31690620e4aef51e63c2de5ee6191f50bf/src/Display/MCU/VUDisplay.hpp#L49-L65