tttapa/Control-Surface #142: Analog inputs as Banks control

sucrz, 2020-03-22:
Hi, 

first thank for the lib.

I'm making a Midi-controller for Ableton Live mostly rotary encoders based. 

The idea is to use a arduino Mega to set the maximum of rotary encoders, 24 in my case. It Work's pretty well so far, but I would like to implemente banks to change midi-channel and I'm facing a issue : 

I've search everywhere in the lib but for what I've found so far controls to manage banks need to be on digital pins. 

I would like to use analog pins as input to control banks. The idea is to use switches with pull-up resistors to "emulate" digital pin. 

I'm not that good in programming and to be honest I'm more confortable reading examples et lib files than using the documentation. 

May be you can give me some lead where to search of what to change in the files so I can use Analog inputs.

tttapa, 2020-03-22:
"Analog" pins are full-fledged digital pins that are also connected to an ADC. You can use the pins A0, A1, ... just like any other digital pin.

sucrz, 2020-03-22:
Thank for the quick reply, 

Sound like a good news, I've dig I little the "IncrementDecrementSelector" class and look like it use "pinmode" and "digitalRead". 

Can you give me some code example tu use banks with analog in?

tttapa, 2020-03-22:
You can just start from the [Bank example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d74/Bank_8ino-example.html), and replace the pin numbers with the analog pins you want to use:

```cpp
IncrementDecrementSelector<4> selector = {
    bank,       // Bank to manage
    {A0, A1},     // push button pins (increment, decrement)
    Wrap::Wrap, // Wrap around
};
```

sucrz, 2020-03-22:
I've already tried but I think I've messed somewhere cause it doesn't worked. 
I've tried again today and it works fin this time. 

Silly question, I didn't tried yet : 
Can I'use "analogs" in as "digitals" to wire rotary encoders (It would be amazing)?

In any case, thanks again for the quick replies and the awesome lib ;)

sucrz, 2020-03-22:
Answer to myself I've tried to wire rotary encoder to "analog in's" and it works... that really good news!