tttapa/Control-Surface #143: Banks with Mux

diegogauffin, 2020-03-22:
Hey! First of all... Thank you for this awsome tool ! 

I want to make banks with multiplexors inputs but i can not do it, cant find any example for this. Thank you a lot !

tttapa, 2020-03-22:
What exactly do you mean by "creating a bank"? Do you want to create bankable MIDI elements on mux pins, or do you want to create a bank selector on mux pins?

You can just replace Arduino pin numbers with `mux.pin(x)` where `x` is the pin number of the mux.

diegogauffin, 2020-03-22:
Im sorry, my english isnt it so good, thank you for the answer. I want to create bankable MIDI elements on mux pins. I have 16 potentiometers on a mux and want to change the bank of some of them . I m just a beginner so  I 've tried something like this :

#include <Control_Surface.h>  
 
USBMIDI_Interface midi;  
 

CD74HC4067 mux = {
  A2,       // Analog input pin
  {2, 3, 4, 5} 
};

Bank<4> bank(2);

 

IncrementDecrementSelector<4> selector = {
    bank,      
    {2, 3},    
    Wrap::Wrap, 
};
 

 

Bankable::CCPotentiometer volumePotentiometers[] = {
  {bank, BankType::CHANGE_CHANNEL},     // bank configuration
  mux.pin(0),                                   // analog pin
  {MIDI_CC::Channel_Volume, CHANNEL_1} // address
};
                                 

Bankable::CCPotentiometer volumePotentiometers[] = {
  {bank, BankType::CHANGE_CHANNEL},     // bank configuration
  mux.pin(1),                                   // analog pin
  {MIDI_CC::Channel_Volume, CHANNEL_2}, // address
};
 

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2020-03-23:
The line `Bankable::CCPotentiometer volumePotentiometers[]` is a declaration of an array. You're trying to initialize it with the arguments for a single Bankable::CCPotentiometer object, so that won't work.  
The second problem is that you're declaring two variables with the same name, in the same scope, which is not allowed.

### Initialization of `Bankable::CCPotentiometer`

If you look at the [**documentation for the `Bankable::CCPotentiometer` constructor**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/dc0/classBankable_1_1CCPotentiometer.html#aa6bc0a85a69766c2aab7b1e7f402a87c), you'll see that it expects 3 arguments:

```cpp
Bankable::CCPotentiometer(const OutputBankConfig &config, // (1)
                          pin_t analogPin,                // (2)
                          const MIDIAddress &address      // (3)
)
```
If you want to create a `Bankable::CCPotentiometer` object, you have to provide these three parameters using [list initialization](https://en.cppreference.com/w/cpp/language/list_initialization):

```cpp
Bankable::CCPotentiometer myPot = {
    config,  // (1)
    pin,     // (2)
    address, // (3)
};
```

### Initialization of the arguments to the `Bankable::CCPotentiometer` constructor

Then you have to click on the types of the arguments, to see how they are constructed. For example, the [**`OutputBankConfig` class**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d24/structOutputBankConfig.html#a9093359859e908d72689b9ab5dc27855):

```cpp
OutputBankConfig(OutputBank &bank,              // (1.1)
                 BankType type = CHANGE_ADDRESS // (1.2)
)
```
Again, using list initialization:
```cpp
OutputBankConfig config = {
    bank,                     // (1.1)
    BankType::CHANGE_CHANNEL, // (1.2)
};
```

The second argument (`type`) has a default value, so it is optional. You can omit it in the initialization list, and the default value will be used:
```cpp
OutputBankConfig config = {
    bank,                     // (1.1)
};
```
Since there is only one argument in the previous initialization, you can abbreviate it by omitting the braces:
```cpp
OutputBankConfig config = bank;                     // (1.1)
```
The pin number is just a number, and the `MIDIAddress` argument of `Bankable::CCPotentiometer` can be constructed in a way very similar to what we've done for `OutputBankConfig`.

So finally, we get:
```cpp
OutputBankConfig config = {
    bank,                     // (1.1)
    BankType::CHANGE_CHANNEL, // (1.2)
};

pin_t pin = mux.pin(0); // (2)

MIDIAddress address = {
    MIDI_CC::Channel_Volume, // (3.1)
    CHANNEL_1,               // (3.2)
};

Bankable::CCPotentiometer myPot = {
    config,  // (1)
    pin,     // (2)
    address, // (3)
};
```
All of this can be combined by just substituting `config`, `pin` and `address` by their initialization:

```cpp
Bankable::CCPotentiometer myPot = {
    {                             // (1)   
        bank,                         // (1.1)
        BankType::CHANGE_CHANNEL,     // (1.2)
    },
    mux.pin(0),                   // (2)
    {                             // (3)
        MIDI_CC::Channel_Volume,      // (3.1)
        CHANNEL_1,                    // (3.2)
    },
};
```
Carefully study the braces used, and compare them to the ones in the previous snippet.

### Initialization of an array of `Bankable::CCPotentiometer`s

Finally, if you want to create an array, you have to initialize a list of single elements.

For example, a single integer is initialized like this:
```cpp
int i = 42;
```
An array of integers is initialized like this:
```cpp
int array[] = {42, 43, 44};
```
To go from a single integer to an array, we just created a list of single integers, separated by commas, and surrounded by braces.

The exact same principle applies to arrays of more complex objects like `Bankable::CCPotentiometer`:

```Bankable::CCPotentiometer myPots[] = {
    myPot0,
    myPot1,
    myPot2,
};
```
Each element `myPot#` is just a single `Bankable::CCPotentiometer` object, and we already know how to construct that:

```cpp
Bankable::CCPotentiometer myPots[] = {
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(0),                   // (2)
        {                             // (3)
            MIDI_CC::Channel_Volume,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
    {                             // myPot1
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(1),                   // (2)
        {                             // (3)
            MIDI_CC::Channel_Volume,      // (3.1)
            CHANNEL_2,                    // (3.2)
        },
    },
    {                             // myPot2
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(2),                   // (2)
        {                             // (3)
            MIDI_CC::Channel_Volume,      // (3.1)
            CHANNEL_3,                    // (3.2)
        },
    },
};
```
Again, study the different levels of braces very carefully, and understand what they are initializers for.

### In general

The same principle applies to all other objects in C++:  

1. **Look up the constructor in the documentation**
2. **Check what types of arguments it expects**
3. **Look up the constructors of these arguments recursively (repeating steps 1-3)**
4. **Use a new level of braces for each level of constructors**