tttapa/Control-Surface #144: Defining what happens at each press of a button in a button matrix

OrionCul, 2020-03-23:
I have a button matrix set up and when a button is pressed, I want to light up a corresponding LED. I've seen numerous ways of doing this in the library, but the issue I'm having is that I'm using an SX1509 I/O expander so I can't use the NoteToLed library or anything. I was wondering if it's possible to define in the arduino file what happens when a button is pressed. For example, something like this possibly:

```cpp
for (int i = 0; i < buttonMatrixRows; i++) {
    for (int j = 0; j < buttonMatrixCols; j++) {
        if (buttonMatrix[i][j].currentState != buttonMatrix[i][j].previousState) {
            if (buttonMatrix[i][j] == LOW) {
                //Light up LED & whatnot
            } else {
               //Disable LED 
            }
        }
    {
}
```

tttapa, 2020-03-23:
There are two main approaches you could take:
1. You could implement the button matrix etc. yourself. The code you posted is a good starting point, and you could also look at [Control Surface's implementation](https://github.com/tttapa/Control-Surface/blob/master/src/AH/Hardware/ButtonMatrix.ipp) for inspiration.
2. You could implement the [ExtendedInputOutputElement interface](https://github.com/tttapa/Control-Surface/blob/master/src/AH/Hardware/ExtendedInputOutput/ExtendedIOElement.hpp) for the SX1509. This is just an adapter between the Control Surface library and an existing SX1509 Arduino library. Basically, you just tell Control Surface how to `digitalWrite` etc. to SX1509 pins. When you do this, you can use pretty much all parts of the library using pins from your IO expander. This is how the library deals with multiplexed pins and pins of shift registers.

The first option is a quick solution if you know some C++, and the end result will have less overhead, but at a cost of having to copy/adapt some parts of the library to make them work with your custom IO functions.  
The second option just requires you to implement a couple of virtual functions (`digitalWrite`, `digitalRead`, `pinMode`, etc.), and from then on, you can just use the library as usual.

OrionCul, 2020-03-23:
For option one, if I were to implement the button matrix, how would I get it to work with the control surface's library to send out MIDI signals? Or would I have to do that myself?

For option two, would I have to implement all of the virtual functions in that class? Also, where would I implement them so that the whole library uses my defined functions instead?

tttapa, 2020-03-23:
You can use `Control_Surface.sendCC(address, value)` and the other similar functions.

You have to provide an implementation for all pure virtual functions, but you can just leave them empty if you don't need them.  
You don't have to define them in a specific place. Just create a class for the expander that inherits from `ExtendedIOElement` and override its methods. Then create an instance of your class, and use `yourInstance.pin(n)` to pass its pins to other parts of the library.