tttapa/Control-Surface #145: Vuleds con max7219?

alejamelissa, 2020-03-24:
Buen dia amigo.
Me gustaria saber si es posible usar un max7219 para el manejo de vumetro led y demas funciones que usan leds feedback, ya que los 74hc595 son mas costosos para implementar varios leds,soy novata, no se como configurarlo, podrias orientarme un poco con un ejemplo si es posible.
Tambien quisiera implementar en mi controlador midi un vumetro leds del master, cosa que no me ha sido posible hacer, solo lo permite para el audio en mcu de los 8 canales, mas no del master. Tira error al introducir MASTER_VOLUME en lugar de VOLUME_1.

tttapa, 2020-03-24:
Yes, you can use the MAX7219 to drive LEDs. I added an example: <https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/dcb/1_8MAX7219-Blink_8ino-example.html>

To use it for VU meters, you can use the `pin` method, like in this example: <https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/d49/6_8MAX7219-NoteLED_8ino-example.html>

I don't think the MCU protocol supports a VU meter for the master channel. You can have a look at what your DAW sends using this example: <https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d85/Mackie-Control-Universal-Reverse-Engineering_8ino-example.html>

alejamelissa, 2020-03-25:
Se me olvido preguntarte si es posible usar varios max7219 y conectarlos como se hace normalmente para letreros por ejemplo, y para llamar encender el led solo seria camviar e tu ejemplo por max7219.pin (120), por ejemplo, y ya con eso se activaria el segundo max7219 o se tiene que hacer alguna configuracion adicional.
Por otra parte las conexiones del max7219 en tu ejemplo no declara pines para la conexion como lo haces con los shift registers   por lo que no se a donde va conectado el max7219 en mi esp32?.
 Muchas gracias

tttapa, 2020-03-25:
Currently, it's not possible to connect multiple MAX7219s in series. You can use them in parallel, with a separate CS/SS pin. Adding support for daisy chaining in software shouldn't be too hard either.

You have to connect the MAX7219 to the SPI bus of the ESP32 (CLK, MOSI, SS) and to a power supply. You'll also need some bypassing caps and a resistor to set the LED driver current, but these should already be present if you're using an LED display module. 

There's a typical application schematic on Maxim's site: https://www.maximintegrated.com/en/products/power/display-power-control/MAX7219.html

alejamelissa, 2020-03-26:
Gracias amigo, por ultimo se me olvido preguntarte cuantos leds va a manejar el max7219 con tu libreria, es decir, voy a poder usar 64 leds en matriz y usarlos todos para los 8 vumetros de 8 leds cada vumetro. O  solo se puede manejar 8 leds con cada max7219. Gracias

tttapa, 2020-03-26:
Each MAX7219 can drive 64 LEDs.

alejamelissa, 2020-03-30:
Hola amigo, te escribo de nuevo para preguntarte si los anillos led de paneo en MCU funcionan, ya que he estado toda la tarde y no he logrado que se encienda el led indicador, he estado con la funcion VPotRingLEDs, y si bien compila y sube a la placa leonardo  no hay respuesta por parte de los leds todos permanecen apagados, he usado los pins   del 2 al 12 y nada, podrias ayudarme.

tttapa, 2020-03-30:
The documentation for VPotRing is indeed lacking: the "track" argument to the VPotRing constructor is a track number from 1 to 8. `MCU::V_POT_1` is the CC address for _sending_ position changes of the VPots, not for receiving the LED status.

alejamelissa, 2020-03-30:
Acabo de intentar con tu configuracion y no ha funcionado, los leds no responden al movimiento de paneo en el daw ni tampoco al movimiento del encoder.
Mirando en la libreria he encontrado la funcion CCLEDBar y queria preguntarte si esta funciona con el protocolo MCU o solo con el MIDI_CC.
Tambien he subido al leonardo el ejemplo de enable-disable y no funciona.

tttapa, 2020-03-30:
I'll have a look at why the VPots don't work later tonight.

CCLEDBar is for generic CC controllers, not for the MCU protocol.

alejamelissa, 2020-03-30:
Vale amigo, te lo agradesco, estare al pendiente de la solucion que puedas darle a los Vpots, seria genial poder implementarlos. Quedó a la espectativa

tttapa, 2020-03-30:
It should be fixed.

This is the code I used to test it:

```cpp
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

using namespace MIDI_Notes;

// Instantiate a MAX7219 with the SPI slave select pin as latch pin
MAX7219 max7219 = SS;

MCU::VPotRingLEDs vpot = {
  max7219.pins().slice<0, 10>(),
  1,
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
  // No need to initialize the MAX7219 or to set pinModes, all of this
  // is handled by Control Surface.
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

alejamelissa, 2020-03-31:
Gracias amigo, ya hice la prueba y va de lujo, ahora estoy tratando de que trabaje en modo punto pero parece que no esta implemetado en la libreria para los vpots, en la configuracion CCLEDBar si he logrado el modo punto

tttapa, 2020-03-31:
All four VPot LED modes are implemented, they are selected by the DAW.