tttapa/Control-Surface #148: is there a way to test midi input?

benwadub, 2020-03-27:
hi Pieter, hope you re fine, is there a way to show midi message coming from midi din input to serial monitor with the debug interface please?
i m waiting for my hex inverter to build the thru and wanted to test my input before adding the thru in a few days!
thanks
I tried the usbdebugmidi interface but nothing happened so I don t know if I made a mistake or if this is not possible to show these event

tttapa, 2020-03-27:
You can route the MIDI data from the DIN input to the USBDebugMIDI_Interface, as shown in the [MIDI Routing example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dc4/MIDI_Pipes-Routing_8ino-example.html).

```cpp
#include <Control_Surface.h>
 
USBDebugMIDI_Interface midiusb;
HardwareSerialMIDI_Interface midiser = Serial1;
 
MIDI_Pipe pipe;
 
void setup() {
  midiser >> pipe >> midiusb; // all incoming midi from serial is sent to USB 
  midiusb.begin();
  midiser.begin();
}
 
void loop() {
  midiusb.update();
  midiser.update();
}
```

benwadub, 2020-03-27:
Thank you so much! 
Another question come :-)
What the difference between hardwareserialmidiîterface and serial midi interface? The sketch seems to work with both so I don t know which one to keep.

tttapa, 2020-03-27:
`SerialMIDI_Interface` is a generic Serial interface, it works for all types of serial ports that implement the abstract Arduino `Stream` interface and have a `begin(unsigned long)` method.

`HardwareSerialMIDI_Interface` is a specific version of `SerialMIDI_Interface` that only works with `HardwareSerial` UARTs. On an Arduino UNO or MEGA, all Serial ports are hardware UARTs, but this is not the case on Teensy boards, Arduino Leonardo, Due, etc.  
If you try `HardwareSerialMIDI_Interface midi = Serial;` on those boards, you'll get a compilation error, because `Serial` is not a hardware UART but an emulated USB CDC serial port.

`SoftwareSerial` is another type of serial interface that works with `SerialMIDI_Interface`, but not with `HardwareSerialMIDI_Interface`.

benwadub, 2020-03-27:
thanks again for the explanation! my controller is really great using your library!
[image]

now I try to make a synth with a raspberry but it seems to be really harder! I m still not able to put midi din it :-)

tttapa, 2020-03-27:
Looks good!

It might be easier to use MIDI over USB on the Raspberry Pi. Using your Teensy 3.5 (IIRC?) this should work out of the box.

benwadub, 2020-03-27:
yes it work like this, but I wanted to be able to use another midi sequencer (my electron digitakt) for sequencing that synth so I ll need to use din!
 for my debug on the teensy I can t find how to declare the midiser withe the software midi interface :-(
```cpp

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
 USBDebugMIDI_Interface midiusb; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos
//USBMIDI_Interface usbmidi;// ecrire // devant la ligne pour desactiver l'interface usb midi
auto &serial = Serial1;// Selectionne le port série à utiliser, enlever // devant les lignes grisées pour permettre à l'arduino de lire ces lignes et remplacer seria1 par serial pour une arduino à la place de teensy

SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};// démarre une interface midi serial au midi baud rate par defaut

MIDI_Pipe pipe;
```

tttapa, 2020-03-27:
> for my debug on the teensy I can t find how to declare the midiser withe the software midi interface :-(

What do you mean? What doesn't work?

benwadub, 2020-03-27:
You told me that for teensy I could not use hardware midi interface so I think I can t use the sketch you just gave me?
I think I have to use serialmidiinterface for midiser? But don t know where ton name midiser in the sketch I gave you

tttapa, 2020-03-27:
You can use `HardwareSerialMIDI_Interface` on Teensy, but only on the Hardware UARTs, i.e. `Serial1`, `Serial2` etc., but not `Serial`.

`midiser` in the example is just an arbitrary name. You can call it whatever you want.

In the code you posted, you can use `midi >> pipe >> midiusb;`.

benwadub, 2020-03-27:
ok thanks! I had fear that if I changed it I would disable something else! stil learning with you thanks! did you got the result from you're exams?

benwadub, 2020-03-27:
hey but I could also use midi pipe to make MIDI Thru over usb from my midi din input with pipe am I right?

benwadub, 2020-03-27:
usb debug doesn't print anything with pipe, but with usb I see event in the monitor, strange I named them with the same name and tried with both separately.
```cpp

#include <Encoder.h>

// celà doit etre écrit avant la library control surface
#include <Control_Surface.h> // Inclu la library control surface téléchargeable ici https://github.com/tttapa/Control-Surface en cliquant le bouton vert a droite
USBDebugMIDI_Interface midiusb; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos
HardwareSerialMIDI_Interface midiser = Serial1;

auto &serial = Serial1;// Selectionne le port série à utiliser
//USBMIDI_Interface midiusb;
MIDI_Pipe pipe;

// démarre une interface midi serial au midi baud rate par defaut

//vous pouvez aussi utiliser l'interfaces midi usb en enlevant les // à la ligne suivante
//USBMIDI_Interface usbmidi;

CD74HC4067 mux1 = {
  A3,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};
Bank<2> bank2 = {-1}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched<2> buttonmute[] = { //ces lignes déclarent le mux boutons fil rouge
  {{bank2,BankType::CHANGE_ADDRESS}, mux1.pin(7), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(6), {94, CHANNEL_2}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(5), {94, CHANNEL_3}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(4), {94, CHANNEL_4}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(3), {94, CHANNEL_5}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(2), {94, CHANNEL_6}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(1), {94, CHANNEL_7}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(0), {94, CHANNEL_8}},
};

Bank<2> bank1 = {-8}; // 2 banks, 8 addresse per banks

CD74HC4067 mux2 = {// deuxième multiplexeur cette fois si avec 16 entrées
  A0,              // analog pin
  {2, 3, 4, 5}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};
//Instantiate multiplexer for 16 switch
Bankable::CCPotentiometer fx [] = {
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(0), {83, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(1), {83, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(2), {83, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(3), {83, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(4), {83, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(5), {83, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(6), {83, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(7), {83, CHANNEL_8}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(8), {82, CHANNEL_1}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(9), {82, CHANNEL_2}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(10), {82, CHANNEL_3}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(11), {82, CHANNEL_4}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(12), {82, CHANNEL_5}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(13), {82, CHANNEL_6}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(14), {82, CHANNEL_7}},
  {{bank1, BankType::CHANGE_ADDRESS},mux2.pin(15), {82, CHANNEL_8}},
};

Bank <2> bank3 = {8}; // 2 banks, 8 addresse per banks

  //Instantiate second multiplexer for 8 potentiometer
CD74HC4067 mux3 = {
  A1
  ,       // Analog input pin
  {2, 3, 4, 5} // Address pins S0, S1, S2};
[... 84 lines cut ...]
  {mux4.pin(0), {74, CHANNEL_8}},
  {mux4.pin(8), {75, CHANNEL_1}},//filtre reso
  {mux4.pin(9), {75, CHANNEL_2}},
  {mux4.pin(10), {75, CHANNEL_3}},
  {mux4.pin(11), {75, CHANNEL_4}},
  {mux4.pin(12), {75, CHANNEL_5}},
  {mux4.pin(13), {75, CHANNEL_6}},
  {mux4.pin(14), {75, CHANNEL_7}},
  {mux4.pin(15), {75, CHANNEL_8}},
};
void setup() {
  Control_Surface.begin(); // initialise la library surface de control
midiser >> pipe >> midiusb; // all incoming midi from serial is sent to USB 
  midiusb.begin();
  midiser.begin();
 

}                           
void loop() {
  midiusb.update();
  midiser.update();

  Control_Surface.loop(); // Update the Control Surface

}```

benwadub, 2020-03-28:
Is it a bug or did I a mistake?

tttapa, 2020-03-28:
I'm not sure.

Have you tried the simple example I posted earlier, with just the MIDI pipe?

> usb debug doesn't print anything with pipe, but with usb I see event in the monitor, strange I named them with the same name and tried with both separately.

What exactly do you mean? What events do you see?

benwadub, 2020-03-28:
i see thes element without debug interface, I think it s normal
[image]
and with the simple code nothing in the monitor too!

tttapa, 2020-03-28:
What code produces these results, and what results are you expecting?

It looks like you're using a non-debug MIDI interface on `Serial`.

benwadub, 2020-03-28:
yes these result are with the non debug interface, with the debug midi interface instead nothing is printed with the exact same code, seems that the debug midi interface doesn't print on monitor when used with pipe

tttapa, 2020-03-28:
That's strange, I'll have a look at it tomorrow.

tttapa, 2020-03-29:
Are you sure you used the right code? 

I tried this code on a Teensy 3.2, with a MIDI input connected to Serial1 (pins 0 and 1), and it printed the MIDI messages it received to the serial monitor without any problems.

```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midiusb;
HardwareSerialMIDI_Interface midiser = Serial1;

MIDI_Pipe pipe;

void setup() {
  // midiusb << pipe << midiser; // all incoming midi from serial is sent to USB
  midiser >> pipe >> midiusb; // all incoming midi from serial is sent to USB

  midiusb.begin();
  midiser.begin();
}

void loop() {
  midiusb.update();
  midiser.update();
}
```

benwadub, 2020-03-29:
T’es it was the code I posted above :-( I ll retest when I ll have time, my second child is just born 3 days ago and is at home from few hours!

benwadub, 2020-03-29:
I just think about something, with that code my pots and faders on the controller already build would sent cc to monitor or just the incoming by midi in would be written?
If it the first option maybe I built a faulty midi in :-)

tttapa, 2020-03-29:
> T’es it was the code I posted above :-(

I'm not really sure where to start debugging then.

When you've got the time, you could try the following code: it sends two test messages every second, one to the `USBDebugMIDI_Interface` directly, and one as if it were coming from the `HardwareSerialMIDI_Interface` input.

```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midiusb;
HardwareSerialMIDI_Interface midiser = Serial1;

MIDI_Pipe pipe;

void setup() {
  midiser >> pipe >> midiusb; // all incoming midi from serial is sent to USB

  midiusb.begin();
  midiser.begin();
}

Timer<millis> timer = 1000; 
ChannelMessage testmsg1 = {NOTE_ON, 0x01, 0x7F, 0x0};
ChannelMessage testmsg2 = {NOTE_ON, 0x02, 0x7F, 0x0};

void loop() {
  midiusb.update();
  midiser.update();
  
  if (timer) { // Send a test message every second
    midiusb.send(testmsg1);             // send to USB directly
    midiser.sourceMIDItoPipe(testmsg2); // send from serial, through pipe, to USB
  }
}
```

The expected output is something like this:
```
Note On         	Channel: 1	Data 1: 0x01	Data 2: 0x7f	Cable: 0
Note On         	Channel: 1	Data 1: 0x02	Data 2: 0x7f	Cable: 0
Note On         	Channel: 1	Data 1: 0x01	Data 2: 0x7f	Cable: 0
Note On         	Channel: 1	Data 1: 0x02	Data 2: 0x7f	Cable: 0
...
```
I've verified this on my Teensy 3.2. Do you get both messages, just one, or no messages at all?

>  I ll retest when I ll have time, my second child is just born 3 days ago and is at home from few hours!

Congratulations! 

> with that code my pots and faders on the controller already build would sent cc to monitor or just the incoming by midi in would be written?

In the code you posted, the controls from your potentiometers etc. are sent to the `Serial1` `HardwareSerialMIDI_Interface`. To send them to USB instead, you can add `midiusb.setAsDefault();` to the beginning of your `setup` (before `Control_Surface.begin()`).

> If it the first option maybe I built a faulty midi in :-)

That's a possible explanation.  
It's not impossible that there's a problem with the library, of course, but if that were the case, I think my Teensy 3.2 should have the same problem, which isn't the case.

benwadub, 2020-03-29:
Thanks a lot for all this I ll try tomorrow!!

benwadub, 2020-03-30:
ok so your test is working but on my code still nothing is written
```cpp
#include <Control_Surface.h>
 
USBDebugMIDI_Interface midiusb;
HardwareSerialMIDI_Interface midiser = Serial1;
 
MIDI_Pipe pipe;
CD74HC4067 mux1 = {//déclare un multiplexeur
  A3,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2,S3
};
using namespace MIDI_Notes;//utilise le fichier nommé midi note dans la bibliothèque afin de ne pas à devoir écrire les note en hexadecimal

NoteButton pads[] = {//ici on déclare des boutons de declenchement de notes
  {mux1.pin(15), {note(C, 2), CHANNEL_1}},// note c2 sur le canal midi1
  {mux1.pin(14), {note(Db, 2), CHANNEL_2}},//ne pas écrire de# mais utilisé les bémols le #ferait planter la compilation du code
  {mux1.pin(13), {note(D, 2), CHANNEL_3}},
  {mux1.pin(12), {note(Eb, 2), CHANNEL_4}},
  {mux1.pin(11), {note(E, 2), CHANNEL_5}},
  {mux1.pin(10), {note(F, 2), CHANNEL_6}},
  {mux1.pin(9), {note(Gb, 2), CHANNEL_7}},
  {mux1.pin(8), {note(G, 2), CHANNEL_8}},// Note C4 on MIDI channel 1
};
 
void setup() {
  midiusb.setAsDefault();
  Control_Surface.begin();
  midiser >> pipe >> midiusb; // all incoming midi from serial is sent to USB 
  midiusb.begin();
  midiser.begin();
}
 
void loop() {
  midiusb.update();
  midiser.update();
}
```I added the midi usb set as default

tttapa, 2020-03-30:
Does my test code forward the MIDI input from the HardwareSerial interface?

The reason the NoteButtons don't send anything is because there's not `Control_Surface.loop()` in your code.

benwadub, 2020-03-30:
I had no control surface loop in the first code, I didn’t know I had to use it here, I ll try it thanks! I didn’t try midi input!

tttapa, 2020-03-30:
`midi.update()` only  handles MIDI input and routing. You need `Control_Surface.loop()` to update all MIDI elements, ExtIO, etc.

benwadub, 2020-03-30:
ok so with the control surface loop I get the controller printed but not the midi in, I think I made a mistake with my midi in interface! I ll debug the hardware to see where problems come from!
thanks again!

benwadub, 2020-03-30:
I only used 3,3v with a 6n138, I. just looked at the prjrc schematic it seems that they use both 3,3 and 5v do you think my problem can be here or 3,3 would work?

tttapa, 2020-03-30:
What schematic?

benwadub, 2020-03-30:
the one found here
https://www.pjrc.com/teensy/td_libs_MIDI.html
I just try adding the 10k resistor from pin 7 of 6n138 to gnd and pin 8 to vin of the teensy but still nothing I can t find where is my problem

tttapa, 2020-03-30:
Alright, let's try the Serial1 input first then:

Disconnect your MIDI input circuit from Serial1, and connect a wire from pin 10 (TX2) to pin 0 (RX1). Make sure nothing else is connected to pins 0, 1, 9 and 10.

Now you can send MIDI from Serial2, it goes through the wire to Serial1, then it's parsed, sent over the MIDI pipe to the USB interface and to the serial monitor.

```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midiusb;
HardwareSerialMIDI_Interface midiser = Serial1;
HardwareSerialMIDI_Interface miditest = Serial2;

MIDI_Pipe pipe;

void setup() {
  midiser >> pipe >> midiusb; // all incoming midi from serial is sent to USB

  midiusb.begin();
  midiser.begin();
  miditest.begin();
}

Timer<millis> timer = 1000; 
ChannelMessage testmsg1 = {NOTE_ON, 0x01, 0x7F, 0x0};
ChannelMessage testmsg2 = {NOTE_ON, 0x02, 0x7F, 0x0};
ChannelMessage testmsg3 = {NOTE_ON, 0x03, 0x7F, 0x0};

void loop() {
  midiusb.update();
  midiser.update();
  miditest.update();
  
  if (timer) { // Send a test message every second
    midiusb.send(testmsg1);             // send to USB directly
    midiser.sourceMIDItoPipe(testmsg2); // send from serial, through pipe, to USB
    miditest.send(testmsg3);            // send from Serial2, looped to Serial1 using a wire
  }
}
```
Expected output: 
```
Note On         	Channel: 1	Data 1: 0x01	Data 2: 0x7f	Cable: 0
Note On         	Channel: 1	Data 1: 0x02	Data 2: 0x7f	Cable: 0
Note On         	Channel: 1	Data 1: 0x03	Data 2: 0x7f	Cable: 0
...
```

benwadub, 2020-03-30:
this test is working nicely

tttapa, 2020-03-30:
That means that there's probably a problem with your MIDI in circuit.

benwadub, 2020-03-30:
yes I begin debugging hardware! I found lot of different resistor values, do you have a clear schematic somewhere for teensy 3.5 in/out/thru? please? the official midi schematic doesn'tgive resistor for 3,3 v

tttapa, 2020-03-30:
You mean R<sub>D</sub>, the resistor between 3.3V and the collector of the optocoupler? It doesn't matter all that much, anything between 180Ω and 330Ω will probably work fine.

tttapa, 2020-03-30:
Make sure that the polarity of the diode is correct, and that you didn't accidentally swap pins 4 and 5 on the MIDI connector. The MIDI input jack shouldn't be connected to ground. 

If you've double-checked the pinout of the optocoupler, you can make sure it works by connecting an LED in series with R<sub>D</sub>, disconnecting the wire going to pin 5 of the MIDI jack and connecting it to ground (i.e. connecting pin 3 of the optocoupler to ground), disconnecting pin 4 from the MIDI jack and connecting it to 3.3V (i.e. connecting pin 2 of the optocoupler to 3.3V through R<sub>B</sub>). The LED should turn on when you connect this last connection, and off when pin 2 of the optocoupler is not connected to 3.3V.

benwadub, 2020-03-30:
I used a 220 ohm, I have to go to bed but I suspect a bad soldering between 6n138 3 and the diode! And for the midi thru I ll have to use a 470ohm like on the midi out?

tttapa, 2020-03-30:
The MIDI through part is simply a copy of the MIDI out circuit, so you have to use the same resistor values. (2×220Ω for 5V, 33Ω and 10Ω for 3.3V)

benwadub, 2020-03-30:
My midi out works with a 47, don t know where I found this, if that works with out it should work with thru?
I make proper schematics of all my wiring each time it pass the test. I could give you them if you want to add it somewhere in the repo for the next noobs like me ;-)?

tttapa, 2020-03-31:
There are some tolerances, and 47Ω is relatively small compared to the 220Ω resistor in the receiver, so it doesn't matter too much.  
Feel free to post your schematics here :-)

benwadub, 2020-03-31:
[image]
Can t find why this midi input doesn’t work! Do you have an idea how I could debug it with a multimeter by example?

tttapa, 2020-03-31:
It's very hard to see what's wrong. You could measure the voltage between pins 6 and 5 of the optocoupler should be 3.3V if nothing is connected to the MIDI input.  
If you connect a MIDI input, it should drop to some lower voltage during transmission. This is very hard to see on a digital multimeter, so it's probably easier to disconnect the MIDI connector, and put 3.3V across the input manually (don't leave out the 220Ω resistor or you'll blow up the optocoupler). Watch the polarity as well, ground should be connected to pin 3 of the optocoupler. This should be the anode of the external diode (the side without the mark). With 3.3V connected to the input, the output voltage between pins 6 and 5 should be close to 0, something like 200 mV.

benwadub, 2020-03-31:
i have 4v between 5 and 6, I wired pin 8 to 5v volt like on the official midi schematic do you think I have to wire all vcc to 3,3?

benwadub, 2020-03-31:
with all wired to 3,3v I get 3.3 between 5 and 6, when note on or note off send I get 3.2 or 3.1

tttapa, 2020-03-31:
Have you tried manually driving the input as described in my previous post?

benwadub, 2020-03-31:
Yes i get the 200mv as expected!

benwadub, 2020-03-31:
Maybe it’s my midi thru that doesn’t work properly!
I built it this way:
Din pin 5 to input of hexinverter through 470 ohm and to rx
Output of hexinverter to din pin 4 and vcc to 3,3v
Thru is good?

tttapa, 2020-03-31:
I don't understand. What are you using the MIDI through for? The RX pin of the Teensy should be connected to pin 6 of the 6N138.

benwadub, 2020-04-01:
I use midi through to transfer incoming event in from midi in to the next controller via midi through

tttapa, 2020-04-01:
Do you have pin 6 of the 6N138 connected to Teensy pin 0?

benwadub, 2020-04-01:
I wrote an answer yesterday that didn’t write here, I don t know why!
I forgot to wire gnd on the hexinverter 😅 so now all is working fine!! Thanks for support! Now let’s get to add a synth on this project with the teensy audio library cause I have free pins on the teensy :-)
Do you think it s possible to use theses two library together?

tttapa, 2020-04-01:
Glad to hear you got it working!

Yes, you can definitely use the Audio library. There are some classes that I've used myself in the [Teensy Audio module](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d5c/group__Audio.html), and you can find some examples for volume control and VU meters.

benwadub, 2020-04-01:
thanks! their library seems to be not as easy as your to use!

tttapa, 2020-04-01:
It depends how you look at it, and the Audio library does a lot of clever stuff under the hood to ensure audio playback without stuttering etc.  
You can also use the graphical Audio System Design Tool: https://www.pjrc.com/teensy/gui/

benwadub, 2020-04-01:
i m looking their tutorial at this moment but I can t find how I could use a note button from your library to trigger by exemple an oscillator or use a potentiometer of your library to manipulate delay feedback or something else

tttapa, 2020-04-01:
The Audio library doesn't have any classes that correspond directly to the hardware, like NoteButton or CCPotentiometer.  
To control the oscillators and effects, you have to call the functions yourself.

For instance (untested)

```cpp
#include <Audio.h>
#include <Control_Surface.h>

AudioSynthWaveformSine   sine1;
AudioOutputI2S           i2s1;
AudioConnection          patchCord1(sine1, 0, i2s1, 0);

FilteredAnalog<> amplitudeControl = A0; // connect a potentiometer to pin A0

void setup() {
  AudioMemory(4);
  FilteredAnalog<>::setupADC();
  sine1.frequency(440);
}

void loop() {
  if (amplitudeControl.update()) {
    sine1.amplitude(amplitudeControl.getFloatValue());
  }
}
```
You can also use the `Button` class, for example: https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/da9/2_8Button_8ino-example.html

benwadub, 2020-04-01:
thanks! it will send me deeper in programming! nice exercise I think!

benwadub, 2020-04-15:
hi Pieter! I begin to have a great synth with my teensy, I just have question implementing less cause I never did it, could you explain me how can I light less according to midi cc values please?
i have a potentiometer that select a waveform with the teensy audio library and I d like to light a different led at every segment of the cc values

one if cc is under 14, one if cc is between 14 and 28... here is the part of the code
```cpp
 if (srlswc == LOW){
      if(swc <= 14){
      Serial.print ("WAVEFORM = SINE ");
      Serial.println ();}
      if(swc > 14 && swc <= 28){
      Serial.print ("WAVEFORM = TRIANGLE ");
      Serial.println ();}
      if(swc > 28 && swc <= 42){
      Serial.print ("WAVEFORM = TRIANGLE VAR ");
      Serial.println ();}
      if(swc > 42 && swc <= 56){
      Serial.print ("WAVEFORM = SAWTOOTH ");
      Serial.println ();}
      if(swc > 56 && swc <= 70){
      Serial.print ("WAVEFORM = SAWTOOTH REV ");
      Serial.println ();}
      if(swc > 70 && swc <= 84){
      Serial.print ("WAVEFORM = SQUARE ");
      Serial.println ();}
      if(swc > 84 && swc <= 98){
      Serial.print ("WAVEFORM = PULSE ");
      Serial.println ();}
      if(swc > 98 && swc <= 113){
      Serial.print ("WAVEFORM = ARBITRARY ");
      Serial.println ();}
      if(swc > 113 && swc <= 127){
      Serial.print ("WAVEFORM = SAMPLE&HOLD ");
      Serial.println ();}
      ```

tttapa, 2020-04-15:
Something like this, maybe?

```cpp
#include <Control_Surface.h>

CCPotentiometer cc = {A0, 0x10};

pin_t ledPins[] = {2, 3, 4, 5, 6, 7, 8, 9, 10 };

void setup() {
  Control_Surface.begin();
  for (pin_t ledPin : ledPins)
    ExtIO::pinMode(ledPin, OUTPUT);
}

void loop() {
  Control_Surface.loop();
  static uint8_t prevWaveform = 0;
  static bool initialized = false;

  uint8_t waveform = cc.getValue() * 9 / 128;
  if (!initialized || waveform != prevWaveform) {
    switch (waveform) {
      case 0: Serial << "WAVEFORM = SINE" << endl; break;
      case 1: Serial << "WAVEFORM = TRIANGLE" << endl; break;
      case 2: Serial << "WAVEFORM = TRIANGLE VAR" << endl; break;
      case 3: Serial << "WAVEFORM = SAWTOOTH" << endl; break;
      case 4: Serial << "WAVEFORM = SAWTOOTH REV" << endl; break;
      case 5: Serial << "WAVEFORM = SQUARE" << endl; break;
      case 6: Serial << "WAVEFORM = PULSE" << endl; break;
      case 7: Serial << "WAVEFORM = ARBITRARY" << endl; break;
      case 8: Serial << "WAVEFORM = SAMPLE&HOLD" << endl; break;
      default: FATAL_ERROR("Invalid waveform", 0xFAFA); // Shouldn't happen
    }
    ExtIO::digitalWrite(ledPins[waveform], HIGH);
    if (initialized)
      ExtIO::digitalWrite(ledPins[prevWaveform], LOW);
    initialized = true;
    prevWaveform = waveform;
  }
}
```

[6 more replies]